LinuxCNC/linuxcnc #1789: G83: add chip breaking support to peck drilling

silopolis, 2022-07-03:
While peck drilling does the job, it is often loosing machine time to fully retract after each dive.
On Sinumerik I had the option to use "chip breaking with stripping/fluffing", eg use chip breaking with the addition of full retracts every set distance.
This was the perfect combination as it allowed to clear chip just as needed while preserving efficiency, to the point that I never used the bare peck drilling cycle.

A nice addition would be to have the option to set either the distance or the number of pecks separating each full retract.

c-morley, 2022-07-03:
G73 does peck drilling without full withdrawal.
I agree the Sinumerik code sounds better.

silopolis, 2022-07-04:
Le lun. 4 juil. 2022 à 01:53, c-morley ***@***.***> a écrit :

> G73 does peck drilling without full withdrawal.
>

Yes, that's "chip breaking" right?!
See my other request about setting retract distance for G73 :)

I agree the Sinumerik code sounds better.
>
Having both chip breaking and full retracts combined is really really nice.

For small deep holes it's a great time saver and avoids the risk of
plunging back into the hole so many times.