LinuxCNC/linuxcnc #1790: G83: add option to plunge at feed rate

silopolis, 2022-07-03:
Specially with very long small diameter drills plunging back in hole at rapid speed is like playing Russian roulette!

Adding a word to switch to feed rate for plunging would be very nice for these cases

revjmoney, 2026-11-09:
I'd like to work on this. Two things I found while reading the current code, and a question before I write anything.

**1. The G83 docs don't describe the rapid back into the hole.** `convert_cycle_g83` in `src/emc/rs274ngc/interp_cycles.cc` (master) does, per peck:

```
cycle_feed(block, plane, x, y, current_depth);                    // feed down one peck
cycle_traverse(block, plane, x, y, r);                            // rapid out to R
cycle_traverse(block, plane, x, y, current_depth + rapid_delta);  // rapid back DOWN into the hole
```

The G83 section of `docs/src/gcode/g-code.adoc` says the third step is "Rapid up either the D value, the G83_PECK_CLEARANCE ...". It is a rapid back down to D above the previous peck bottom. That rapid back in is exactly what this issue is worried about, and the docs hide it. I can send a small docs PR for that on its own.

**2. Plunging back at feed may already be reachable with the D word from #2528, with one small fix.** With D at least the depth drilled so far, the re-entry point `current_depth + D` is at or above R, so the descent back into the hole is all feed. Today, though, a D larger than that makes the cycle rapid *up* past R first (to `current_depth + D`) and then feed down through air from there. Clamping the re-entry point to R:

```
cycle_traverse(block, plane, x, y, MIN(current_depth + rapid_delta, r));  // every plane checks r >= bottom, so MIN is right in all four
```

would make "large D" mean "feed all the way back in from R", with no new word. G73 has the same shape.

Would you take the clamp (plus a docs note on using D for this), or do you prefer a separate explicit option? I'll start on whichever you pick, docs fix first.

grandixximo, 2026-11-10:
Thanks, both findings check out.

The docs fix is welcome on its own: step 4 of the G83 section should say the tool rapids back down to D above the bottom of the previous peck, not "rapid up".

The clamp to R is fine as a separate small change, in G83 and G73: a re-entry point above R is never useful. I would not use it as the answer to this issue, though. It makes "D larger than the hole depth" a hidden mode, and feeding back in at the cutting F all the way from R is slow on a deep hole, which is exactly the case this issue is about.

For the option itself I'd prefer an explicit `I` word on G83: the feed rate for the move back into the hole, from R down to D above the previous bottom. No `I` keeps today's rapid. `I` is free there: `check_other_codes` in `interp_check.cc` rejects it on G83 today in every plane ("I word with no G2, G3, ... or G87 to use it"), so no existing program changes meaning. Things to watch:

- Add `G_83` to the motion list of that I check, nothing more.
- Read `I` per block like D (`block->i_flag`), not through `_setup.cycle_i`. That one is sticky and belongs to G87, so an earlier G87 would leak its I into G83.
- G68.2 and G68.4 also take I. A line carrying both G83 and G68.2/G68.4 should be an error rather than have both read the same I.
- `I` <= 0 is an error. Inverse time is already rejected in canned cycles, so `I` is always units per minute.
- The move back in ends at `MIN(current_depth + D, r)`, so the clamp still applies; the next peck stays at F as today.

revjmoney, 2026-11-10:
Thanks @grandixximo, that's a clear plan. I'll do it as three separate PRs, in this order:

1. Docs: fix step 4 of the G83 section (rapid back down to D above the previous peck bottom).
2. Clamp the re-entry point to R in G83 and G73.
3. The `I` word on G83, following your list: `G_83` added to the I check in `check_other_codes`, I read per block from `block->i_flag` (not `_setup.cycle_i`), G83 with G68.2/G68.4 on the same line is an error, I <= 0 is an error, and the move back in still ends at `MIN(current_depth + D, r)`. With docs and tests.