LinuxCNC/linuxcnc #1794: Issue with tool_probe_m6 in stdglue.py

brycedjohnson, 2022-07-03:
My first time using probing or tool touch off so I could be missing something.

## Here are the steps I follow to reproduce the issue:

Set this remap in your ini `REMAP=M6 modalgroup=10 python=tool_probe_m6`

Which points to the function in linuxcnc/nc_files/remap_lib/python-stdglue/stdglue.py

## This is what I expected to happen:
Successful tool length offset

## This is what happened instead:
```
        self.params[4999] = emccanon.GET_EXTERNAL_POSITION_X()
        self.params[4998] = emccanon.GET_EXTERNAL_POSITION_Y()
        self.params[4997] = emccanon.GET_EXTERNAL_POSITION_Z()
```
Appear to grab whatever coor system you are in (G54) in mycase.  But the move back happens in G53 and it tosses a bunch of limit errors.

[image]

Printing those params shows
[image]

Also haven't quite verified this but the G38.2 probe seems to be working in whatever coor system you are in as well.  

I am currently working around by using switching to zeroed out G59.3 `self.execute("G59.3")` before the GET_EXTERNAL_POSITION and using a G54 `self.execute("G54")` after the tool length is set.  Then I can: touch off probe to tool length sensor - zero probe to workpiece - switch to tool1 and probe to tool length sensor - Cutter is zeroed on top of workpiece

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`):

lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 10 (buster)
Release:	10
Codename:	buster

 * I am using this kernel version (shown by `uname -a`):

uname -a
Linux cncpc 4.19.0-20-rt-amd64 #1 SMP PREEMPT RT Debian 4.19.235-1 (2022-03-17) x86_64 GNU/Linux

 
 * I am running ...
   * [x ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):
 LINUXCNC - 2.9.0-pre0-6934-g45a17c155

 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):
 Qtdragon

 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): 
Ethernet Mesa

brycedjohnson, 2022-07-03:
Also a related PR #1793 for an apparent type-o

c-morley, 2022-07-06:
Unfortunately this was code I was working on and I got lost in other priorities.

brycedjohnson, 2022-07-06:
No problem @c-morley , it was a good starting place to figure out how all the probe stuff works.  I'll do some PRs if I do improvements besides my workaround.

c-morley, 2022-07-07:
Yes please do, I would like to get it right.