LinuxCNC/linuxcnc #1813: Don't flush queue when spindle speed is set but stays same

PetteriAimonen, 2022-07-09:
## Here are the steps I follow to reproduce the issue:

 1. Generate G-code using [OpenBuilds CAM](https://cam.openbuilds.com/)
 2. Set `G64` path blending mode
 3. Run the file and observe slow motion

The generated G-code will be like this:

    G1 F1000 X100.000 Y100.000 Z-2.000 S10000
    G1 F1000 X100.100 Y100.010 Z-2.000 S10000
    G1 F1000 X100.200 Y100.020 Z-2.000 S10000
    G1 F1000 X100.300 Y100.040 Z-2.000 S10000
    G1 F1000 X100.400 Y100.060 Z-2.000 S10000

## This is what I expected to happen:

Segments should be blended together instead of stopping many times per millimeter.

## This is what happened instead:

Machine feeds very slowly and `linuxcnctop` shows that queue length is always 0 or 1.

## Information about my hardware and software:

LinuxCNC 2.8.2.11.g6a3d0a434 using the official distribution, with AXIS gui.

## Cause of bug

This happens because OpenBuilds CAM puts `S10000` on every line, which rs274 then converts to `SET_SPINDLE_SPEED` command.  

Currently `SET_SPINDLE_SPEED` unconditionally calls `flush_segments()`:
https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/task/emccanon.cc#L1918-L1926

I suggest adding a comparison against currently set speed first, and skip the function if there is no change.
I can pull request the change if wanted.

c-morley, 2022-07-11:
pull request would be good please.

petterreinholdtsen, 2022-07-15:
Perhaps a good start could be to create a test case for this.  Not sure where it should go, but suspect tests/trajectory-planner/ might be a good place for it.  @c-morley, perhaps you can provide some guidance to where in the code such logic should be implemented?