LinuxCNC/linuxcnc #1872: pncconf: add 7i96s - dont work in 2.8

zz912, 2022-07-27:
Mr . phillc54  made this commit:
https://github.com/LinuxCNC/linuxcnc/commit/21ad34edb4f4d6c65308840bc3d88cd5a5d0175a

So I downloaded version 2.8 and build it.
Pncconf had 7i96s. It is good, but it dont work.

I do not know what went wrong. Did phillc54 only try to add 7i96s to 2.9 and it got into 2.8 by mistake?

There are two ways to do it so that 7i96s can be used even in version 2.8.
**The first way** is to port hostmot2 from version 2.9 to version 2.8.

**The second way** is to modify
/home/zdenek/linuxcnc/linuxcnc-2.8/src/emc/usr_intf/pncconf/private_data.py
so that the original 2.8 hostmot2 accepts it.

I tried it here:
https://forum.linuxcnc.org/27-driver-boards/46356-can-i-use-7i96s-with-lcnc-2-8

The result is a working 7i96s definition for hostmot2 2.8:
` # 7i96s ####################
    ['7i96s-Internal Data', '7i96', '7i96s_d', '7i96', 'hm2_eth',
        1,3, 0,0, 1,1, 0,0, 5,2, 1,1, [],0,0,0,0,0,0,0, 1, 34, 33, 200, [1, 2, 3],
        # TAB 1
        [S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],
        [S.GPIOI, 0],[S.SSR0, 100],[S.SSR0, 101],[S.SSR0, 102],[S.SSR0, 103],[S.GPIOO, 100],[S.GPIOO, 101],
        [S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],
        # TAB 2
        [S.STEPB,0],[S.STEPA,0],[S.STEPB,1],[S.STEPA,1],[S.STEPB,2],[S.STEPA,2],[S.STEPB,3],[S.STEPA,3],[S.STEPB,4],[S.STEPA,4],
        [S.ENCA,0],[S.ENCB,0],[S.ENCI,0],[S.RXDATA0,0],[S.TXDATA0,0],[S.TXEN0,0],[S.PWMP,0],
        [S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],
        # TAB 3
        [S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],
        [S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],[S.GPIOI, 0],
        [S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],[S.NUSED,0],],`

Just now I wanted to make a Pull request to add this definition because no one responded to my idea in the forum. I saw Phil's commit so I created this Issue.

I would be very happy if 7i96s were added to version 2.8.
I would prefer the first way: port hostmot2 from version 2.9 to version 2.8

Should you choose to go the other way. I can elaborate more on the differences between hosmot2 2.8 and 2.9.

hansu, 2022-07-27:
https://github.com/LinuxCNC/linuxcnc/commit/21ad34edb4f4d6c65308840bc3d88cd5a5d0175a is on 2.8 branch.

>So I downloaded version 2.8 and build it.

Did you download it from http://buildbot.linuxcnc.org/dists/buster/ ?
It seems like the [buildbot](http://buildbot.linuxcnc.org/buildbot/waterfall) is not finished yet and so the files not updated yet.
Maybe give it a try a bit later.

zz912, 2022-07-27:
I made this:
[image]

and then this:
> cd linuxcnc-2.8/src
> ./autogen.sh
> ./configure --with-realtime=uspace
> make
> sudo make setuid

hansu, 2022-07-27:
Did you run  `scripts/rip-environment`?

hansu, 2022-07-27:
>I think that this way I will get to the data from which the version 2.8.3 will be created in the future. It is so?

But you can also use a deb package from the buildbot for installing.

zz912, 2022-07-27:
> Did you run `scripts/rip-environment`?
Yes, I use it for run pncconf, halshow, halrun ......

Why are you ask?

hansu, 2022-07-27:
So I guess you did not?
Let me assume you did this:
```
cd linuxcnc-2.8/src
./autogen.sh
./configure --with-realtime=uspace
make
sudo make setuid
linuxcnc <path to your ini file>
```
This builds the correct version, but if you don't run the rip-environment script, you just start the (old) version which is installed on the system. You need to run this script to set up a run-in-place environment. See also http://linuxcnc.org/docs/2.8/html/code/building-linuxcnc.html#_building_for_run_in_place for more information.

zz912, 2022-07-27:
> > I think that this way I will get to the data from which the version 2.8.3 will be created in the future. It is so?
> 
> But you can also use a deb package from the buildbot for installing.

I know there are many ways to build an LCNC. But I don't know what you want to tell me by this? Is this related to the 7i96s issue in pncconf 2.8? I use a RIP install to have the latest data.

zz912, 2022-07-27:
> So I guess you did not? Let me assume you did this:
>
> ```
> cd linuxcnc-2.8/src
> ./autogen.sh
> ./configure --with-realtime=uspace
> make
> sudo make setuid
> linuxcnc <path to your ini file>
> ```
>
> This builds [...]

I made:
sudo apt-get remove linuxcnc
before I download source code from github.

phillc54, 2022-07-27:
I see you mention hostmot2, I am guessing that 2.8 hostmot2 does not support the 7i96s.
If that is the case then I will revert the 2.8 commit.
You should then be using master branch if that is the only branch that supports the 7i96s.

zz912, 2022-07-27:
Could you port hostmot2 from version 2.9 to version 2.8?

phillc54, 2022-07-27:
Sorry, I know nothing about the hostmot2 code.

zz912, 2022-07-27:
Thank you for your cooperation and I am closing this issue.

closed 2022-07-27 by zz912