LinuxCNC/linuxcnc #1873: Gmoccapy - [RS274NGC] - FEATURES

zz912, 2022-07-27:
I integrated probe for autotoolmeasurement:
I followed the instructions:
http://linuxcnc.org/docs/stable/html/gui/gmoccapy.html#gmoccapy:auto-tool-measurement

`[RS274NGC]`
`# Enables the reading of INI and HAL values from gcode`
`FEATURES=12`
``
`# is the sub, with is called when a error during tool change happens, not needed on every machine configuration`
`ON_ABORT_COMMAND=O <on_abort> call`
``
`# The remap code`
`REMAP=M6  modalgroup=6 prolog=change_prolog ngc=change epilog=change_epilog`

I wanted to see what the FEATURES parameter means.
I found:
http://linuxcnc.org/docs/stable/html/config/ini-config.html#gcode:ini-features
`[NOTE] The above six options were controlled by the FEATURES bitmask in versions of LinuxCNC prior to 2.8. This INI tag will no longer work.`

ZincBoy, 2022-08-09:
The FEATURES=12 line in the documentation should be replaced with:

INI_VARS = 1
HAL_PIN_VARS = 1

However, the default behavior is for both of those to be set to 1 (enabled) so they are redundant.  The FEATURES=12 can be completely removed without effect unless the INI has explicitly set INI_VARS or HAL_PIN_VARS to 0.

hansu, 2022-08-12:
See also here for more details: https://github.com/LinuxCNC/linuxcnc/pull/1915

hansu, 2022-08-18:
See updated docs here: http://linuxcnc.org/docs/html/config/ini-config.html

closed 2022-08-18 by hansu