LinuxCNC/linuxcnc #1893: Documents Fail to Build on Debian 11

jethornton, 2022-08-03:
I did a fresh install of Debian 11 as shown here
https://gnipsel.com/linuxcnc/debian-11.html

Then cloned and built a deb for LinuxCNC 2.9 (Master) as described here.
https://gnipsel.com/linuxcnc/debian-11-emc.html

If I do debian/configure then debuild -uc -us the build ends with many errors in the documents.

JT

jethornton, 2022-08-03:
I tried a different switch as suggested by Turboss debuild -b -uc this is the errors
[debian-11.txt](https://github.com/LinuxCNC/linuxcnc/files/9255424/debian-11.txt)

JT

silopolis, 2022-08-06:
So, it looks like it builds by manually installing po4a from testing and it's dependency

Only lintian seems not so happy

```
dpkg-deb: building package 'linuxcnc-doc-en' in '../linuxcnc-doc-en_2.9.0~pre0_all.deb'. 	dpkg-deb --build debian/linuxcnc-doc-es .. dpkg-deb: building package 'linuxcnc-doc-es' in '../linuxcnc-doc-es_2.9.0~pre0_all.deb'. dpkg-genbuildinfo dpkg-genchanges >../linuxcnc_2.9.0~pre0_amd64.changes dpkg-genchanges: info: including full source code in upload dpkg-source --after-build . dpkg-buildpackage: info: full upload; Debian-native package (full source is included) Now running lintian linuxcnc_2.9.0~pre0_amd64.changes ... E: linuxcnc source: duplicate-globbing-patterns 2 in lines 214 214 214 E: linuxcnc source: duplicate-globbing-patterns GPL in lines 214 214 214 E: linuxcnc source: duplicate-globbing-patterns License: in lines 214 214 214 E: linuxcnc source: duplicate-globbing-patterns Version in lines 214 214 214 E: linuxcnc source: duplicate-globbing-patterns src/emc/linuxcnc.h in lines 214 344 E: linuxcnc source: duplicate-globbing-patterns src/hal/hal_lib.c in lines 214 339 E: linuxcnc source: duplicate-globbing-patterns src/hal/hal_priv.h in lines 214 339 E: linuxcnc source: duplicate-globbing-patterns src/libnml/nml/* in lines 362 374 E: linuxcnc source: duplicate-globbing-patterns the in lines 384 384 E: linuxcnc-uspace: malformed-override Unknown tag elevated-privileges in line 8 E: linuxcnc-uspace: malformed-override Unknown tag elevated-privileges in line 9 W: linuxcnc source: debian-rules-uses-unnecessary-dh-argument dh ... --parallel (line 39) W: linuxcnc source: debian-watch-file-in-native-package W: linuxcnc-doc-es: desktop-command-not-in-package usr/share/applications/linuxcnc-documentation_es.desktop xdg-open W: linuxcnc-doc-es: desktop-command-not-in-package usr/share/applications/linuxcnc-gettingstarted_es.desktop xdg-open W: linuxcnc-doc-zh-cn: desktop-command-not-in-package usr/share/applications/linuxcnc-gettingstarted_zh_CN.desktop xdg-open W: linuxcnc-uspace: executable-not-elf-or-script usr/lib/python3/dist-packages/qtvcp/lib/mdi_text.py W: linuxcnc-uspace: executable-not-elf-or-script usr/lib/tcltk/linuxcnc/bin/cbutton.tcl W: linuxcnc-uspace: groff-message usr/share/man/man9/toggle.9.gz 19: can't open '../man/images/toggle.ps': No such file or directory W: linuxcnc-uspace: groff-message usr/share/man/man9/toggle2nist.9.gz 28: can't open '../man/images/toggle2nist.ps': No such file or directory W: linuxcnc-uspace: lacks-unversioned-link-to-shared-library usr/lib/libpyplugin.so.0 usr/lib/libpyplugin.so W: linuxcnc source: mismatched-override debian-rules-uses-unnecessary-dh-argument dh ... --parallel (line 32) W: linuxcnc-doc-es: mismatched-override desktop-command-not-in-package usr/share/applications/linuxcnc-documentation_es.desktop usr/bin/see W: linuxcnc-doc-es: mismatched-override desktop-command-not-in-package usr/share/applications/linuxcnc-gettingstarted_es.desktop usr/bin/see W: linuxcnc-doc-zh-cn: mismatched-override desktop-command-not-in-package usr/share/applications/linuxcnc-gettingstarted_zh_CN.desktop usr/bin/see W: linuxcnc-uspace: mismatched-override groff-message usr/share/man/man9/toggle.9.gz (line 1) 17: can't open '../man/images/toggle.ps': No such file or directory W: linuxcnc-uspace: mismatched-override groff-message usr/share/man/man9/toggle2nist.9.gz (line 1) 26: can't open '../man/images/toggle2nist.ps': No such file or directory W: linuxcnc source: missing-debian-source-format W: linuxcnc source: newer-standards-version 4.6.0 (current is 4.5.1) W: linuxcnc-uspace: no-manual-page usr/bin/melfagui W: linuxcnc-uspace: no-manual-page usr/bin/millturngui W: linuxcnc-uspace: no-manual-page usr/bin/qtplasmac_gcode W: linuxcnc-uspace: no-manual-page usr/bin/tool_mmap_read W: linuxcnc-uspace: no-manual-page usr/bin/tool_watch W: linuxcnc-uspace: setuid-binary usr/bin/linuxcnc_module_helper 4755 root/root W: linuxcnc-uspace: setuid-binary usr/bin/rtapi_app 4755 root/root W: linuxcnc-uspace: shared-library-lacks-prerequisites usr/lib/linuxcnc/modules/boss_plc.so W: linuxcnc-uspace: shared-library-lacks-prerequisites usr/lib/linuxcnc/modules/charge_pump.so W: linuxcnc-uspace: shared-library-lacks-prerequisites usr/lib/linuxcnc/modules/corexykins.so W: linuxcnc-uspace: shared-library-lacks-prerequisites usr/lib/linuxcnc/modules/counter.so W: linuxcnc-uspace: shared-library-lacks-prerequisites usr/lib/linuxcnc/modules/debounce.so W: linuxcnc-uspace: shared-library-lacks-prerequisites usr/lib/linuxcnc/modules/encoder.so W: linuxcnc-uspace: shared-library-lacks-prerequisites usr/lib/linuxcnc/modules/encoder_ratio.so W: linuxcnc-uspace: shared-library-lacks-prerequisites usr/lib/linuxcnc/modules/hal_gm.so W: linuxcnc-uspace: shared-library-lacks-prerequisites usr/lib/linuxcnc/modules/hm2_7i43.so W: linuxcnc-uspace: shared-library-lacks-prerequisites usr/lib/linuxcnc/modules/hm2_7i90.so W: linuxcnc-uspace: shared-library-lacks-prerequisites usr/lib/linuxcnc/modules/hm2_pci.so W: linuxcnc-uspace: shared-library-lacks-prerequisites usr/lib/linuxcnc/modules/hm2_test.so W: linuxcnc-uspace: shared-library-lacks-prerequisites usr/lib/linuxcnc/modules/homecomp.so W: linuxcnc-uspace: shared-library-lacks-prerequisites usr/lib/linuxcnc/modules/homemod.so W: linuxcnc-uspace: shared-library-lacks-prerequisites usr/lib/linuxcnc/modules/matrix_kb.so W: linuxcnc-uspace: shared-library-lacks-prerequisites usr/lib/linuxcnc/modules/mesa_7i65.so W: linuxcnc- [... cut ...] b/linuxcnc/modules/threads.so W: linuxcnc-uspace: shared-library-lacks-prerequisites usr/lib/linuxcnc/modules/tpcomp.so W: linuxcnc-uspace: shared-library-lacks-prerequisites usr/lib/linuxcnc/modules/watchdog.so W: linuxcnc-uspace: shared-library-lacks-prerequisites usr/lib/linuxcnc/modules/weighted_sum.so W: linuxcnc source: upstream-metadata-in-native-source N: 14 hints overridden (1 error, 13 warnings); 17 unused overrides Finished running lintian.
```

What do our packagers think about this?

petterreinholdtsen, 2022-08-10:
It is unclear to me if the issue here is that the package building fail, or if lintian find some issues.  Can you explain a bit more exactly what the problem is?

silopolis, 2022-08-11:
Le jeu. 11 août 2022 à 00:27, petterreinholdtsen ***@***.***>
a écrit :

> It is unclear to me if the issue here is that the package building fail,
> or if lintian find some issues. Can you explain a bit more exactly what the
> problem is?
>

Originally the packages were not building.
This was solved by installing po4a 0.67 package from testing in Bullseye.
Only remained lintian issues.

petterreinholdtsen, 2022-08-11:
[Jérémie Tarot]
> Originally the packages were not building.
> This was solved by installing po4a 0.67 package from testing in Bullseye.
> Only remained lintian issues.

Ah.  Then perhaps this issue should be closed?

-- 
Happy hacking
Petter Reinholdtsen

silopolis, 2022-08-12:
Le jeu. 11 août 2022 à 19:45, petterreinholdtsen ***@***.***>
a écrit :

> [Jérémie Tarot]
> > Originally the packages were not building.
> > This was solved by installing po4a 0.67 package from testing in Bullseye.
> > Only remained lintian issues.
>
> Ah. Then perhaps this issue should be closed?

Yep, left it open just to leave opportunity for packagers to take note of
lintian messages if necessary

petterreinholdtsen, 2022-08-12:
[Jérémie Tarot]
> Yep, left it open just to leave opportunity for packagers to take note
> of lintian messages if necessary

When I build using 'debuild', I get the messages at the end of the
build, and I also check out the buildpot output for lintian issues.

-- 
Happy hacking
Petter Reinholdtsen

silopolis, 2022-08-12:
Le ven. 12 août 2022 à 13:46, petterreinholdtsen ***@***.***>
a écrit :

> [Jérémie Tarot]
> > Yep, left it open just to leave opportunity for packagers to take note
> > of lintian messages if necessary
>
> When I build using 'debuild', I get the messages at the end of the
> build, and I also check out the buildpot output for lintian issues.
>

Master at work 👌😎

Message ID: ***@***.***>
>