LinuxCNC/linuxcnc #1914: Reorder documentation

hansu, 2022-08-12:
I am thinking of reordering the sections of the documentation into a more logical order. And maybe add cross headings.
I have something like this in mind. What do you think of that?

#### Getting Started & Configuration

- General User Information
- Getting Started with LinuxCNC
- Configuration Wizards
- Configuration
- HAL (Hardware Abstraction Layer)
- Hardware Drivers
- Hardware Examples
- Classicladder
- Advanced Topics
- Integrator Information

#### Usage

- User Interfaces
- G-code Programming

#### Customization & Development

- Virtual Control Panels
- User Interface Programming
- Developer Information
- Glossary, Copyright, History & Overview

c-morley, 2022-08-12:
seems reasonably to me.
Integrator  info seems like it could use a different title 'integrator whitpages' - maybe that's to obscure.

Chris

smoe, 2022-08-14:
Also liking it. Modbus is currently explained in the context of ClassicLadder - should that possibly move to a more general part?

silopolis, 2022-08-14:
Le dim. 14 août 2022 à 19:09, Steffen Möller ***@***.***> a
écrit :

> Also liking it. Modbus is currently explained in the context of
> ClassicLadder - should that possibly move to a more general part?
>

Modbus should have it's chapter in a "Fielbuses" or such of the Integrator
guide.

ClassicLadder should have it's own "PLC Programming" part in the same
guide, waiting to be doubled by a sibling OpenPLC subpart if we someday
integrate with the latter. Modbus there should only deal with
implementation with said logic controllers.

hansu, 2022-08-17:
>Integrator  info seems like it could use a different title 'integrator whitpages' - maybe that's to obscure.

Not sure if that is better.

hansu, 2022-08-18:
Looks now so: http://buildbot.linuxcnc.org/doc/scratch/v2.9.0~pre0~1914-reorder-documentation~2d7001abe/html/
(I don't know why the images are missing on the build bot)

The Integrator Information could be moved to Configuration:
Maybe an own section for steppers would make sense.
[image]

closed 2022-09-09 by hansu

silopolis, 2022-10-11:
Hello friends

Le ven. 12 août 2022 à 09:52, Hans ***@***.***> a écrit :

> I am thinking of reordering the sections of the documentation into a more
> logical order. And maybe add cross headings.
> I have something like this in mind. What do you think of that?
> Getting Started & Configuration
>
>    - ...
>
> Usage
>
>    - ...
>
> Customization & Development
>
>    - ...
>
>
I think we had already started discussing this in one of our long threads
in the beginnings or before po4a work started. I would have to dig into
mailbox...

Anyway I've always thought, along the line traced by bigger players, that
documentation should be:
* Split between users documentation and reference manuals, the latter
holding G-code ref, HAL ref, INI ref, code ref...
Maybe those could (should?) be auto-generated with adequate sources
documentation. Same as we do with man pages, which I believe also belong to
the reference content category.
* Users Manuals should be targeted at archetypal use cases, personnas for
the Agile crowd, being:
** NC  user/operator: those making chips with lcnc
** NC integrators: those buiding, retrofitting and maintaining lcnc powered
machines
** NC developers and contributors: those writing code, tests, docs... As
well as translators.

Depending on the amount of potentially shared and/or overlapping content,
and to avoid a cross referencing maze for the readers, a common matters
volume could be bundled and added to the list.

When a user/operator content for a specific machine type (mill, lathe,
plasma...) becomes thick enough, it would in turn be split into its own
opus... If we don't decide to separate them from the start.

I believe this structure would build an overall clear and easy to
understand overall table of content, as well as a series of targeted
actionable documents of an hopefully non daunting and reasonably printable
size for each specific kind of user.

Icing on the cake, this slicing could open opportunities to alleviate
weight of docs on the build time by allowing targeted rebuilds of edited
doc(s)... Provided another round of significant work on the build system.
Do I ear CMake?! 😉

Message ID: ***@***.***>
>

PR #1972 (2022-09-09): Reorder documentation index
Like mentioned in #1914.
Now also adapted the PDF version.
See http://buildbot.linuxcnc.org/doc/scratch/v2.9.0~pre0~1914-reorder-documentation~da5be815b/ for a preview.

Closes #1914