LinuxCNC/linuxcnc #1924: Should libnml become a separate LinuxCNC project?

smoe, 2022-08-14:
I feel that libnml is a gem of LinuxCNC that is not finding the recognition it deserves. Just as a hunch, I could for instance imagine that BOINC would want to use that, too. Would it be well-perceived in the community if libnml moves into https://github.com/LinuxCNC/libnml?

Disadvantage:
 * We would need to wait a few months until this is an official package in Debian and other distributions.

Advantages:
 * Additional users of LinuxCNC infrastructure with accompanying eyeballs / developments, or pointers to other technology that would provide an alternative implementation that we should then discuss to adopt if that is superior.

Supporting (if this works out as I hope):
 * Separation of work among LinuxCNC devs
 * Separate (and likely more intense) testing
 * Additional documentation

silopolis, 2022-08-14:
Le dim. 14 août 2022 à 19:17, Steffen Möller ***@***.***> a
écrit :

> I feel that libnml is a gem of LinuxCNC that is not finding the
> recognition it deserves. Just as a hunch, I could for instance imagine that
> BOINC would want to use that, too. Would it be well-perceived in the
> community if libnml moves into https://github.com/LinuxCNC/libnml?
>

I'd support this as I would for HAL separation 👍🙏
This fits perfectly the meta-NC nature of the project.

smoe, 2022-08-14:
I guess that the community would not be ready yet for a separation of HAL. Something more low-level-ish like libnml may be a good exercise to learn how it goes. If there are no adopters of that technology then we can just leave it all as it is. Just, we already know that HAL has adopters that are external to LinuxCNC. This effectively means that we also have external adopters of libnml, but we have not singled that library out, yet.

silopolis, 2022-08-14:
Le dim. 14 août 2022 à 20:22, Steffen Möller ***@***.***> a
écrit :

> I guess that the community would not be ready yet for a separation of HAL.
>
Not as a whole, but MachinKit having already done it means some of us are,
or at least have been!

smoe, 2022-08-15:
@hansu , could you please put this on the agend for the next Open Mike session?

hansu, 2022-08-15:
>@hansu , could you please put this on the agend for the next Open Mike session?

:heavy_check_mark:

petterreinholdtsen, 2022-08-15:
Personally, I believe splitting up the code base should only be done on
request by someone already using the particular fraction of linuxcnc
today.  Is there such request?  To me, splitting out the library is way
down on the priority list, below integrating new components added to
machinekit but not linuxcnc, below making it reproducible, below regular
live ISO build, automated GUI testing and a lot of other development
work that need to take place.

This is not to say that those that want to spend time on the  split
should not do so, but that they should have a clear idea of the
advantage of it and be willing to spend time on maintaining such
separate git repo.

-- 
Happy hacking
Petter Reinholdtsen

smoe, 2022-08-15:
I disagree, obviously. We should find if any of the points you mentioned do somehow benefit from LinuxCNC being external to the  source tree. From how I perceive libnml, this is mostly invariant and the oldest code base in the project: It is already separated, just not formally so.

c-morley, 2022-08-19:
There was actually a lot of talk (at one time) to replace NML. 
If your serious about putting in work to separate it, then maybe discussing replacing it would be pertinent.

petterreinholdtsen, 2022-08-20:
[c-morley]
> There was actually a lot of talk (at one time) to replace NML.

What could it be replaced with?  Where was this discussed?

-- 
Happy hacking
Petter Reinholdtsen

jepler, 2022-08-20:
it was a goal of machinekit to replace nml with something based on some other tech, zeromq. Doing a local search of my personal maiing list archive, it looks like "zeromq" as a search term brings up a number of vintage posts.

silopolis, 2022-08-20:
Le sam. 20 août 2022 à 16:35, Jeff Epler ***@***.***> a
écrit :

> it was a goal of machinekit to replace nml with something based on some
> other tech, zeromq. Doing a local search of my personal maiing list
> archive, it looks like "zeromq" as a search term brings up a number of
> vintage posts.
>

Interesting, thanks 👍

c-morley, 2022-08-20:
> it was a goal of machinekit to replace nml with something based on some other tech, zeromq. Doing a local search of my personal maiing list archive, it looks like "zeromq" as a search term brings up a number of vintage posts.

Didn't you and Seb start on something similar at a fest once too?

jepler, 2022-08-20:
The thing Seb & I started was a "plain C" API to LinuxCNC, but it didn't necessarily get rid of libnml .. it was about the problem that the "most native API" was C++-only and that's a bad fit for exposing to scripting languages as well (today) to trendy young languages like rust. It didn't get very far.

smoe, 2022-08-20:
I found https://stackshare.io/stackups/mqtt-vs-zeromq helpful. I did not know that there was a motivation to see libnml completely removed.

andypugh, 2022-08-21:
On Sun, 21 Aug 2022 at 00:50, Steffen Möller ***@***.***>
wrote:

> I found https://stackshare.io/stackups/mqtt-vs-zeromq helpful. I did not
> know that there was a motivation to see libnml completely removed.
>

See:
https://sourceforge.net/p/emc/mailman/emc-developers/thread/1BC4D539-F8CE-42DB-9AD5-6AE2F5986D5E%40mah.priv.at/#msg29646911

> Message ID: ***@***.***>
>

-- 
atp

silopolis, 2022-08-22:
Le lun. 22 août 2022 à 01:56, andypugh ***@***.***> a écrit :

> On Sun, 21 Aug 2022 at 00:50, Steffen Möller ***@***.***>
> wrote:
>
>
> See:
>
> https://sourceforge.net/p/emc/mailman/emc-developers/thread/1BC4D539-F8CE-42DB-9AD5-6AE2F5986D5E%40mah.priv.at/#msg29646911

So much interesting stuff there! 😲🤩👍

silopolis, 2022-08-22:
Le lun. 22 août 2022 à 03:48, Jérémie Tarot ***@***.***> a écrit :

>
>
> Le lun. 22 août 2022 à 01:56, andypugh ***@***.***> a
> écrit :
>

See:
>>
>> https://sourceforge.net/p/emc/mailman/emc-developers/thread/1BC4D539-F8CE-42DB-9AD5-6AE2F5986D5E%40mah.priv.at/#msg29646911
>
>

Definitely a lot of things squeeze out of this thread after reading the
second half!

It's also somehow good to see that components independency and
distributability (ie network and sync), programming interfaces upgrade and
standardization (zmq, and orocos IIUC), and other architectural choices are
shared vision by many community members and for a long time...

Clever folks in this discussion raised a lot of technical and planning
points I'll gather in some form of document...

A significant part of tve messages also deal with licensing issues.
Specially, a potential legal incompatibility with ZeroMQ GPLv3 is raised,
notably due to parts of LinuxCNC being GPLv2 only.
Work on upgrading this situation to get all code GPLv2+ is said to be
needed... Has this been done? What's our current licences portfolio?

PS: as my tech understanding and prog skills are still both in their early
infancy, please correct any error or misunderstanding. Thanks in advance 🙏

silopolis, 2022-10-11:
Le lun. 15 août 2022 à 13:41, Hans ***@***.***> a écrit :

> @hansu <https://github.com/hansu> , could you please put this on the
> agend for the next Open Mike session?
>
> ✔️
>

Could we use:

* issues for meetings planning and preparation
* gh wiki for minutes

Would also allow linking in projects if we come to put them to use

silopolis, 2022-10-11:
Le sam. 20 août 2022 à 10:42, petterreinholdtsen ***@***.***>
a écrit :

> [c-morley]
> > There was actually a lot of talk (at one time) to replace NML.
>
> What could it be replaced with? Where was this discussed?

I'm curious too!