LinuxCNC/linuxcnc #1927: LinuxCNC 2.8.3

zz912, 2022-08-14:
I looked at website:
[image]

It is not true:
[image]

I would like to recap the situation with the 7i96s card.

1) PCW made a 7i96s card for LCNC version 2.9
2) Mr. phillc54 made commit for LCNC 2.8 + 7i96s
It was a mistake because he didn't know that LCNC 2.8 had the old hostmot2
https://github.com/LinuxCNC/linuxcnc/issues/1872
3) Mr. phillc54 uncommit for LCNC 2.8 + 7i96s
4) I didn't like that the 7i96s card is only for non-stable LCNC 2.9 and I made an imperfect configuration for LCNC 2.8 + 7i96s
https://github.com/LinuxCNC/linuxcnc/pull/1897
5) At the same time, a commit was created to update hostmot2
https://github.com/LinuxCNC/linuxcnc/pull/1899
6) If hostmot2 is updated, my imperfect configuration for LCNC 2.8 + 7i96s will disappear and Mr . phillc54 restores what he undid.

silopolis, 2022-09-13:
Sorry missed it 😬
May be better directed to LinuxCNC/wlo repository.

zz912, 2022-09-13:
What is wlo? What wlo mean?

silopolis, 2022-09-13:
Le mar. 13 sept. 2022 à 19:27, zz912 ***@***.***> a écrit :

> What is wlo? What wlo mean?
>

https://github.com/LinuxCNC/wlo

Www.Linuxcnc.Org

zz912, 2022-09-14:
Thank you. This is exactly what i wanted.

closed 2022-09-14 by zz912