LinuxCNC/linuxcnc #1929: Documentation: How to sep numbers from units

smoe, 2022-08-15:
This is a side-issue, I know. Please forgive.

We want units with all our values, but not directly adjacent to the number. A separation that is too wide is unfortunate, and no line break should be allowed. 

On https://mailtrap.io/blog/nbsp/#:~:text=is%20actually%20one%20of,and%20put%20into%20separate%20lines I learned about &amp;thisp; which looks quite right to me as in '&thinsp;' or 2&thinsp;mm. &amp;hairsp; as in '&hairsp;' or 2&hairsp;mm here in Github looks wider than in my atom asciidoc preview. Here I find that ok, too. As a referenc: This '' or 2mm is with no separation.

What I find a bit weird is that with electrical units like Ampère or Volt the distance I would want to be smaller than for length units. Does that make any sense to anyone? I suggest to start with &amp;thinsp; and if any decision is ever made to change that to something else, this would easily be done in some automated manner. @hansu , @silopolis, @petterreinholdtsen ?

silopolis, 2022-08-15:
I believe there must be real typographic rules for this 🤔

smoe, 2022-08-15:
I could not find any such formal document. Closest was https://forum.wordreference.com/threads/typography-rules-for-units-of-measurements.2781694 that claimed that there is no such. I chuckled at their claim that only non-techies would introduce blanks since they would have been told to do so.

hansu, 2022-08-18:
In [The International System of Units](https://www.bipm.org/documents/20126/41483022/SI-Brochure-9.pdf/fcf090b2-04e6-88cc-1149-c3e029ad8232) the width of this space seems to be not mentioned. Only that there should be a space.
So its up to the typographer or to the specification of the regarding organisation.
So it's up to us how we want it I guess :slightly_smiling_face: 

See also 
https://tex.stackexchange.com/questions/127817/how-much-space-between-number-and-unit/324137#324137
https://english.stackexchange.com/questions/105553/should-thin-spaces-be-used-between-numerals-and-units

petterreinholdtsen, 2022-10-11:
[Steffen Möller]
> This is a side-issue, I know. Please forgive.
>
> We want units with all our values, but not directly adjacent to the
> number. A separation that is too wide is unfortunate, and no line
> break should be allowed.

I would go with NO-BREAK SPACE (U+00A0), as it easier accessible on
keyboards and charsets than all the other space variations.

We need to make sure a2x, asciidoc, pandoc and dblatex all handle the
character used to ensure the HTML and PDF editions are correct.

Perhaps you can prepare a paragraph for the adoc style guide about this?
-- 
Happy hacking
Petter Reinholdtsen