LinuxCNC/linuxcnc #1930: Package menu entry errors

phillc54, 2022-08-15:
Menu in package installation of 2.9 from the buildbot:
Missing menu entries for PnCconf and StepConf.
Latency Test appears under Documentation.

SebKuzminsky, 2022-08-17:
This is partially fixed in master now.  As of 68680aa8 i see the following behavior:

# With just `linuxcnc-uspace` installed
Everything shows up in `Accessories`, but should not:
[image]

Everything shows up in `CNC` too, which is good, but this should be the only place.  And `latency-test` is in `CNC/Documentation`, should be in `CNC`:
[image]

# With all the docs installed
Again everything is in `Accessories`, where it doesn't belong.
[image]

The docs all show up in `CNC`, but some are in `CNC/Documentation` and some in `CNC/G-Code Quick-Reference`
[image]
[image]

phillc54, 2022-08-18:
Similar in Buster with 6c25f7eec, there is a slight difference with G-Code Quick Reference
[image]
[image]

petterreinholdtsen, 2022-08-18:
Where a menu entry show up in the menu structure is controlled using the
Categories values in the .desktop file.

Perhaps these values need a cleanup?

-- 
Happy hacking
Petter Reinholdtsen

commit bece5d38 (2022-08-21): Clean up Linux XDG menu entry information

Make sure all doc entries use category X-CNC-DOC and not category Utility.
Make sure all entries use Keywords and ensure all entries use the same
value ordering.

Removed obsolete menu entries and ensure es, fr and zh_CN have the same
set of documentation related menu entries.

Fixes #1930.
closed 2022-08-21 by petterreinholdtsen