LinuxCNC/linuxcnc #1933: Gmoccapy not starting "out of the box" on Linux (or WSL)

smoe, 2022-08-16:
Sidenote: I need to thank @hansu to also check WSL. That cannot be used for anything time-critical because of the latencies (330&hairsp;µs for base -> 980 while typing this report, 184&hairsp;µs for servo -> 566) on a 10 yo i7 and Win10, but we should find out where its limits are. The one or other educational value this should well have.

After installation on Debian (regular Linux machine or WSL) the direct execution of /usr/bin/gmoccapy demands first of an environment variable CONFIG_DIR and once that is given, also one for LINUXCNC_TCL_DIR, so I tried
```
LINUXCNC_TCL_DIR=/usr/lib/tcltk/linuxcnc CONFIG_DIR=/usr/share/gmoccapy gmoccapy
```
which led to another error:
```
Traceback (most recent call last):
  File "/usr/bin/gmoccapy", line 5365, in <module>
    app = gmoccapy(sys.argv)
  File "/usr/bin/gmoccapy", line 172, in __init__
    self.stat.poll()
SystemError: <method 'poll' of 'linuxcnc.stat' objects> returned NULL without setting an exception
```
This is a bit cryptic. Preferably, some default paths would have been chosen and the user would just start the binary.

smoe, 2022-08-16:
Well. I should first have executed linuxcnc. But that is what gmoccapy could have told me, too. The gmoccapy screen starts up (with errors) with the 3 axes mill simulation, but not without a first error box because of
```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gladevcp/iconview.py", line 118, in __init__
    pixbuf = Gtk.IconTheme.get_default().load_icon("gtk-home", 48, 0)
gi.repository.GLib.GError: gtk-icon-theme-error-quark: Icon 'gtk-home' not present in theme Adwaita (0)
```
that I could click away, but then the later file icon error kills it. A copy of stdout/err is added below.
```
moeller@DESKTOP-CPG6JS1:~$ linuxcnc
LINUXCNC - 2.9.0~pre0
Machine configuration directory is '/home/moeller/linuxcnc/configs/sim.gmoccapy'
Machine configuration file is 'gmoccapy.ini'

check_config:
  Unexpected multiple values [JOINT_0]HOME: 0.000 10
  Unexpected multiple values [JOINT_1]HOME: 0.000 10
  Unexpected multiple values [JOINT_2]HOME: 0.0 -10

Starting LinuxCNC...
linuxcnc TPMOD=tpmod HOMEMOD=homemod EMCMOT=motmod
Note: Using POSIX non-realtime
Found file(REL): ./core_sim.hal
Found file(REL): ./spindle_sim.hal
Found file(REL): ./simulated_home.hal
link (updating variable file): No such file or directory
/usr/lib/python3/dist-packages/gladevcp/hal_lightbutton.py:20: PyGIWarning: PangoCairo was imported without specifying a version first. Use gi.require_version('PangoCairo', '1.0') before import to ensure that the right version gets loaded.
  from gi.repository import Pango, PangoCairo
NVD3D10: CPU cyclestats are disabled on client virtualization
NVD3D10: CPU cyclestats are disabled on client virtualization
Found an error!
The following information may be useful in troubleshooting:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gladevcp/iconview.py", line 118, in __init__
    pixbuf = Gtk.IconTheme.get_default().load_icon("gtk-home", 48, 0)
gi.repository.GLib.GError: gtk-icon-theme-error-quark: Icon 'gtk-home' not present in theme Adwaita (0)

'IconFileSelection' object has no attribute 'store'
'IconFileSelection' object has no attribute 'iconView'
'IconFileSelection' object has no attribute 'store'
'IconFileSelection' object has no attribute 'iconView'
'IconFileSelection' object has no attribute 'on_btn_jump_to'
'IconFileSelection' object has no attribute 'store'
'IconFileSelection' object has no attribute 'iconView'
'IconFileSelection' object has no attribute 'store'
'IconFileSelection' object has no attribute 'iconView'
0  =  /usr/bin/gmoccapy
1  =  -ini
2  =  /home/moeller/linuxcnc/configs/sim.gmoccapy/gmoccapy.ini
Entry = trivkins
Entry = coordinates=xyz
found the following coordinates xyz

**** GMOCCAPY GETINIINFO ****
Number of joints = 3
3 COORDINATES found = xyz
joint 0 = axis x
joint 1 = axis y
joint 2 = axis z
{0: 'x', 1: 'y', 2: 'z'}

**** GMOCCAPY GETINIINFO ****

[KINS] KINESTYPE is trivkins
**** GMOCCAPY GETINIINFO ****
No MIN_ANGULAR_VELOCITY entry found in [DISPLAY] of INI file
Using default value of 0.1 degree / min
**** GMOCCAPY GETINIINFO ****
No DEFAULT_ANGULAR_VELOCITY entry found in [DISPLAY] of INI file
[... 136 lines cut ...]
    self.widgets.IconFileSelection1.show_filelabel(False)
  File "/usr/lib/python3/dist-packages/gladevcp/iconview.py", line 404, in show_filelabel
    self.file_label.hide()
AttributeError: 'IconFileSelection' object has no attribute 'file_label'

Jog Pin Changed
x+
Jog Button released = x+
Jog Pin Changed
x-
Jog Button released = x-
Jog Pin Changed
y+
Jog Button released = y+
Jog Pin Changed
y-
Jog Button released = y-
Jog Pin Changed
z+
Jog Button released = z+
Jog Pin Changed
z-
Jog Button released = z-
on tool change
```

hansu, 2022-08-16:
>gi.repository.GLib.GError: gtk-icon-theme-error-quark: Icon 'gtk-home' not present in theme Adwaita (0)

Oh, I thought was was fixed with https://github.com/LinuxCNC/linuxcnc/pull/1548

smoe, 2022-08-16:
Ah. Maybe it is. I was using the Debian package that may just not have been updated. I then suggest I'll have another look when https://github.com/LinuxCNC/linuxcnc/pull/1887 is out.

hansu, 2022-08-16:
A version before April 25th?

SebKuzminsky, 2022-08-17:
Yes, as you discovered, LinuxCNC GUIs expect to be started by the `linuxcnc` launcher script, not by a human "by hand" from the terminal or similar.

Can you check if this problem still exists in the current tip of master, and close the issue if not?

smoe, 2022-08-18:
Traveling, will check once I am back. /usr/bin is not the correct location for anything that usr is not meant to execute. Should we move all the GUIs to /usr/lib/linuxcnc/bin?

hansu, 2023-01-14:
@smoe have you checked that? I guess you are now back from travelling ;-)

smoe, 2023-01-14:
I do not see that error when now starting GMOCCAPY correcly on Linux, did not yet run WSL again - thank you for the ping.

closed 2023-01-14 by smoe