LinuxCNC/linuxcnc #1941: Failed to compile v2.9.0-pre0-7465-gc8dd11e89e deb packages

ihabmmali, 2022-08-19:
# The issue tracker is not a support forum

## Here are the steps I follow to reproduce the issue:
 1. cd [Linuxcnc source directory]
 2. git pull origin master
 3.  dpkg-buildpackage -b -uc
 

## This is what I expected to happen:
successful compilation and generation of deb packages

## This is what happened instead:

Several warning regarding missing files 
dh_install: warning: Cannot find (any matches for) "usr/share/man/man1/halstreamer.1" (tried in ., debian/tmp)

dh_install: warning: linuxcnc-uspace missing files: usr/share/man/man1/halstreamer.1
dh_install: warning: Cannot find (any matches for) "usr/share/man/man1/hy_gt_vfd.1" (tried in ., debian/tmp)

dh_install: warning: linuxcnc-uspace missing files: usr/share/man/man1/hy_gt_vfd.1
dh_install: warning: Cannot find (any matches for) "usr/share/man/man1/linuxcnc.1" (tried in ., debian/tmp)

dh_install: warning: linuxcnc-uspace missing files: usr/share/man/man1/linuxcnc.1
dh_install: warning: Cannot find (any matches for) "usr/share/man/man1/pi500_vfd.1" (tried in ., debian/tmp)

dh_install: warning: linuxcnc-uspace missing files: usr/share/man/man1/pi500_vfd.1
dh_install: warning: Cannot find (any matches for) "usr/share/man/man1/sendkeys.1" (tried in ., debian/tmp)

dh_install: warning: linuxcnc-uspace missing files: usr/share/man/man1/sendkeys.1
dh_install: warning: Cannot find (any matches for) "usr/share/man/man1/svd-ps_vfd.1" (tried in ., debian/tmp)

dh_install: warning: linuxcnc-uspace missing files: usr/share/man/man1/svd-ps_vfd.1
dh_install: warning: Cannot find (any matches for) "usr/share/man/man1/thermistor.1" (tried in ., debian/tmp)

dh_install: warning: linuxcnc-uspace missing files: usr/share/man/man1/thermistor.1
dh_install: warning: Cannot find (any matches for) "usr/share/man/man1/wj200_vfd.1" (tried in ., debian/tmp)

dh_install: warning: linuxcnc-uspace missing files: usr/share/man/man1/wj200_vfd.1
dh_install: warning: Cannot find (any matches for) "usr/share/man/man1/xhc-whb04b-6.1" (tried in ., debian/tmp)

dh_install: warning: linuxcnc-uspace missing files: usr/share/man/man1/xhc-whb04b-6.1

Followed By:

dh_install: error: missing files, aborting
make: *** [debian/rules:38: binary] Error 255

## It worked properly before this:
Recent code pull on 8/19/2022

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`): Debian 11
 * I am using this kernel version (shown by `uname -a`): 5.18.0-2-rt-amd64
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [X ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):v2.9.0-pre0-7465-gc8dd11e89e

 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):qtdragon
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): Mesa cards

petterreinholdtsen, 2022-08-21:
I suspect this issue is fixed by pull request #1942.

-- 
Happy hacking
Petter Reinholdtsen

SebKuzminsky, 2022-08-21:
I believe it was commits 46855065243f0826c6bffdaa9098d35f9010c2b6 and 1ab6d720ba5990eb8517d60499ead90a0f64f537 that introduced this build failure (but only on distros newer than the ones our CI tests, i.e. where `dh_missing` can fail the build instead of just warn). 

But yes, #1942 should fix it.

annhan, 2022-08-23:
- I pulled #1942 but the error is stilling

petterreinholdtsen, 2022-08-23:
[NhanPham]
> - I pulled #1942 but the error is stilling

#1942 is in master now.  Do you get the same build error if you use
debuild from the devscripts package, without any command line options?
This tool do more checking, and might give a better indication on what
is wrong.

My test build of master work, not sure what is special about your setup.

-- 
Happy hacking
Petter Reinholdtsen

annhan, 2022-08-23:
> [NhanPham]
> - I pulled #1942 but the error is stilling
> #1942 is in master now. Do you get the same build error if you use debuild from the devscripts package, without any command line options? This [...]

i built in master brand now. Just type:

```
git pull
cd debian
./configure no-docs uspace
cd ..
dpkg-buildpackage -b -uc
```

and got same above error

petterreinholdtsen, 2022-08-23:
[NhanPham]
> i built in master brand now. Just type:
>
> ```
> git pull
> cd debian
> ./configure no-docs uspace
> cd ..
> dpkg-buildpackage -b -uc
> ```

I tested a few variations, and you can drop 'uspace' from the list
above.  The 'no-docs' option break the build.  No idea why, but it is
probably a bug in the recent build system adjustments.

If you want to build before it is fixed, I recommend 'debian/configure
&& debuild' to get installable .deb files.

-- 
Happy hacking
Petter Reinholdtsen

ihabmmali, 2022-08-23:
The latest build system from the current pull also introduces these dependencies: asciidoc source-highlight asciidoc-dblatex

These appear to be documentation build related but the system now seems to ignore the "no-docs" option @SebKuzminsky

annhan, 2022-08-24:
still error. i got 

```
dh_install: linuxcnc-uspace missing files: usr/lib/python3
```

petterreinholdtsen, 2022-08-24:
[Eyecon]
> These appear to be documentation build related but the system now
> seems to ignore the "no-docs" option @SebKuzminsky

It is not ignored, but the no-docs build need some of the tools earlier
only used by the docs build.

Missing build dependencies would be detected by debuild, which is one of
the reasons I am more interested in test builds using that tool than
ones using dpkg-buildpackage.
-- 
Happy hacking
Petter Reinholdtsen

annhan, 2022-08-24:
> [NhanPham]
> i built in master brand now. Just type: ``` git pull cd debian ./configure no-docs uspace cd .. dpkg-buildpackage -b -uc ```
> I tested a few variations, and you can drop 'uspace' from the [...]

i typed ```debian/configure && debuild``` and still get the error:

```
dh_install: Cannot find (any matches for) "usr/lib/python3" (tried in ., debian/tmp)

dh_install: linuxcnc-uspace missing files: usr/lib/python3
	install -d debian/.debhelper/generated/linuxcnc-uspace-dev
	install -d debian/linuxcnc-uspace/
	cp --reflink=auto -a debian/tmp/etc debian/linuxcnc-uspace//
	install -d debian/linuxcnc-uspace//usr/bin
	cp --reflink=auto -a debian/tmp/usr/bin/5axisgui debian/tmp/usr/bin/axis debian/tmp/usr/bin/axis-remote debian/tmp/usr/bin/classicladder debian/tmp/usr/bin/debuglevel debian/tmp/usr/bin/elbpcom debian/tmp/usr/bin/genserkins debian/tmp/usr/bin/gladevcp debian/tmp/usr/bin/gladevcp_demo debian/tmp/usr/bin/gmoccapy debian/tmp/usr/bin/gremlin_view debian/tmp/usr/bin/gs2_vfd debian/tmp/usr/bin/gscreen debian/tmp/usr/bin/hal-histogram debian/tmp/usr/bin/hal_input debian/tmp/usr/bin/hal_manualtoolchange debian/tmp/usr/bin/halcmd debian/tmp/usr/bin/halcmd_twopass debian/tmp/usr/bin/halmeter debian/tmp/usr/bin/halreport debian/tmp/usr/bin/halrmt debian/tmp/usr/bin/halrun debian/tmp/usr/bin/halsampler debian/tmp/usr/bin/halscope debian/tmp/usr/bin/halshow debian/tmp/usr/bin/halstreamer debian/tmp/usr/bin/haltcl debian/tmp/usr/bin/halui debian/tmp/usr/bin/hbmgui debian/tmp/usr/bin/hexagui debian/tmp/usr/bin/hy_gt_vfd debian/tmp/usr/bin/hy_vfd debian/tmp/usr/bin/image-to-gcode debian/tmp/usr/bin/inivar debian/tmp/usr/bin/io debian/tmp/usr/bin/iov2 debian/tmp/usr/bin/latency-histogram debian/tmp/usr/bin/latency-plot debian/tmp/usr/bin/latency-test debian/tmp/usr/bin/lineardelta debian/tmp/usr/bin/linuxcnc debian/tmp/usr/bin/linuxcnc_info debian/tmp/usr/bin/linuxcnc_module_helper debian/tmp/usr/bin/linuxcnc_var debian/tmp/usr/bin/linuxcnclcd debian/tmp/usr/bin/linuxcncmkdesktop debian/tmp/usr/bin/linuxcncrsh debian/tmp/usr/bin/linuxcncsvr debian/tmp/usr/bin/linuxcnctop debian/tmp/usr/bin/maho600gui debian/tmp/usr/bin/max5gui debian/tmp/usr/bin/mb2hal debian/tmp/usr/bin/mdi debian/tmp/usr/bin/melfagui debian/tmp/usr/bin/milltask debian/tmp/usr/bin/millturngui debian/tmp/usr/bin/mitsub_vfd debian/tmp/usr/bin/monitor-xhc-hb04 debian/tmp/usr/bin/motion-logger debian/tmp/usr/bin/moveoff_gui debian/tmp/usr/bin/ngcgui debian/tmp/usr/bin/panelui debian/tmp/usr/bin/pi500_vfd debian/tmp/usr/bin/pmx485 debian/tmp/usr/bin/pmx485-test debian/tmp/usr/bin/pncconf debian/tmp/usr/bin/puma560gui debian/tmp/usr/bin/pumagui debian/tmp/usr/bin/pyngcgui debian/tmp/usr/bin/pyui debian/tmp/usr/bin/pyvcp debian/tmp/usr/bin/pyvcp_demo debian/tmp/usr/bin/qtplasmac-cfg2prefs debian/tmp/usr/bin/qtplasmac-materials debian/tmp/usr/bin/qtplasmac-plasmac2qt debian/tmp/usr/bin/qtplasmac_gcode debian/tmp/usr/bin/qtvcp debian/tmp/usr/bin/rotarydelta debian/tmp/usr/bin/rs274 debian/tmp/usr/bin/rtapi_app debian/tmp/usr/bin/scaragui debian/tmp/usr/bin/schedrmt debian/tmp/usr/bin/scorbot-er-3 debian/tmp/usr/bin/sendkeys debian/tmp/usr/bin/shuttle debian/tmp/usr/bin/sim-torch debian/tmp/usr/bin/sim_pin debian/tmp/usr/bin/simulate_probe debian/tmp/usr/bin/stepconf debian/tmp/usr/bin/svd-ps_vfd debian/tmp/usr/bin/teach-in debian/tmp/usr/bin/thermistor debian/tmp/usr/bin/tool_mmap_read debian/tmp/usr/bin/tool_watch debian/tmp/usr/bin/tooledit debian/tmp/usr/bin/touchy debian/tmp/usr/bin/update_ini debian/tmp/usr/bin/vfdb_vfd debian/tmp/usr/bin/vfs11_vfd debian/tmp/usr/bin/wj200_vfd debian/tmp/usr/bin/xhc-hb04 debian/tmp/usr/bin/xhc-hb04-accels debian/tmp/usr/bin/xhc-whb04b-6 debian/tmp/usr/bin/xyzac-trt-gui debian/tmp/usr/bin/xyzbc-trt-gui debian/linuxcnc-uspace//usr/bin/
	install -d debian/linuxcnc-uspace//usr/lib
	cp --reflink=auto -a debian/tmp/usr/lib/libcanterp.so.0 debian/tmp/usr/lib/liblinuxcnchal.so.0 debian/tmp/usr/lib/liblinuxcncini.so.0 debian/tmp/usr/lib/libnml.so.0 debian/tmp/usr/lib/libposemath.so.0 debian/tmp/usr/lib/libpyplugin.so.0 debian/tmp/usr/lib/librs274.so.0 debian/tmp/usr/lib/libtooldata.so.0 debian/tmp/usr/lib/linuxcnc debian/linuxcnc-uspace//usr/lib/
dh_install: missing files, aborting
	install -d debian/.debhelper/generated/linuxcnc-uspace
	install -d debian/.debhelper/generated/linuxcnc-doc-en
	install -d debian/.debhelper/generated/linuxcnc-doc-fr
	install -d debian/.debhelper/generated/linuxcnc-doc-es
	install -d debian/.debhelper/generated/linuxcnc-doc-zh-cn
make: *** [debian/rules:38: binary] Error 25
dpkg-buildpackage: error: fakeroot debian/rules binary subprocess returned exit status 2
debuild: fatal error at line 1182:
dpkg-buildpackage -us -uc -ui failed

```

ihabmmali, 2022-08-24:
> It is not ignored, but the no-docs build need some of the tools earlier only used by the docs build. 
What I dont understand is why the no-docs build didn't need these packages before but now it does. I had to install 877MB of additional tools and language files not relevant to the main build just to get the build to work

>Missing build dependencies would be detected by debuild, which is one of the reasons I am more interested in test builds using that tool than ones using dpkg-buildpackage. 
I didn't see any difference in behavior between debuild and dpkg-buildpackage but I don't know much about the debian packaging system. To me both debuild and dpkg-build report missing dependencies and won't let the build proceed before the dependencies are satisfied

petterreinholdtsen, 2022-08-24:
[Eyecon]
> What I dont understand is why the no-docs build didn't need these
> packages before but now it does. I had to install 877MB of additional
> tools and language files not relevant to the main build just to get
> the build to work

This happened in ca8510d9a4f0560e57cac8868c90c74cd715a592 and it contain
this explanation:

  The manpages go in linuxcnc-uspace.deb (not in linuxcnc-doc-en), so
  they are built with the architecture-dependent ("binary") debs.
    
  Since some manpages are written in asciidoc rather than roff, these
  build dependencies are needed in Build-Depends (which is satisfied for
  both architecture-dependent and architecture-independent builds), not
  just in Build-Depends-Indep.

The no-docs option disable building the HTML and PDF editions that would
get separate linuxcnc-doc packages, but man pages are still created.

-- 
Happy hacking
Petter Reinholdtsen

ihabmmali, 2022-08-24:
> [Eyecon]
> What I dont understand is why the no-docs build didn't need these packages before but now it does. I had to install 877MB of additional tools and language files not relevant to the main build just to get the build to work
> This happened in [ca8510d](https://github.com/LinuxCNC/linuxcnc/commit/ca8510d9a4f0560e57cac8868c90c74cd715a592) and it contain this explanation: The manpages go in linuxcnc-uspace.deb (not in linuxcnc-doc-en), so they are built with the architecture-dependent ("binary") debs. Since some manpages are written in asciidoc rather than roff, these build dependencies are needed in Build-Depends (which is satisfied for both architecture-dependent and architecture-independent builds), not just in Build-Depends-Indep. The no-docs option disable building the HTML and PDF editions that would get separate linuxcnc-doc packages, but man pages are still created.
> […](#)
> -- Happy hacking Petter Reinholdtsen

Thanks Petter, that makes sense. So any idea what we can do to generate the missing man pages causing dh_install to fail?

cnc-nexT, 2022-08-28:
Hello [petterreinholdtsen], [annhan]
since I have the same problem not beeing able to build the *.deb's .... is there any progress with that matter ? or any idea when it can be expected ?
What is a solution to work around that problem ?

petterreinholdtsen, 2022-08-28:
[udoS]
> Hello [petterreinholdtsen], [annhan]
> since I have the same problem not beeing able to build the *.deb's
> .... is there any progress with that matter ? or any idea when it can
> be expected ?

No idea.

> What is a solution to work around that problem ?

The obvious workaround is to drop the no-docs debian/configure option.

-- 
Happy hacking
Petter Reinholdtsen

closed 2022-08-28 by petterreinholdtsen

annhan, 2022-08-29:
i removed  no-docs in configure. But i am still getting error:

```
dh_install: Cannot find (any matches for) "usr/lib/python3" (tried in ., debian/tmp)

dh_install: linuxcnc-uspace missing files: usr/lib/python3
	install -d debian/.debhelper/generated/linuxcnc-uspace-dev
	install -d debian/linuxcnc-uspace/
	cp --reflink=auto -a debian/tmp/etc debian/linuxcnc-uspace//
	install -d debian/linuxcnc-uspace//usr/bin
	cp --reflink=auto -a debian/tmp/usr/bin/5axisgui debian/tmp/usr/bin/axis debian/tmp/usr/bin/axis-remote debian/tmp/usr/bin/classicladder debian/tmp/usr/bin/debuglevel debian/tmp/usr/bin/elbpcom debian/tmp/usr/bin/genserkins debian/tmp/usr/bin/gladevcp debian/tmp/usr/bin/gladevcp_demo debian/tmp/usr/bin/gmoccapy debian/tmp/usr/bin/gremlin_view debian/tmp/usr/bin/gs2_vfd debian/tmp/usr/bin/gscreen debian/tmp/usr/bin/hal-histogram debian/tmp/usr/bin/hal_input debian/tmp/usr/bin/hal_manualtoolchange debian/tmp/usr/bin/halcmd debian/tmp/usr/bin/halcmd_twopass debian/tmp/usr/bin/halmeter debian/tmp/usr/bin/halreport debian/tmp/usr/bin/halrmt debian/tmp/usr/bin/halrun debian/tmp/usr/bin/halsampler debian/tmp/usr/bin/halscope debian/tmp/usr/bin/halshow debian/tmp/usr/bin/halstreamer debian/tmp/usr/bin/haltcl debian/tmp/usr/bin/halui debian/tmp/usr/bin/hbmgui debian/tmp/usr/bin/hexagui debian/tmp/usr/bin/hy_gt_vfd debian/tmp/usr/bin/hy_vfd debian/tmp/usr/bin/image-to-gcode debian/tmp/usr/bin/inivar debian/tmp/usr/bin/io debian/tmp/usr/bin/iov2 debian/tmp/usr/bin/latency-histogram debian/tmp/usr/bin/latency-plot debian/tmp/usr/bin/latency-test debian/tmp/usr/bin/lineardelta debian/tmp/usr/bin/linuxcnc debian/tmp/usr/bin/linuxcnc_info debian/tmp/usr/bin/linuxcnc_module_helper debian/tmp/usr/bin/linuxcnc_var debian/tmp/usr/bin/linuxcnclcd debian/tmp/usr/bin/linuxcncmkdesktop debian/tmp/usr/bin/linuxcncrsh debian/tmp/usr/bin/linuxcncsvr debian/tmp/usr/bin/linuxcnctop debian/tmp/usr/bin/maho600gui debian/tmp/usr/bin/max5gui debian/tmp/usr/bin/mb2hal debian/tmp/usr/bin/mdi debian/tmp/usr/bin/melfagui debian/tmp/usr/bin/milltask debian/tmp/usr/bin/millturngui debian/tmp/usr/bin/mitsub_vfd debian/tmp/usr/bin/monitor-xhc-hb04 debian/tmp/usr/bin/motion-logger debian/tmp/usr/bin/moveoff_gui debian/tmp/usr/bin/ngcgui debian/tmp/usr/bin/panelui debian/tmp/usr/bin/pi500_vfd debian/tmp/usr/bin/pmx485 debian/tmp/usr/bin/pmx485-test debian/tmp/usr/bin/pncconf debian/tmp/usr/bin/puma560gui debian/tmp/usr/bin/pumagui debian/tmp/usr/bin/pyngcgui debian/tmp/usr/bin/pyui debian/tmp/usr/bin/pyvcp debian/tmp/usr/bin/pyvcp_demo debian/tmp/usr/bin/qtplasmac-cfg2prefs debian/tmp/usr/bin/qtplasmac-materials debian/tmp/usr/bin/qtplasmac-plasmac2qt debian/tmp/usr/bin/qtplasmac_gcode debian/tmp/usr/bin/qtvcp debian/tmp/usr/bin/rotarydelta debian/tmp/usr/bin/rs274 debian/tmp/usr/bin/rtapi_app debian/tmp/usr/bin/scaragui debian/tmp/usr/bin/schedrmt debian/tmp/usr/bin/scorbot-er-3 debian/tmp/usr/bin/sendkeys debian/tmp/usr/bin/shuttle debian/tmp/usr/bin/sim-torch debian/tmp/usr/bin/sim_pin debian/tmp/usr/bin/simulate_probe debian/tmp/usr/bin/stepconf debian/tmp/usr/bin/svd-ps_vfd debian/tmp/usr/bin/teach-in debian/tmp/usr/bin/thermistor debian/tmp/usr/bin/tool_mmap_read debian/tmp/usr/bin/tool_watch debian/tmp/usr/bin/tooledit debian/tmp/usr/bin/touchy debian/tmp/usr/bin/update_ini debian/tmp/usr/bin/vfdb_vfd debian/tmp/usr/bin/vfs11_vfd debian/tmp/usr/bin/wj200_vfd debian/tmp/usr/bin/xhc-hb04 debian/tmp/usr/bin/xhc-hb04-accels debian/tmp/usr/bin/xhc-whb04b-6 debian/tmp/usr/bin/xyzac-trt-gui debian/tmp/usr/bin/xyzbc-trt-gui debian/linuxcnc-uspace//usr/bin/
	install -d debian/linuxcnc-uspace//usr/lib
	cp --reflink=auto -a debian/tmp/usr/lib/libcanterp.so.0 debian/tmp/usr/lib/liblinuxcnchal.so.0 debian/tmp/usr/lib/liblinuxcncini.so.0 debian/tmp/usr/lib/libnml.so.0 debian/tmp/usr/lib/libposemath.so.0 debian/tmp/usr/lib/libpyplugin.so.0 debian/tmp/usr/lib/librs274.so.0 debian/tmp/usr/lib/libtooldata.so.0 debian/tmp/usr/lib/linuxcnc debian/linuxcnc-uspace//usr/lib/
dh_install: missing files, aborting
	install -d debian/.debhelper/generated/linuxcnc-uspace
	install -d debian/.debhelper/generated/linuxcnc-doc-en
	install -d debian/.debhelper/generated/linuxcnc-doc-fr
	install -d debian/.debhelper/generated/linuxcnc-doc-es
	install -d debian/.debhelper/generated/linuxcnc-doc-zh-cn
make: *** [debian/rules:38: binary] Error 25
dpkg-buildpackage: error: fakeroot debian/rules binary subprocess returned exit status 2
debuild: fatal error at line 1182:
dpkg-buildpackage -us -uc -ui failed
```

before merge these functions, i built deb package success

petterreinholdtsen, 2022-08-29:
[NhanPham]
> i removed  no-docs in configure. But i am still getting error:
>
> ```
> dh_install: Cannot find (any matches for) "usr/lib/python3" (tried in
> ., debian/tmp)

The no-docs issue should be fixed in the latest master, with commit
7b337b3d90831b2310c17c860ebcdf0a2aeb912e.

If your build is missing usr/lib/python3 (in
debian/tmp/usr/lib/python3/), something else went wrong earlier in the
build.  I can not reproduce this on Debian 11/Bullseye or Debian
12/Bookworm.

> before merge these functions, i built deb package success

Can you provide the commit ID when it was last building for you?  A
complete build log would be useful too.

-- 
Happy hacking
Petter Reinholdtsen

annhan, 2022-08-29:
> Can you provide the commit ID when it was last building for you?  A
complete build log would be useful too.
ID of commit in my last built:   4c34757702019f090c3f564b7669395bceebaa12

Buildinfo file :
[linuxcnc_2.9.0~pre0_amd64.buildinfo.txt](https://github.com/LinuxCNC/linuxcnc/files/9441559/linuxcnc_2.9.0.pre0_amd64.buildinfo.txt)

* Error build logfile:
[error_log_j1.txt](https://github.com/LinuxCNC/linuxcnc/files/9442390/error_log_j1.txt)

* Error in terminal

```
Reading 205/205 dependency files
Done reading dependencies
Reading 228/228 realtime dependency files
Done reading realtime dependencies
Reading 205/205 dependency files
Done reading dependencies
Reading 228/228 realtime dependency files
Done reading realtime dependencies
Reading 205/205 dependency files
Done reading dependencies
Reading 228/228 realtime dependency files
Done reading realtime dependencies
Updating index cache for path `/home/mwork/linuxcnc/src/../docs/man/man1'. Wait...done.
troff: man1/xhc-whb04b-6.1:746: warning [p 183, 1.7i]: can't break line
 fakeroot debian/rules binary
Reading 205/205 dependency files
Done reading dependencies
Reading 228/228 realtime dependency files
Done reading realtime dependencies
/bin/bash: ldconfig: command not found
make[2]: [Makefile:728: install-kernel-indep] Error 127 (ignored)
Reading 205/205 dependency files
Done reading dependencies
Reading 228/228 realtime dependency files
Done reading realtime dependencies
dh_install: Cannot find (any matches for) "usr/lib/python3" (tried in ., debian/tmp)

dh_install: linuxcnc-uspace missing files: usr/lib/python3
dh_install: missing files, aborting
make: *** [debian/rules:38: binary] Error 25
dpkg-buildpackage: error: fakeroot debian/rules binary subprocess returned exit status 2

```

thank you so much

petterreinholdtsen, 2022-08-29:
[NhanPham]
> * Error build logfile:
> [error_log_j1.txt](https://github.com/LinuxCNC/linuxcnc/files/9442390/error_log_j1.txt)

This file show the problem.  Somehow, your python setup stuff its files
in debian/tmp/usr/lib/python3.7/ instead of
debian/tmp/usr/lib/python3/.  No idea why, nor how to avoid it.

This is the first line in the log documenting the problem:

  install -o root -d -m 0755 \
    /home/mwork/linuxcnc/debian/tmp/usr/lib/python3.7/site-packages \
    /home/mwork/linuxcnc/debian/tmp/usr/lib/python3.7/site-packages/rs274

A workaround is to change usr/lib/python3 in debian/linuxcnc.install.in
to usr/lib/python3.7.  No idea how to properly fix it in a generic way.

How come you got python 3.7?  According to
<URL:https://tracker.debian.org/pkg/python3-defaults>, Debian 11 got 3.9
and Debian 12 got 3.10.  Are you building on Debian 10, aka
oldstable/buster?

-- 
Happy hacking
Petter Reinholdtsen

annhan, 2022-08-29:
> How come you got python 3.7?  According to
<URL:https://tracker.debian.org/pkg/python3-defaults>, Debian 11 got 3.9
and Debian 12 got 3.10.  Are you building on Debian 10, aka
oldstable/buster?

Yes, i am building in debian 10 buster.  I installed from ISO which i downloaded from linuxcnc website.
Maybe i  installed python 3.7 for my job.
but i can build success in previous version.
thank you so much

petterreinholdtsen, 2022-08-29:
[NhanPham]
> Yes, i am building in debian 10 buster.  I installed from ISO which i
> downloaded from linuxcnc website.

Aha.  I guess that explain it.  I doubt it it useful to try to keep the
master branch building on Debian 10 Buster aka Oldstable, and I spend my
time focusing on Debian 11 and 12.

I've spent some time preparing a LinuxCNC live ISO build for Debian 12.
Hope it will be published soon.

-- 
Happy hacking
Petter Reinholdtsen

cnc-nexT, 2022-08-29:
No idea.
 What is a solution to work around that problem ?
The obvious workaround is to drop the no-docs debian/configure option.

I have a Debian11 5.0.17 PREEMT installed.
I got: git clone https://github.com/LinuxCNC/linuxcnc.git linuxcnc-dev
No missing dependencies;
linuxcnc-dev/debian > /configure uspace          I tried with and without  ( no-docs )
linuxcnc-dev> dpkg-buildpackage -b -uc         as per ( Building LinuxCNC ) Html-doc
I have a /usr/lib/python3/dist-packages/  
and a    /usr/lib/python3.9

dh_install: warning: Cannot find ( .... ) " /usr/lib/python3 " ........
dh_install: error: missing files, aborting

<"A workaround is to change usr/lib/python3 in debian/linuxcnc.install.in"> NOP

Thanks for any useful help:

annhan, 2022-08-29:
> 

B1. Get your python version. can you check version of python in workfolder/linuxcnc/debian/tmp/usr/lib/python?
B2. Change line 111  in workfolder/linuxcnc/debian/linuxcnc.install.in file.
from ```usr/lib/python``` to ```usr/lib/python_Version_of_you``` exam: ```usr/lib/python3.9```
B3. 
```
cd debian
./configure no-docs uspace
cd..
dpkg-buildpackage -b -uc
```

cnc-nexT, 2022-08-29:
sorry,... my mistake; there is. python3/dist-packages/

cnc-nexT, 2022-08-29:
same for me ! But /python3

cnc-nexT, 2022-08-29:
what is wrong here ? 
this is now not a one day problem;
A bare Debian Linux11 xfce Installation; 5.0.17 by the way;
Clone the git LinuxCNC
doesn't work and no solution at the horizon ?

cnc-nexT, 2022-08-29:
Is This topic considered to be solved/closed ; or ?

petterreinholdtsen, 2022-08-30:
[udoS]
> Is This topic considered to be solved/closed ; or ?

Yes.  The issue with the missing manual pages breaking the build is
fixed.

--
Happy hacking
Petter Reinholdtsen

cnc-nexT, 2022-08-30:
NO --> IT IS NOT

read my previous post. 
> A bare Debian Linux11 xfce Installation; 5.0.17 by the way;
> Clone the git LinuxCNC
> doesn't work 
as of 2022.08.29-18:23 CET
So if there is a WORKING solution just let me know; maybe I have missed it;

Best regards

petterreinholdtsen, 2022-08-30:
[udoS]
> NO --> IT IS NOT
>
> read my previous post.

The error reported by Eyecon as the first message in this issue is as
far as I know solved in 7b337b3d90831b2310c17c860ebcdf0a2aeb912e.

Other issues have been mentioned by other people as well, beliving it to
be identical to this issue, which is fine of course, but that do not
really change the state of this reported issue.  I assume here no-one
really intended to hijack this issue for another topic.  I see no
advantage of lumping independent issues with independent solutions into
the same github issue.  If you have a different error than the one
reported by Eyecon, I suggest creating a new issue for it.  At the start
it might have seen like other build problems were related, but during
the discussion it has become obvious to me that they are not.  What make
you believe your current issue is the same as the one reported by
Eyecon?

-- 
Happy hacking
Petter Reinholdtsen

cnc-nexT, 2022-08-30:
@petterreinholdtsen
I will leave it at that; as in my previous posts I have the same problem building Linuxcnc *.deb with the same dh_install missing files as mentioned in this post. Wether or not using no-docs option. What I normally do.
Last time I build successfully was on 2022.06.22 with option no-docs; Without wouldn't work.
So, I was just looking into this in the hope to find a way to build. Looks like this is not the case.

Sorry if I wasted your time
Best Regards

commit 7b337b3d (2022-08-28): Adjusted build rules to build man pages also in the no-docs setting

The manual pages are part of the linuxcnc-uspace package and need to
be built and installed even when not building doc packages.

Fixes #1941.