LinuxCNC/linuxcnc #1943: diagnostic in carousel.comp may indicate problem

jepler, 2022-08-20:
While rebasing #1278 I noticed this diagnostic in carousel.comp:
```
hal/components/carousel.comp:249:21: warning: ‘align_pin’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  249 |         align_pin = ! align_pin; // Not supported, but wiggle the pin to escape
      |                     ^~~~~~~~~~~
```

`align_pin` is a function level variable, but it looks like the use on line 249 is intended to give it an alternating value down below where the state of `align_pin` is used to control passing from state 6 to 8 in the positioning state machine.

If `align_pin` is used uninitialized, it _might_ sometimes appear to be working, as it may get a left-over value that changes from time to time, but there's no guarantee this would actually happen.  Instead, the code should probably be changed so that this alternation isn't required, perhaps by explicitly skipping the states that don't apply to an E(dge)-type carousel.

This is all conjecture since I don't have anything to try the component on.

andypugh, 2022-08-21:
The weird thing is, I don't have anything to try it on either....

carousel.comp was a response to a clear need, but not one I had myself. 

The idea here was to bypass an edge case. Maybe align_pin could be initialised to zero in setup. Or that line could be deleted. 

But, it seems to me that even if it is unitialised junk, the first time through it will be interpreted as boolean true or false, and from that point on will toggle true to false and false to true every servo cycle?

petterreinholdtsen, 2022-08-21:
[andypugh]
> The weird thing is, I don't have anything to try it on either....

I use carousel, but have no idea how to test that particular pin.
Anyone care to give some recipe on how to test it?

-- 
Happy hacking
Petter Reinholdtsen

andypugh, 2022-08-21:
There seems to be a problem with it during jogging. https://forum.linuxcnc.org/38-general-linuxcnc-questions/46405-carousel-comp-jog-fwd-still-an-issue?start=10#248412

spumco1234, 2022-08-22:
I'm the edge-case Andy helped out.  And yes, jogging is still malfunctioning in one direction when the pin alignment slow search function is enabled.  No issues in counts mode with the slow search turned off.

Is there anything I can do to help with diagnosing or testing?

jepler, 2022-08-24:
In #1953 I added an explicit initialization of the local `align_pin` variable that was the subject of the diagnostic. This fixes the undefined behavior diagnostic, but will almost definitely make 'E'dge-type carousels get stuck at state 6 or 7 consistently instead of progressing unpredictably due to the undefined behavior resulting from the uninitialized value.

andypugh, 2022-08-24:
Is that the right PR number? It seems unrelated.

petterreinholdtsen, 2022-10-11:
[spumco1234]
> I'm the edge-case Andy helped out.  And yes, jogging is still
> malfunctioning in one direction when the pin alignment slow search
> function is enabled.  No issues in counts mode with the slow search
> turned off.
>
> Is there anything I can do to help with diagnosing or testing?

Perhaps you can provide a simulated test case in a form similar to the
other collections in tests/ to demonstrate your problem?  This way we
can all reproduce the problem and hopefully also find a fix for it.

-- 
Happy hacking
Petter Reinholdtsen

spumco1234, 2022-10-13:
> Perhaps you can provide a simulated test case in a form similar to the other collections in tests/ to demonstrate your problem? This way we can all reproduce the problem and hopefully also find a fix for it.

Not really sure how to provide a simulated test case.  An explanation of the observed behavior is in the forum thread discussing the issue.  I can explain it again here if that would help.

https://forum.linuxcnc.org/38-general-linuxcnc-questions/46405-carousel-comp-jog-fwd-still-an-issue?start=10#248503

Thinking about it, I'm not sure how a simulation would work.  The counts mode depends not only on a stepgen output to move the carousel, but also an input pin that seems to trigger the 'slow search' function Andy wrote.  If somone set up a sim and used halshow to watch the stepgen counts go up/down as it reacts to the carousel.comp commands, something would have to trigger the pocket sense pin at approximately the right time to get carousel to do the slow search (back off pin).

To be honest, I punted and turned off the slow-search function.  My ATC is now in 'dumb' counts mode and I'm relying on the stepper drive to alarm if it misses steps.  The bug is still there in carousel.comp, but it's a non-issue for me now.

andypugh, 2023-05-07:
Other work has superceded this

closed 2023-05-07 by andypugh