LinuxCNC/linuxcnc #1944: Is it possible to support remote scenarios for LinuxCNC

summerlotus513, 2022-08-20:
I noticed that there is a framework QtQuickVcp on another derivative version of LinuxCNC (machinekit), with modules such as halltalk and mkwrapper, it is possible for me to remotely control my machines in the LAN on any platform. I have also tried it and it does meet some of my needs. I noticed that LinuxCNC seems to have no related content, or does not provide a framework similar to QtQuickVcp for our secondary development.
Is it possible to support QtQuickVcp or other forms of remote solutions in the future? I see that libnml seems to be supported, but IMHO, it is currently not easy to use in the remote control scenario, or it may be because my configuration method is incorrect.

andypugh, 2022-08-20:
What do you want to do? Machinekit has better support for GUI and realtime on different CPUs, but there are other ways to remote control LinuxCNC, depending on where you want to draw the boundary.

silopolis, 2022-08-20:
Le sam. 20 août 2022 à 12:56, andypugh ***@***.***> a écrit :

> Machinekit has better support for GUI and realtime on different CPUs,
>

Could you elaborate please?

We need a page somewhere to list and detail differences with MK, with a
note about their backportability when they can be desirable.

but there are other ways to remote control LinuxCNC, depending on where you
> want to draw the boundary.
>

AFAIR, we don't have page in docs about ways to remote control LinuxCNC or
split components on different CPUs.

Just created #1949 and #1950 to gather these informations.

andypugh, 2022-08-21:
> Le sam. 20 août 2022 à 12:56, andypugh ***@***.***> a écrit :
> Machinekit has better support for GUI and realtime on different CPUs,
> Could you elaborate please? 

Machinekit decided to commit hard to the Beaglebone Black (The PRU's are very interesting). But it has lousy graphics performance so they worked on running motion / interp(?) on the BBB and GUIs on remote machines. 

That's as much as I know. I ceased involvement in MK fairly early on, when it was made clear that my input was not welcome.

smoe, 2022-08-28:
I have seen a raspberry pi configured for windows remote desktop - ah, here: https://www.youtube.com/watch?v=g-fzD8sEcIk

I would volunteer to add a respective description for our documentation.