LinuxCNC/linuxcnc #1952: Gtk tooltable offsets are not beeing applied, when offset was zero before

hansu, 2022-08-21:
This affects gscreen and gmoccapy.

### Steps to reproduce the issue:
1. Change to a tool with an offset = 0.0 
2. Change back to a tool with an offset not equal zero

After that, the tool offset is still zero. But I noticed that the machine first goes to the position of the new tool, and then back to the old position.
After 

[image]

[image]

After restarting LinuxCNC and referencing, the offset is applied correcty: 

[image]

hansu, 2022-10-04:
This happens only when using the tool change without G43 - which shouldn't apply any offsets. So far ok, but according to the docs: "Please note, that GMOCCAPY takes care of its own to update the offsets, sending an G43 after any tool change, but not in auto mode!"
So it seams that G43 is not sent reliably.

The main problem was, that I uncommented the POSTGUI HAL file which

But I have a question for @gmoccapy:

gmoccapy, 2023-10-07:
> The main problem was, that I uncommented the POSTGUI HAL file which connected gmoccapy.tooloffset-z to motion.tooloffset.z
> I only wonder why this connection has to be made externally - @gmoccapy can you clarify this? 

This is, because it can be done more easily in Hal than in the GUI code. IMHO that is also a part a GUI should not take care about, but LinuxCNC itself.

The bug is not related to gmoccapy. After a tool offset of Z=0 is set, the G43 is canceled somewhere else.

As RENE is redefining the complete tool handling in LinuxCNC we must wait for his final push before going on with this issue.

gmoccapy, 2026-10-28:
Can be closed as gmoccapy unfortunately does not support any more auto G43 due to very very rare  problems.

Norbert

closed 2026-10-28 by gmoccapy