LinuxCNC/linuxcnc #1954: incorrect manpage grouping in html doc index

jepler, 2022-08-22:
I noticed that rather than being grouped under the headings "API calls" everything has been moved down from the correct sections to "API calls: general". http://linuxcnc.org/docs/devel/html/#man_3hal

hansu, 2022-08-27:
I found the line causing this: https://github.com/LinuxCNC/linuxcnc/commit/168289d3fed81377f218e98048bcb30be5d3d4b5#r82354641.
Maybe @andypugh knows best how to fix this while keeping his improvement.

andypugh, 2022-08-30:
Sorry, can't really help. If I ever understood this it was only for long enough to do what I wanted to do.

rmu75, 2025-12-28:
To me it looks like API stuff is proberly grouped in the documentation section. Is there (still) something to do?

closed 2025-12-28 by jepler