LinuxCNC/linuxcnc #1974: v2.9.0-pre0-7604-g9aa4ee9e77 and beyond package build errors

ihabmmali, 2022-08-29:
## Here are the steps I follow to reproduce the issue:
 1. cd [Linuxcnc source directory]
 2. git pull origin master
 3. cd debian
 4. ./configure no-docs uspace
 5. cd ..
 6.  dpkg-buildpackage -b -uc
 

## This is what I expected to happen:
successful compilation and generation of deb packages

## This is what happened instead:
dh_missing: warning: usr/bin/z_level_compensation exists in debian/tmp but is not installed to anywhere 
	The following debhelper tools have reported what they installed (with files per package)
	 * dh_install: linuxcnc-uspace (255), linuxcnc-uspace-dev (13)
	 * dh_installdocs: linuxcnc-uspace (2), linuxcnc-uspace-dev (0)
	 * dh_installman: linuxcnc-uspace (0), linuxcnc-uspace-dev (0)
	If the missing files are installed by another tool, please file a bug against it.
	When filing the report, if the tool is not part of debhelper itself, please reference the
	"Logging helpers and dh_missing" section from the "PROGRAMMING" guide for debhelper (10.6.3+).
	  (in the debhelper package: /usr/share/doc/debhelper/PROGRAMMING.gz)
	Be sure to test with dpkg-buildpackage -A/-B as the results may vary when only a subset is built
	If the omission is intentional or no other helper can take care of this consider adding the
	paths to debian/not-installed.
dh_missing: error: missing files, aborting
make: *** [debian/rules:38: binary] Error 255
dpkg-buildpackage: error: fakeroot debian/rules binary subprocess returned exit status 2

## It worked properly before this:

The z compensation module introduced in 9aa4ee9e77ae17e624763dca98a1cdc9edd7cbb3 causes the error, suspecting a similar issue that broke the "no-docs" build before.  

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`): Debian 11
 * I am using this kernel version (shown by `uname -a`): 5.18.0-2-rt-amd64
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [X ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):v2.9.0-pre0-7604-g9aa4ee9e77

 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):qtdragon
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): Mesa cards

c-morley, 2022-08-30:
I pushed (I think) a fix for the z_level_compensation error. Thank you for reporting.
If this fixes the error then please close the issue.
Thanks.

petterreinholdtsen, 2022-08-30:
Fixed with 885c2597ec4440619989fc9842c6e81afe5a2aaf .

closed 2022-08-30 by petterreinholdtsen