LinuxCNC/linuxcnc #1987: M70/M72 doesn't properly restore feed rate when using G93 between them

jallwine, 2022-09-01:
## Here are the steps I follow to reproduce the issue:

g1_error.ngc:

    %
    (logappend,feed.txt)
    G94 G90
    
    G1 F60 X0
    (log,Should be 60: #<_feed>)
    
    M70
    
    G93
    (log,After G93: #<_feed>)
    
    M72
    (log,After M72: #<_feed>)
    (logclose)
   
    G1 X1
    
    %

 1. Run g1_error.ngc

## This is what I expected to happen:
I expect the machine to move to X0 then to X1 at 60 inches per minute. I would also expect the following to print to feed.txt:

    Should be 60: 60.000000
    After G93: 60.000000
    After M72: 60.000000

## This is what happened instead:
The interpreter errors with `Cannot do g1 with zero feed rate`. feed.txt:

    Should be 60: 60.000000
    After G93: 60.000000
    After M72: 0.000000

## Information about my hardware and software:

I'm using a fork of LinuxCNC 2.8 (EMCApplication+MachineKit-HAL).

jallwine, 2022-09-02:
Similarly, the motion mode doesn't seem to be restored:
```
%

(logappend,motion_mode.txt)

G94 G90 G20

G0 X0
(log,Should be 0: #<_motion_mode>)

M70

G1 F60 X1
(log,Should be 10: #<_motion_mode>)

M72
(log,Should be 0: #<_motion_mode>)
(logclose)

X0

%
```

Running the above results in the same `Cannot do g1 with zero feed rate` and the following output to motion_mode.txt:
```
Should be 0: 0.000000
Should be 10: 10.000000
Should be 0: 10.000000
```

I would expect the above to move to X0 at rapid rate, then to X1 at 60 inches per minute, then back to X0 at rapid rate.

jallwine, 2022-09-02:
Upon closer inspection of the documentation, I guess motion mode isn't supposed to be restored: http://linuxcnc.org/docs/2.8/html/gcode/m-code.html#mcode:m70-saved-state

andypugh, 2023-10-04:
Sorry for the delay. I found a workaround, which seems to have only a very minor cost.

closed 2023-10-04 by andypugh