LinuxCNC/linuxcnc #1996: HAL user programs don't unload when linuxcnc is shutdown

c-morley, 2022-09-04:
# The issue tracker is not a support forum

The LinuxCNC issue tracker is to report bugs in the software.
If you have a question about how to use the software, use one of the other methods detailed on our community support page: http://linuxcnc.org/community/

(delete this section before submitting your bug report)

## Here are the steps I follow to reproduce the issue:

 1.
 2.
 3.

## This is what I expected to happen:

## This is what happened instead:

## It worked properly before this:
(If the behavior changed after making a particular change in hardware or
software, describe the change you think is responsible.  E.g., "after upgrading
from LinuxCNC 2.7.3 to 2.7.4")

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`):
 * I am using this kernel version (shown by `uname -a`):
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card):

c-morley, 2022-09-04:
hmm while now it just seems slow to kill (qtdragon) where qtdragon_hd seemed to hang in the ending phase. using qtdragon/hd in master. I will try other guis soon.
Just interested if anyone else found this too.

petterreinholdtsen, 2022-09-04:
[c-morley]
> hmm while now it just seems slow to kill (qtdragon) where qtdragon_hd
> seemed to hang in the ending phase. using qtdragon/hd in master. I
> will try other guis soon.  Just interested if anyone else found this
> too.

The description of the issue here is too short for me to understand if I
experience the problem or not.  Can you tell us more how to reproduce
it, and how to discover it?

--
Happy hacking
Petter Reinholdtsen

c-morley, 2022-09-04:
I have 5 halmeters loaded via a HAL file (for debugging) when I exit Qtdragon_hd they sometimes stay loaded.
when that happens,it looks like qtdragon_hd doesn't completely finish exiting. 
It seems to coincide with my last pull but I'm in the middle of coding so it's hard to say much for sure.

c-morley, 2022-09-04:
Really seems to be the HAL programs, If I close all the halmeters before shutting qtdragon, qtdragon closes quickly/normally.

The differenence between qtdragon and qtdragon/hd, is the qtdragon_hd loads a userland HAL/python program for auto leveling. I dare say the extra userland program is causing qtdragon_hd to close slowly/never.
More testing as I work

andypugh, 2022-09-05:
Does the levelling comp have a handler for KeyboardInterrupt ? That generally appears in the main loop of a userspace comp.

c-morley, 2022-09-05:
That's a good thought but it does have one, but set up a bit unusually.
I will try playing with it a bit to see if that is the difference.
I can load and unload inside halrun normally.
But I see a difference when using qtdragon (vrs qtdragon_hd), which didn't load this new component and if I close the halmeters before closing qtdragon, it closes normally. I say normally but qt program sometimes segfaults on shutdown - but that's a separate, not new behavior.
Thanks for e suggestion.