LinuxCNC/linuxcnc #1997: gmoccapy: wrong unit in gremlin DRO when machine units are inch

hansu, 2022-09-04:
When starting gmoccapy with 
```
LINEAR_UNITS = inch
```
and 
```
RS274NGC_STARTUP_CODE = G20
```
the DRO in gremlin shows mm. Switching to G21 and then back to G20 makes the correct unit in the gremin DRO appear.

See: https://forum.linuxcnc.org/gmoccapy/46799-preview-in-mm-while-dro-s-are-in-inch#251168

gmoccapy, 2023-10-07:
I am not able to reproduce this behavior, it has been solved somehow.
I close this issue, if the problem does occur again, it may be reopened

hansu, 2023-10-07:
Did you try with 2.8?

hansu, 2023-10-07:
Yes, in 2.8.4 this issue is still present, not in 2.9:

[image]

gmoccapy, 2023-10-07:
This is solved, I do not know who fixed this issue, but i is solved.

closed 2023-10-07 by gmoccapy

hansu, 2023-10-08:
I tried with the latest commit on the 2.8 branch...

gmoccapy, 2023-10-08:
@hansu ; I do not have any computer here in Stuttgart supporting Python2, to check this issue.
If it still poersist, please reopen and fix. 

Should be only comparing both gmoccapy files from 2.8 and 2.9

Norbert