LinuxCNC/linuxcnc #1999: GMoccapy doesnot support optional stop - no button to select it.

c-morley, 2022-09-04:
from
https://forum.linuxcnc.org/gmoccapy/46805-optional-stop-m1#251181

It seems in 2.8 and master the code for optional stop is commented out and there is no button on screen.
No reason is given in the code comments.

gmoccapy, 2022-09-11:
That is right. It has been commented out, because the reload button has been placed on that button list and we only have place for 10 button, so I decided to not show the optional  stops button, as I have never used that feature.

c-morley, 2022-09-13:
Ok. Maybe we should add that to the docs. Optional stop is pretty standard CNC stuff and is supported by linuxcnc, so it's seemed a bit confusing that they couldn't find it.

hansu, 2022-09-13:
We could add an option whether do display the "run optional blocks button" or the "use optional stop button"...

andypugh, 2022-09-13:
Slightly crazy idea, but t could display one or the other or none based on the G-code file content.

hansu, 2022-09-13:
I think that would just lead to more confusion.
Another crazy idea: What about a combined button which enables (or skips) optional blocks and stops on M1?

gmoccapy, 2022-09-13:
@andypugh 
good idea, but what if gcode contains optional blocks and optional stops?

@hansu 
That would lead in a difficult solution, first press optioal stops, second press optional blocks, and thirt press of the button both

On the macro button list and also on 9 axis machines I introduced Next and Previous button, that shows the next buttons that did not have place on the screen. What about a settings entry 
- Show optional Blocks button in Auto mode
- Show optional stop button in Auto mode 

If the user check both an arrow will be shown as in Marco side and so he can reach the hidden button.
default entry would than be the actual screen layout

Unfortunately I will not have time to code that in this year.

Norbert

hansu, 2022-09-13:
> good idea, but what if gcode contains optional blocks and optional stops?

Then one function has to be the dominant one if both are present in the G-code.

> That would lead in a difficult solution, first press optioal stops, second press optional blocks, and thirt press of the button both

I was thinking of a button with only two states. So either enable both or none.

c-morley, 2022-09-14:
The issue only really matters if you need optional stop and block delete. If you use '/m1' then you get optional stop by using block delete. That is what the user figured out in the forum, if you read it all.

I think K.I.S.S is in order here:
So either support (ideally) both with separate buttons or document that it was a conscious decision, so we don't need to dig into the code and guess why it's commented out.

hansu, 2022-09-28:
>What about a settings entry
>
>    Show optional Blocks button in Auto mode
>    Show optional stop button in Auto mode
>
>If the user check both an arrow will be shown as in Marco side and so he can reach the hidden button.

Or we could allow one additional button when the screen resolution / window size allows it.

c-morley, 2022-10-01:
could use a drop down menu from a button - something like this:
https://stackoverflow.com/questions/9132440/gtk-drop-down-menu-to-toolbar-item

hansu, 2022-10-01:
You mean like this?
[image]

c-morley, 2022-10-01:
exactly !

c-morley, 2022-10-01:
To get really fancy you can change the button image based on current state of m1 and block delete.

c-morley, 2022-10-01:
The only problem I can think of is that I think GMoccapy allows external pins to select those buttons and so this would break that paradigm. i am not sure how common that feature is though.

hansu, 2022-10-01:
>The only problem I can think of is that I think GMoccapy allows external pins to select those buttons and so this would break that paradigm. i am not sure how common that feature is though.

Quite common AFAIK

c-morley, 2022-10-01:
Then maybe the HAL pin would only work on the blockdelete - just to keep it consistent with older systems. not ideal but still useful.

hansu, 2022-10-01:
Yes. Would fit better to a fixed number of horizontal buttons than my suggestion:

>Or we could allow one additional button when the screen resolution / window size allows it.

gmoccapy, 2023-10-07:
I have done as first solution:
I added two hal pin, one for optional stops and one for blockdelete to gmoccapy
I added a sample config to show how to add a custom panel to connect a hal toggle button to block delete.

I will try to findd a way to get more space in the bottom bar to add an additional button.
I will discus this with Hansu during the next weeks.

That is why I closed this issue

closed 2023-10-07 by gmoccapy

hansu, 2023-10-08:
Fixed in https://github.com/LinuxCNC/linuxcnc/commit/7d4f3a3b5400dba3e677a1230e6c21abdcb17bd3

hansu, 2023-10-08:
There is unfortunately still an issue (I tested with gmoccapy 9_axis).
At the beginning, there are still too many buttons. Only after clicking on the right arrow button, the number of buttons get reduced:

[image]

Furthermore I noticed that also the minimal height is 836 pixels with 9 axes. This is because of the jog buttons:
[image]

Maybe we should show those buttons in this case in 5 columns and decrease the size. So something like this:

[image]

hansu, 2023-10-08:
Forget the deleted message (wrong content and wrong issue :see_no_evil: )

gmoccapy, 2023-10-08:
You are wellcome