LinuxCNC/linuxcnc #2000: QTPVCP critical error from latest Debian package in SID

rodw-au, 2022-09-05:
**Reported error:**
[QTvcp][CRITICAL]  Aborted from Error Dialog
 Qtvcp encountered an error.  The following information may be useful in troubleshooting:
LinuxCNC Version  : 2.9.0~pre0

Traceback (most recent call last):
  File "/usr/bin/qtvcp", line 511, in <module>
    APP = QTVCP()
  File "/usr/bin/qtvcp", line 310, in __init__
    window.handler_instance.initialized__()
  File "/usr/share/qtvcp/screens/qtplasmac/qtplasmac_handler.py", line 317, in initialized__
    self.init_widgets()
  File "/usr/share/qtvcp/screens/qtplasmac/qtplasmac_handler.py", line 932, in init_widgets
    self.w.filemanager.button2.setText(_translate('HandlerClass', 'USER'))
AttributeError: 'FileManager' object has no attribute 'button2'

 (qtvcp:503)
/usr/bin/linuxcnc: line 963:  4641 Segmentation fault      $EMCDISPLAY -ini "$INIFILE" $EMCDISPLAYARGS $EXTRA_ARGS
Shutting down and cleaning up LinuxCNC...
task: 1239 cycles, min=0.000005, max=0.016119, avg=0.010249, 0 latency excursions (> 10x expected cycle time of 0.010000s)
Note: Using POSIX realtime
LinuxCNC terminated with an error.  You can find more information in the log:
    /home/debian/linuxcnc_debug.txt
and
    /home/debian/linuxcnc_print.txt
as well as in the output of the shell command 'dmesg' and in the terminal

## Here are the steps I follow to reproduce the issue:

 1. Install debian bookworm
 2. install linuxcnc-uspace
 3.  QTplasmac sim opens OK
 4. dist-upgrade to sid
 5. Open another new QTplasmac sim
 6. Error ocurrs

## This is what I expected to happen:
it would open as it does on Bookworm
## This is what happened instead:
QVCP critical error as reported

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`): Debian Sid and linuxcnc-uspace package
 * I am using this kernel version (shown by `uname -a`): 5.19
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org) - debian package in SID
   * [ ] A binary I built myself
   * [ x] A binary version from some other source besides linuxcnc.org - debian package in SID
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): QTplasmac
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): Running on VMNware virtual player

phillc54, 2022-09-05:
AFAIK that was fixed in commit #3b21586e

c-morley, 2022-09-05:
It's fixed. Unfortunately you'll have to wait till someone makes another Debian upload.

rodw-au, 2022-09-05:
I can confirm rebuilding source on SID resolved the issue.
Perhaps the current package in SID can be stopped from going to Bookworm?
We should not be sending broken code to Debian.

I really wanted to find out what version of QTplasmac was included in the new package.

@c-morley thanks.
I guess I now know within 24 hours what version of QTplasmac is included :)

Perhaps we should leave the issue open pending review by some of the others...

petterreinholdtsen, 2022-09-05:
[c-morley]
> It's fixed. Unfortunately you'll have to wait till someone makes
> another Debian upload.

Packages in unstable with release critical bugs (meaning bugs with
severity serious or grave) in unstable but not testing will not
propagate to testing.  So if you want to block linuxcnc from propagating
to bookworm you can file a bug at bugs.debian.org to get the effect.
-- 
Happy hacking
Petter Reinholdtsen

c-morley, 2022-09-05:
Well critical is relative. In this case QtPlasmac will fail to load - I am sure there are others but not all.
I am sure we will upload again.
But I still struggle with our rational of uploading/releasing code without communicating effectively  with the people in the project. Nor uploading development code. 
I assumed we are 'working out the kinks' for inclusion.
I hope people understand that it is all unstable at the moment.

rodw-au, 2022-09-05:
> Well critical is relative. In this case QtPlasmac will fail to load - I am sure there are others but not all. I am sure we will upload again. But I still struggle with our rational of [...]

I do agree here. I think also developers could also take more care with the code they release. eg of the last 9 updates to QTplasmac, 7 appear to be reverts or error fixes on an initial glance. There are two active developers, maybe they could check each other's work?

Let me know if we should report on Debian. I think others should  make that decision....

c-morley, 2022-09-05:
It's a development branch. One can only check so much so often. In this particular case it was only a day to discovery and fix the problem.
It just happened to be the day that the upload was done.

smoe, 2022-09-05:
Sent the bug report to Debian to stop the transition - don't have the number, yet, will edit this comment once this arrives.

SebKuzminsky, 2022-09-05:
I'll make a new dsc for upload, and mark [Debian Bug #1019212](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1019212) as fixed in the changelog.

rodw-au, 2022-09-07:
Tested this morning and can confirm the issue is resolved in Sid. Nice work guys! Much appreciated. 
Closing this. I am sure there are plenty of other things in Sid that will be uncovered.

closed 2022-09-07 by rodw-au

petterreinholdtsen, 2022-10-11:
[c-morley]
> It's a development branch. One can only check so much so often. In
> this particular case it was only a day to discovery and fix the
> problem.
> It just happened to be the day that the upload was done.

It is a development branch, but it is still possible to develop changes
in separate branches and share branches between people without putting
it in linuxcnc/master.  Personally I have several branches where I work
on various changes either reported as pull requests or not yet shared
outside my private github repository collection.

If mistakes happen often and cause breakage to reach unsuspected users,
it might be a good reason to reconsider the quality assurance process.
Perhaps the QTVCP module should get more automated tests to detect these
problems up front instad of after the fact?  Perhaps the two active
developers should test and review each others code before sending
patches to master?  I do not know what will work in this case, but see
that the indicators mentioned might tell us something.

If you are unsure how to test GUI components, I am happy to help if you
approach me on IRC.  A simple test demonstrating one concept is
available as the test-gui branch in 
URL: https://github.com:petterreinholdtsen/linuxcnc/ >.

-- 
Happy hacking
Petter Reinholdtsen

c-morley, 2022-10-12:
The particular error was caused when QtPlasmac referenced an internal name of an object in a  widget.
An update happened to change the internal name of the object. there is really no practiical test for this.
The closest would be to some how automatically load every screen and somehow automatically click on things.
and you must do this on every commit. 
 
The position that master should be good enough for users to use daily, so no breakage is tolerable is not a very practical policy - if you want any serious development.
The same for having a policy that you should work in a development branch until a feature is perfect and finished. Well besides that master is a development branch.

Writing test cases / reviewing each others work is a good idea for sure but one must balance that with the development work it's self. there is only so much man hours available. We could use more developers.

I personally think we should put more energy into releasing more often so users don;t need to use master as often.

andypugh, 2022-10-12:
> I personally think we should put more energy into releasing more often so users don;t need to use master as often.

By "We" you mean me?

petterreinholdtsen, 2022-10-12:
[andypugh]
> By "We" you mean me?

No idea what he ment, but personally I read it to mean the team of
developers, not you in particular.  I do hope you do not feel alone and
abandoned in the task of making new releases.  We are in this together.

-- 
Happy hacking
Petter Reinholdtsen

c-morley, 2022-10-12:
I meant we. You are currently the sharp point of the stick of the mechanics of the actual release I suppose but it takes everyone to have a release ready.