LinuxCNC/linuxcnc #2011: RTAI build broken due to "Don't build tpcomp, it's only useful out-of-tree" commit

NTULINUX, 2022-09-08:
git revert 07e1e991df2e62f0549712a0babb1c83e348dd1b fixes this error:

```
[ntu@ntu-debian]:~/linuxcnc/src$ make
Reading 204/204 dependency files
Done reading dependencies
MAKEFLAGS="" \
/usr/bin/python3.10 modsilent.py make KBUILD_EXTRA_SYMBOLS=/usr/realtime/modules/Module.symvers -C /home/ntu/linux SUBDIRS=`pwd` CC=gcc V=0 modules
make[1]: Entering directory '/home/ntu/linux'
make[1]: *** [Makefile:1549: _module_/home/ntu/linuxcnc/src] Error 2
make[1]: Leaving directory '/home/ntu/linux'
make: *** [Makefile:541: modules] Error 2
```

Had to modify src/Makefile to remove the `$(PYTHON) modsilent.py` to figure out what's going on.

Upon further discovery, was able to get more useful output:

```
[ntu@ntu-debian]:~/linuxcnc/src$ make
Reading 204/204 dependency files
Done reading dependencies
MAKEFLAGS="" \
make KBUILD_EXTRA_SYMBOLS=/usr/realtime/modules/Module.symvers -C /home/ntu/linux SUBDIRS=`pwd` CC=gcc V=0 modules
make[1]: Entering directory '/home/ntu/linux'
make -f ./scripts/Makefile.build obj=/home/ntu/linuxcnc/src
/home/ntu/linuxcnc/src/hal/components/Submakefile:3: /home/ntu/linuxcnc/src/objects/hal/components/tpcomp.mak: No such file or directory
make[2]: *** No rule to make target '/home/ntu/linuxcnc/src/objects/hal/components/tpcomp.mak'.  Stop.
make[1]: *** [Makefile:1546: _module_/home/ntu/linuxcnc/src] Error 2
make[1]: Leaving directory '/home/ntu/linux'
make: *** [Makefile:541: modules] Error 2
```

Which lead to me to use `git log -S tpcomp` and reverted 07e1e991df2e62f0549712a0babb1c83e348dd1b

Build works now.

petterreinholdtsen, 2022-09-08:
Very glad to hear from someone using RTAI, as that build profile seem to
get little testing with no easily available kernel available for
automated testing on github or from Debian. :)

[NTU]
> git revert 07e1e991df2e62f0549712a0babb1c83e348dd1b fixes this error:

So the patch to drop building the useless tpcomp component was
incomplete.  The question is how it was incomplete. :)

> Which lead to me to use `git log -S tpcomp` and reverted
> 07e1e991df2e62f0549712a0babb1c83e348dd1b
>
> Build works now.

Are you sure the issue is not leftover generated dependencies in your
build tree?  Did you try from a fresh git clone or from a tree built
earlier?  I have no idea how else this could happen, but welcome ideas.

-- 
Happy hacking
Petter Reinholdtsen

NTULINUX, 2022-09-08:
`rm -rf * && git reset --hard HEAD` everything was definitely clean! I'm not sure how it's incomplete so I just reverted it. It should be a simple fix for anyone who knows makefiles well.

petterreinholdtsen, 2022-09-08:
[NTU]
> `rm -rf * && git reset --hard HEAD` everything was definitely clean!
> I'm not sure how it's incomplete so I just reverted it.

Very good, the 'rm -rf *' should have removed any lingering generated
dependency information.

> It should be a simple fix for anyone who knows makefiles well.

I very much welcome such simple fix.  Apparently I do not know makefiles
well enough, because I fail to find any reason for your problem in
src/hal/components/Submakefile (the only makefile fragment mentioning
tpcomp) or elsewhere in the code base.  I do hope you know makefiles
well enough to spot it, but fear it might remain unsolved if neither of
us find it. :)

-- 
Happy hacking
Petter Reinholdtsen

andypugh, 2022-09-08:
On Thu, 8 Sept 2022 at 06:26, petterreinholdtsen ***@***.***>
wrote:

>
> Very glad to hear from someone using RTAI, as that build profile seem to
> get little testing with no easily available kernel available for
> automated testing on github or from Debian. :)

I use RTAI pretty much exclusively on my real hardware. But I haven't been
doing compiles very much recently.

-- 
atp

NTULINUX, 2022-09-08:
Hey Andy, I'll be pushing a 4.19.256/257 kernel over the weekend, Debian Bullseye/11 doesn't have any RTAI packages, do you want to set those up once I've updated everything?

andypugh, 2022-09-09:
> Hey Andy, I'll be pushing a 4.19.256/257 kernel over the weekend, Debian Bullseye/11 doesn't have any RTAI packages, do you want to set those up once I've updated everything, assuming #2012 fixes this?

You mean drop the kernel in here: https://www.linuxcnc.org/dists/bullseye/base/binary-amd64/ ?

closed 2022-09-09 by jepler

NTULINUX, 2022-09-09:
Yes please! 6f285604ac1a1b58b2d65d5904ffec3998a833ef will have to be at least partially reverted for RTAI Debian package support.

PR #2012 (2022-09-09): Fix kmod rtai build
@NTULINUX
Closes: #2011