LinuxCNC/linuxcnc #2013: Error handling with multiple user interfaces

c-morley, 2022-09-09:
This is a long known and fundamental problem wih multiple programs that read linuxcnc's error channel.
Since the messages are consumed when read, it creates a race condition that potentially interferes with catching an handling errors.

This goes hand in hand with there being no definitive way to know if a specific MDI command finished without error.
For instance when creating probing screens, abort or other motion errors will stop the current command but the program doesn't know to stop sending more commands. Thinking deeper, using MDI is really not the best way to make such programs, but currently that is all we have.

added to this is python programs each run in their own context, hence each read the error channel.

While banging up against this problem for years, I have pondered solutions.
The one that seems to make sense to me, is nothing new  - ZMQ or equivalent broadcasting the errors, probably with a serial number.

I could actually trial this by adding a 'shim' program between linuxcnc and the python Gstat library.
Ultimately though, I would think it would be better if linuxcnc it's self used this broadcasting program to talk to every thing.

There are other related things that could go with this discussion, such as the error messages that comes to the GUI can not easily be filtered, which is really annoying. I would think linuxcnc sending some sort of code to the GUI and the the integrator decide what to filter out and what to actually print to the user would be far better.

This issue marks that we know of the problem, and hopefully open some discussion on the best solution to work on.

petterreinholdtsen, 2022-09-09:
[c-morley]
> While banging up against this problem for years, I have pondered solutions.
> The one that seems to make sense to me, is nothing new  - ZMQ or
> equivalent broadcasting the errors, probably with a serial number.

Given that the Zero Message Queue is basicly a networked
Publis-Subscribe patter,
<URL: https://en.wikipedia.org/wiki/Publish%E2%80%93subscribe_pattern >,
and LinuxCNC already have message passing abilities, might it not be
simpler to extend the Linuxcnc message passing system to also do
Publish-Subscribe.  It seem to me like a simpler solution than
introducing the complexities of ZMQ configuration and maintenance.

Of course ZMQ would provide some new features, like being able to pass
the messages over the Internet.  I am just not sure the increased
network exposure is worth it, security and complexity wise, to improve
error handling in LinuxCNC.

-- 
Happy hacking
Petter Reinholdtsen

c-morley, 2022-09-09:
Extending NML might be easier, I don't know - hopefully someone in the know will comment.

One reason to switch messaging systems (there was talk of getting rid of NML at one time) would be that there are ready made libraries that are maintained, tested, improved and documented by someone else that are experts in that field.
The documentation and c/python libraries fact is worth the consideration alone.
Of course I don't know how much work it would be to switch error handling to something else either. Again hopefully someone who knows will comment.
Machinekit uses it with HAL, but I'm not sure if they got any farther then that.

It would also be nice for user programs to talk to each other in a some way besides HAL, So maybe status and error handling could be extended / re-libraried.

Anyways it sure would be nice to make potentially lethal machines, abort properly, with a useful message.