LinuxCNC/linuxcnc #2024: software encoder causes error when spun 'rtapi_app: caught signal 8 - dumping core'

samcoinc, 2022-09-11:
1) Run the attached sample config in master.  
2) hook an encoder to printer port pin 10 and 11  (I am using a 2000 count - 500 line)
3) run sample config and spin the encoder

Expected:  Nothing should happen.
Observed: linuxcnc crashes with 'rtapi_app: caught signal 8 - dumping core'

I narrowed between these 2 commits then did a git bisect

c8074d7a002850eb550e27cfd4cd80f494c8a5ea  good
0c79d367b05d820fe6434a9a474789392802c561  bad

Looks like it is 
Author: andypugh <andy@bodgesoc.org>
Date:   Mon Jul 19 23:20:27 2021 +0100

    encoder: Add missing-tooth index
    
    This allows a single channel to supply both position and index, most likely
    to be useful for IO-limited lathe configurations.
    
    Signed-off-by: andypugh <andy@bodgesoc.org>

:040000 040000 b453ce776bffebee855943a019c8e122c1747aea 1789e99975963859271560f1473a6104cb0c3c67 M	docs
:040000 040000 e3ffd27f364a8c7d5933948cd7973a0f1212fbf6 e075746587105c65fc33efeffac5f0cce64c0883 M	src

There might be a hal only example that would show this - maybe with a encoder and an sim encoder?  I don't know what the actual issue is - Talking with bari - it might be an issue with 2 or more transitions within a base period. 

[sample.tar.gz](https://github.com/LinuxCNC/linuxcnc/files/9543190/sample.tar.gz)

andypugh, 2022-09-11:
This looks to be a divide by zero, probably on line 486. I will attempt to reproduce the problem some time next week.

samcoinc, 2022-09-11:
Thank you!   I will see if I can come up with a simple hal file that will reproduce the problem (without hardware)

andypugh, 2022-09-11:
If you are bisecting then can you just comment out that line, recompile and re-test? You don't need what it does if not using missing-index (yet).

samcoinc, 2022-09-11:
Ok - this is a short hal file that will crash it - it seems I had to put the sim encoder in a faster thread than the encoder to make it error - also needed to send it a sin wave to make it fail..  It will error seconds after loading.

loadrt threads name1=fastsimthread period1=10000 name2=encoderthread period2=100000 name3=servothread period3=1000000 
loadrt encoder num_chan=1
loadrt sim_encoder
loadrt siggen 

addf encoder.capture-position servothread
addf encoder.update-counters encoderthread
addf sim-encoder.update-speed servothread 
addf sim-encoder.make-pulses fastsimthread 
addf siggen.0.update servothread

net a sim-encoder.0.phase-A encoder.0.phase-A
net b sim-encoder.0.phase-B encoder.0.phase-B
net signal siggen.0.sine sim-encoder.0.speed

setp sim-encoder.0.ppr 500
setp encoder.0.position-scale 2000
setp siggen.0.amplitude 100

start

samcoinc, 2022-09-11:
q

> 

I remarked out that line

cntr->limit_dt += 0.1 * ((*(cntr->missing_teeth) + 0.5) * (delta_time / delta_counts))

so far no crash

andypugh, 2022-09-12:
It doesn't seem to crash for me, which is making the fix more difficult. 
I might have to try a preempt-rt build.

samcoinc, 2022-09-12:
well poop - I was using rt-preempt.   I just tried it again to make sure what I posted was what would make it crash

linuxcnc@linuxcnc-pc:~/linuxcnc-dev/src$ halcmd -kf
halcmd: loadrt threads name1=fastsimthread period1=10000 name2=encoderthread period2=100000 name3=servothread period3=1000000
Note: Using POSIX realtime
halcmd: loadrt encoder num_chan=1
halcmd: loadrt sim_encoder
halcmd: loadrt siggen
halcmd: 
halcmd: addf encoder.capture-position servothread
halcmd: addf encoder.update-counters encoderthread
halcmd: addf sim-encoder.update-speed servothread
halcmd: addf sim-encoder.make-pulses fastsimthread
halcmd: addf siggen.0.update servothread
halcmd: 
halcmd: net a sim-encoder.0.phase-A encoder.0.phase-A
halcmd: net b sim-encoder.0.phase-B encoder.0.phase-B
halcmd: net signal siggen.0.sine sim-encoder.0.speed
halcmd: 
halcmd: setp sim-encoder.0.ppr 500
halcmd: setp encoder.0.position-scale 2000
halcmd: setp siggen.0.amplitude 100
halcmd: 
halcmd: start
halcmd: rtapi_app: caught signal 8 - dumping core

andypugh, 2022-09-12:
I got it to happen with preempt-rt. I am not sure why it should make a difference, though.

andypugh, 2022-09-12:
Should be fixed by 9471c059d133f7d14363f6a7cdcf87a4de4666e6

closed 2022-09-12 by andypugh