LinuxCNC/linuxcnc #2048: hal_speaker: dlopen: /usr/lib/linuxcnc/modules/hal_speaker.so: cannot open shared object file: No such file or directory

ipsod, 2022-09-18:
## Here are the steps I follow to reproduce the issue:

 1.  Install Debian 12 and LinuxCNC according to guide here: https://gnipsel.com/linuxcnc/index.html
 2.  create stepper sim_inch config with wizard  (ran on terminal with command `linuxcnc`)
 3.  try to launch machine

## This is what I expected to happen:
not an error

## This is what happened instead:
```
LINUXCNC - 2.9.0~pre0
Machine configuration directory is '/home/self/linuxcnc/configs/by_interface.parport.stepper'
Machine configuration file is 'sim_inch.ini'
Starting LinuxCNC...
linuxcnc TPMOD=tpmod HOMEMOD=homemod EMCMOT=motmod
Note: Using POSIX realtime
Found file(lib): /usr/share/linuxcnc/hallib/core_stepper.hal
Found file(REL): ./sim_pinout.hal
hal_speaker: dlopen: /usr/lib/linuxcnc/modules/hal_speaker.so: cannot open shared object file: No such file or directory
./sim_pinout.hal:5: waitpid failed /usr/bin/rtapi_app hal_speaker
./sim_pinout.hal:5: /usr/bin/rtapi_app exited without becoming ready
./sim_pinout.hal:5: insmod for hal_speaker failed, returned -1
Shutting down and cleaning up LinuxCNC...
Note: Using POSIX realtime
LinuxCNC terminated with an error.  You can find more information in the log:
    /home/self/linuxcnc_debug.txt
and
    /home/self/linuxcnc_print.txt
as well as in the output of the shell command 'dmesg' and in the terminal

```

## It worked properly before this:
this is a fresh install

## Information about my hardware and software:
 * I am using this Linux distribution and version (often, shown by `lsb_release -a`):

>  No LSB modules are available.
> Distributor ID: Debian
> Description:    Debian GNU/Linux bookworm/sid
> Release:        testing
> Codename:       bookworm

 * I am using this kernel version (shown by `uname -a`):

>  Linux machine 5.19.0-1-rt-amd64 #1 SMP PREEMPT_RT Debian 5.19.6-1 (2022-09-01) x86_64 GNU/Linux

 * I am running ...

>  installation from debian 12 `apt install linuxcnc`

 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):

> 2.9.0~pre0+git20220906.02744cdef6-1 

 * I am using this user interface (GUI)

whatever is default with launch command `linuxcnc` 

 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): 

parallel port, but not sure that's applicable for sim

petterreinholdtsen, 2022-09-18:
[ipsod]
> hal_speaker: dlopen: /usr/lib/linuxcnc/modules/hal_speaker.so: cannot
> open shared object file: No such file or directory

So, what is the content of your /usr/lib/linuxcnc/modules/?  I lack
hal_speaker.so in my /usr/lib/linuxcnc/modules/ directory.  Not quite
sure why.
-- 
Happy hacking
Petter Reinholdtsen

ipsod, 2022-09-18:
`ls /usr/lib/linuxcnc/modules/`

```
5axiskins.so          estop_latch.so      match8.so             siggen.so
abs_s32.so            feedcomp.so         matrix_kb.so          sim_axis_hardware.so
abs.so                filter_kalman.so    max31855.so           sim_encoder.so
and2.so               flipflop.so         maxkins.so            sim_home_switch.so
axistest.so           gantry.so           mesa_7i65.so          sim_matrix_kb.so
bin2gray.so           gearchange.so       mesa_pktgyro_test.so  sim_parport.so
biquad.so             genhexkins.so       message.so            simple_tp.so
bitslice.so           genserkins.so       millturn.so           sim_spindle.so
bitwise.so            gray2bin.so         minmax.so             sphereprobe.so
bldc.so               hal_bb_gpio.so      modmath.so            spindle_monitor.so
blend.so              hal_gm.so           motmod.so             spindle.so
boss_plc.so           hal_lib.so          moveoff.so            stepgen.so
carousel.so           hal_parport.so      mult2.so              steptest.so
charge_pump.so        hal_pi_gpio.so      multiclick.so         streamer.so
clarke2.so            hal_ppmc.so         multiswitch.so        sum2.so
clarke3.so            histobins.so        mux16.so              supply.so
clarkeinv.so          hm2_7i43.so         mux2.so               thc.so
classicladder_rt.so   hm2_7i90.so         mux4.so               thcud.so
comp.so               hm2_eth.so          mux8.so               threads.so
constant.so           hm2_pci.so          mux_generic.so        threadtest.so
conv_bit_float.so     hm2_rpspi.so        near.so               timedelay.so
conv_bit_s32.so       hm2_spi.so          not.so                timedelta.so
conv_bit_u32.so       hm2_test.so         offset.so             time.so
conv_float_s32.so     homecomp.so         ohmic.so              tof.so
conv_float_u32.so     homemod.so          oneshot.so            toggle2nist.so
conv_s32_bit.so       hostmot2.so         or2.so                toggle.so
conv_s32_float.so     hypot.so            orient.so             ton.so
conv_s32_u32.so       ilowpass.so         pentakins.so          tpmod.so
conv_u32_bit.so       integ.so            pid.so                tp.so
conv_u32_float.so     invert.so           plasmac.so            tripodkins.so
conv_u32_s32.so       joyhandle.so        pumakins.so           tristate_bit.so
corexy_by_hal.so      knob2float.so       pwmgen.so             tristate_float.so
corexykins.so         latencybins.so      rosekins.so           trivkins.so
counter.so            lcd.so              rotarydeltakins.so    updown.so
dbounce.so            limit1.so           rotatekins.so         userkins.so
ddt.so                limit2.so           sample_hold.so        watchdog.so
deadzone.so           limit3.so           sampler.so            wcomp.so
debounce.so           lincurve.so         scale.so              weighted_sum.so
demux.so              lineardeltakins.so  scarakins.so          xhc_hb04_util.so
differential.so       linuxcnc            scope_rt.so           xor2.so
edge.so               logic.so            scorbot-kins.so       xyzac-trt-kins.so
encoder_ratio.so      lowpass.so          select8.so            xyzbc-trt-kins.so
encoder.so            lut5.so             serport.so
eoffset_per_angle.so  maj3.so             setsserial.so

```

`ls /usr/lib/linuxcnc/modules/hal*`

```
/usr/lib/linuxcnc/modules/hal_bb_gpio.so  /usr/lib/linuxcnc/modules/hal_parport.so
/usr/lib/linuxcnc/modules/hal_gm.so       /usr/lib/linuxcnc/modules/hal_pi_gpio.so
/usr/lib/linuxcnc/modules/hal_lib.so      /usr/lib/linuxcnc/modules/hal_ppmc.so
```

petterreinholdtsen, 2022-09-18:
I just noticed this in src/Makefile

  ifneq ($(BUILD_SYS),uspace)
  [...]
  obj-$(CONFIG_HAL_SPEAKER) += hal_speaker.o
  hal_speaker-objs := hal/drivers/hal_speaker.o $(MATHSTUB)
  [...]
  endif

I believe the effect is that the hal_speaker component is not built in
uspace mode.

I do not really know what the component need, perhaps it can not work
with the uspace flavor of linuxcnc?

-- 
Happy hacking
Petter Reinholdtsen

PR #2054 (2022-11-07): Enable hal_speaker driver also for uspace flavor
Switch from including <asm/io.h> to using <sys/io.h> which work better in user space.  Unsure if this will work outside x86 architectures.

Fixes #2048.
commit 39ffc82a (2022-11-07): Enable hal_speaker driver also for uspace flavor

Switch from including <asm/io.h> to using <sys/io.h> which work better
in user space.  Unsure if this will work outside x86 architectures.

Fixes #2048.
closed 2022-11-07 by petterreinholdtsen