LinuxCNC/linuxcnc #2052: debian/configure: Re-add support for RTAI

NTULINUX, 2022-09-19:
With Bullseye RTAI kernels available (4.19.257) 6f285604ac1a1b58b2d65d5904ffec3998a833ef should be at least partially reverted.

petterreinholdtsen, 2022-09-19:
[NTU]
> With Bullseye RTAI kernels available (4.19.257)
> 6f285604ac1a1b58b2d65d5904ffec3998a833ef should be at least partially
> reverted.

Perhaps you can identify which parts should be reverted and create a
pull request to reintroduce updated RTAI support?

--
Happy hacking
Petter Reinholdtsen

andrewheeler82, 2022-09-20:
I don't know if I understand this correctly, but officially there is no RTAI kernel from the Debian developers. The kernel 4.19.257 is from their development. That's why I think this change https://github.com/LinuxCNC/linuxcnc/commit/6f285604ac1a1b58b2d65d5904ffec3998a833ef was made. But I'm happy to be enlightened because I personally prefer RTAI.

NTULINUX, 2022-09-20:
> I don't know if I understand this correctly, but officially there is no RTAI kernel from the Debian developers. The kernel 4.19.257 is from their development. That's why I think this change [6f28560](https://github.com/LinuxCNC/linuxcnc/commit/6f285604ac1a1b58b2d65d5904ffec3998a833ef) was made. But I'm happy to be enlightened because I personally prefer RTAI.

There has never been an RTAI kernel from the Debian developers, however the ability to build LinuxCNC Debian packages with RTAI support has been an option for as long as I remember up until that commit. LinuxCNC would host their own RTAI kernel packages in their repository with corresponding LinuxCNC packages as well. The ability to switch between PREEMPT_RT and RTAI installations should and has been available for quite some time all using the package manager thanks to the LinuxCNC repository.

andrewheeler82, 2022-09-20:
many thanks for the information. My last RTAI kernel was Debian 7 32-bit.

NTULINUX, 2022-09-20:
Ah, I found them!

https://www.linuxcnc.org/dists/buster/base/binary-amd64/

I was looking in `2.8-rt` instead of `base`.

andrewheeler82, 2022-09-20:
Ok 👍🏻 nice.

NTULINUX, 2022-10-09:
Need help:

```
From 9b840b0c01f49022379a46e29a9a86e9d0f54062 Mon Sep 17 00:00:00 2001
From: Alec Ari <neotheuser@ymail.com>
Date: Sun, 9 Oct 2022 00:14:04 -0500
Subject: BROKEN: Revert "debian/configure: Drop outdated support for rtai and
 xenomai"

BROKEN
This reverts commit 6f285604ac1a1b58b2d65d5904ffec3998a833ef.
BROKEN

---
 debian/configure                     | 144 +++++++++++++++++++++++----
 debian/linuxcnc-uspace-rtai.files.in |   1 +
 2 files changed, 123 insertions(+), 22 deletions(-)
 create mode 100644 debian/linuxcnc-uspace-rtai.files.in

diff --git a/debian/configure b/debian/configure
index f5a9683549..1973b1cf0e 100755
--- a/debian/configure
+++ b/debian/configure
@@ -18,7 +18,22 @@ usage () {
 $P: Set up debian/ files to build for a particular kernel
 
 Usage:
-  $P [no-docs]
+    $P uspace [noauto | rtai | rtai=packagename] ...
+        Build for a userspace realtime systems or simulator
+
+    $P sim
+        Deprecated synonym for '$P uspace'
+
+    $P [kernel-version]
+        Build for the installed RTAI realtime kernel specified by
+        [kernel-version], for example "3.4.9-rtai-686-pae"
+
+    $P -r
+        Build for the currently running kernel for RTAI realtime only
+
+    $P -a
+        If the currently running kernel has realtime extensions, use it.
+        Otherwise use any available realtime kernel it can find.
 
   Extra options:
     no-docs
@@ -28,15 +43,48 @@ EOF
 
 cd "${0%/*}"
 
-if [ "$1" == "-h" -o "$1" == "-help" -o "$1" == "--help" ]; then
+if [ $# -eq 0 -o "$1" == "-h" -o "$1" == "-help" -o "$1" == "--help" ]; then
     usage
+    if [ $# -ne 1 ]; then
+        exit 1
+    fi
     exit 0
 fi
 
+if [ $1 = "-r" ]; then
+    set -- `uname -r`
+elif [ $1 = "-a" ]; then
+    KERNEL="$(uname -r)"
+    if [ ! -d "/usr/realtime-$KERNEL" ]; then
+        KERNEL=$(ls -1d /usr/realtime-* 2> /dev/null | cut -d - -f 2- | head -1)
+        if [ -z "$KERNEL" ]; then
+            echo "no realtime kernels found!"
+            exit 1
+        fi
[... 154 lines cut ...]
+else
+    cp linuxcnc-dev.files.in linuxcnc-dev.files
+    echo "$MODULE_PATH/Module.symvers" >> linuxcnc-dev.files
+fi
+
+subst rules.in > rules; chmod +x rules
 
 subst rules.in > rules
 chmod +x rules
@@ -190,4 +290,4 @@ fi
 
 
 rm -f ../build-stamp
-echo "successfully configured for '$DISTRIB_NAME'.."
+echo "successfully configured for '$DISTRIB_NAME'-'$TARGET$TARGET_EXTRA'.."
diff --git a/debian/linuxcnc-uspace-rtai.files.in b/debian/linuxcnc-uspace-rtai.files.in
new file mode 100644
index 0000000000..0b05b4f794
--- /dev/null
+++ b/debian/linuxcnc-uspace-rtai.files.in
@@ -0,0 +1 @@
+/usr/lib/libuspace-rtai.so.0
-- 
2.30.2
```

andypugh, 2022-10-10:
What help do you need?

NTULINUX, 2022-10-11:
I'm having trouble adding back in RTAI support. If you apply that patch, can you let me know at least how close I am and any pointers? I also don't know what the syntax is to call the debian configure script to build RTAI.

andypugh, 2022-10-11:
To configure for RTAI if you are booted into the RTAI kernel then it is 

./debian/configure -r 

I might find time to try your patch, but I am hardly even keeping up with personal emails at the moment.

petterreinholdtsen, 2022-11-15:
[Andy Pugh]
> To configure for RTAI if you are booted into the RTAI kernel then it is 
>
> ./debian/configure -r 
>
> I might find time to try your patch, but I am hardly even keeping up
> with personal emails at the moment.

Can it only be built after booting into a RTAI kernel?  If it can be
build by just installing some build dependency, we could get away with
adjusting .github/workflows/ci.yml to install this extra dependency from
somewhere (like we do with po4a) and then build the RTAI kernel.  I
assume it can not be tested, but we could test the build that way.

--
Happy hacking
Petter Reinholdtsen

andypugh, 2022-11-16:
I haven't tried, but my feeling is that it needs the kernel headers at the very least.

NTULINUX, 2022-11-16:
Yes, you need the headers. `-r` looks like it uses `uname -r` so you'd need to use the `$P [kernel-version]` one.

petterreinholdtsen, 2022-11-16:
[NTU]
> Yes, you need the headers. `-r` looks like it uses `uname -r` so you'd
> need to use the `$P [kernel-version]` one.

Can you tell exactly which package is needed, and how you call
d/configure and src/configure?  It would make it more obvious how to
adjust the github CI setup.

-- 
Happy hacking
Petter Reinholdtsen

petterreinholdtsen, 2022-11-16:
[Jérémie Tarot]
> I think Seb said it needs to load and unload kernel modules too, at
> least for testing, in the release manager/builbot thread

I think we all agree that testing is out of scope for a github build of
RTAI.  My goal is to get the build going to detect when the code is not
building.

-- 
Happy hacking
Petter Reinholdtsen

silopolis, 2022-11-16:
Le mer. 16 nov. 2022 à 12:24, petterreinholdtsen ***@***.***>
a écrit :

> [Jérémie Tarot]
> > I think Seb said it needs to load and unload kernel modules too, at
> > least for testing, in the release manager/builbot thread
>
> I think we all agree that testing is out of scope for a github build of
> RTAI. My goal is to get the build going to detect when the code is not
> building.

oh ok 🖐

NTULINUX, 2022-11-17:
If the build system issued `runtests` then it could make sure RTAI is working ok.

NTULINUX, 2024-07-04:
I was able to build LinuxCNC RTAI debian packages with my branch here:

https://github.com/NTULINUX/linuxcnc-rtai/tree/ntulinux/2.9-rtai

Took @andypugh 's `andypugh/2.9-rtai` branch and rebased it on top of the latest checkout of the `2.9` branch and it built. I wasn't able to test RTAI itself though because it panics on this laptop that I'm temporarily using.