LinuxCNC/linuxcnc #2055: Please revert: Removed the option of manually inserting the parallel port

openlunchbox, 2022-09-20:
## Here are the steps I follow to reproduce the issue:

 1. Run Stepconf wizard
 2. Start, create or modify a config, forward to "Parallel Port" config
 3. Parport Number:  0 or no port number is given for some LPT cards

## This is what I expected to happen:
I wish to enter the base address of the parallel port

## This is what happened instead:
I am only given a choice of 0 or maybe 1, hex addresses are no longer allowed
If the kernel does not detect and load the LPT driver for the LPT port, the user has to edit the config file manually with an editor
Some ports will not work when set to 0 or 1 

## It worked properly before this:
https://github.com/LinuxCNC/linuxcnc/commit/06a34258f8603b46614f2f47186b02a6e65edebc
Removed the option of manually inserting the parallel port
This commit broke this feature 

## Information about my hardware and software:

 Any version of LCNC since the above commit.
Several different LPT cards require manual edits of the base address register.

petterreinholdtsen, 2022-09-21:
Perhaps @nicokid can comment on this?  He removed the feature two years ago, according to git history.

openlunchbox, 2022-09-22:
Many LPT cards are not automagically identified so you have to enter the base address manually. By not allowing the user this ability the user can't test stepping in stepconf. The user has to edit the HAL file with an editor and then try to test by running LCNC.

c-morley, 2022-10-02:
https://github.com/LinuxCNC/linuxcnc/commit/2ffc0a11c8a8e25e3fec204bf6f0216ac7f35e2f
Needs testing

petterreinholdtsen, 2022-10-02:
[c-morley]
> https://github.com/LinuxCNC/linuxcnc/commit/2ffc0a11c8a8e25e3fec204bf6f0216ac7f35e2f
> Needs testing

Works.

<URL: https://github.com/LinuxCNC/linuxcnc/commits/fix-github-ci-pere >

-- 
Happy hacking
Petter Reinholdtsen

c-morley, 2022-10-04:
Pushed to master. Thank you.

closed 2022-10-04 by c-morley