LinuxCNC/linuxcnc #2060: Section 3 man pages not included in linuxcnc-uspace deb

petterreinholdtsen, 2022-09-22:
Thanks to PR #2059, I had a look at some of the manual pages in section 3 of the man page collection, only to discover that the debian package is missing all of these.  This show the problem:
```
% ls docs/man/man3/
EXPORT_FUNCTION.3rtapi          hal_start_threads.3hal            rtapi_is.3rtapi
hal_add_funct_to_thread.3hal    hal_stream.3hal                   rtapi_list.3rtapi
hal_bit_t.3hal                  hal_type_t.3hal                   rtapi_module_param.3rtapi
hal_bool.3hal                   hal_u32_t.3hal                    RTAPI_MP_ARRAY_INT.3rtapi
hal_create_thread.3hal          hal_unlink.3hal                   RTAPI_MP_ARRAY_LONG.3rtapi
hal_del_funct_from_thread.3hal  hm2_allocate_bspi_tram.3hm2       RTAPI_MP_ARRAY_STRING.3rtapi
hal_exit.3hal                   hm2_bspi_set_read_function.3hm2   RTAPI_MP_INT.3rtapi
hal_export_funct.3hal           hm2_bspi_setup_chan.3hm2          RTAPI_MP_LONG.3rtapi
hal_float_t.3hal                hm2_bspi_set_write_function.3hm2  RTAPI_MP_STRING.3rtapi
hal_get_lock.3hal               hm2_bspi_write_chan.3hm2          rtapi_mutex.3rtapi
hal_init.3hal                   hm2_pktuart_read.3hm2             rtapi_open_as_root.3rtapi
hal_link.3hal                   hm2_pktuart_send.3hm2             rtapi_outb.3rtapi
hal_malloc.3hal                 hm2_pktuart_setup.3hm2            rtapi_parport.3rtapi
hal_param_alias.3hal            hm2_tram_add_bspi_frame.3hm2      rtapi_pci.3rtapi
hal_param_bit_new.3hal          hm2_uart_read.3hm2                rtapi_print.3rtapi
hal_param_bit_newf.3hal         hm2_uart_send.3hm2                rtapi_prio.3rtapi
hal_param_float_new.3hal        hm2_uart_setup.3hm2               rtapi_prio_highest.3rtapi
hal_param_float_newf.3hal       intro.3hal                        rtapi_prio_lowest.3rtapi
hal_param_new.3hal              intro.3rtapi                      rtapi_prio_next_higher.3rtapi
hal_param_s32_new.3hal          MODULE_AUTHOR.3rtapi              rtapi_prio_next_lower.3rtapi
hal_param_s32_newf.3hal         MODULE_DESCRIPTION.3rtapi         rtapi_region.3rtapi
hal_param_u32_new.3hal          MODULE_LICENSE.3rtapi             rtapi_release_region.3rtapi
hal_param_u32_newf.3hal         PM_ROTATION_VECTOR.3              rtapi_request_region.3rtapi
hal_parport.3hal                rtapi_app_exit.3rtapi             rtapi_set_msg_level.3rtapi
hal_pin_alias.3hal              rtapi_app_main.3rtapi             rtapi_shmem.3rtapi
hal_pin_bit_new.3hal            rtapi_atomic.3rtapi               rtapi_shmem_delete.3rtapi
hal_pin_bit_newf.3hal           rtapi_bool.3rtapi                 rtapi_shmem_getptr.3rtapi
hal_pin_float_new.3hal          rtapi_byteorder.3rtapi            rtapi_shmem_new.3rtapi
hal_pin_float_newf.3hal         rtapi_clock_set_period.3rtapi     rtapi_slab.3rtapi
hal_pin_new.3hal                rtapi_delay.3rtapi                rtapi_snprintf.3rtapi
hal_pin_port_new.3hal           rtapi_delay_max.3rtapi            rtapi_stdint.3rtapi
hal_pin_port_newf.3hal          rtapi_device.3rtapi               rtapi_string.3rtapi
hal_pin_s32_new.3hal            rtapi_div_u64.3rtapi              rtapi_strlcpy.3rtapi
hal_pin_s32_newf.3hal           rtapi_div_u64_rem.3rtapi          rtapi_task_delete.3rtapi
hal_pin_u32_new.3hal            rtapi_exit.3rtapi                 rtapi_task_new.3rtapi
hal_pin_u32_newf.3hal           rtapi_firmware.3rtapi             rtapi_task_pause.3rtapi
hal_port.3hal                   rtapi_get_clocks.3rtapi           rtapi_task_resume.3rtapi
hal_ready.3hal                  rtapi_get_msg_level.3rtapi        rtapi_task_self.3rtapi
hal_s32_t.3hal                  rtapi_get_time.3rtapi             rtapi_task_start.3rtapi
hal_set_constructor.3hal        rtapi_gfp.3rtapi                  rtapi_task_wait.3rtapi
hal_set_lock.3hal               rtapi_inb.3rtapi                  undocumented.3hal
hal_signal_delete.3hal          rtapi_init.3rtapi                 undocumented.3rtapi
hal_signal_new.3hal             rtapi_io.3rtapi
% dpkg -L linuxcnc-uspace|grep man3
% 
```

smoe, 2022-12-27:
At least the one I tested from above list is in the -dev package.
```
dpkg -L linuxcnc-uspace-dev|grep man
```

petterreinholdtsen, 2022-12-27:
Steffen Möller ***@***.***> writes:

> At least the one I tested from above list is in the -dev package.
> ```
> dpkg -L linuxcnc-uspace-dev|grep man
> ```

Not quite sure what you are talking about here.  'dpkg -L
linuxcnc-uspace-dev|grep man|wc -l' on my send show 119 library man
pages in the dev package.  Do you expect them to be in the
linuxcnc-uspace package?  If so, why?

-- 
Happy hacking
Petter Reinholdtsen

smoe, 2022-12-28:
Maybe I misunderstood. You said the Debian package would be missing some man pages. And I have shown where I had found them, which was just the -dev package. I think they are fine in the -dev package.

petterreinholdtsen, 2022-12-29:
You are right.  This was just a misunderstanding on my end.

closed 2022-12-29 by petterreinholdtsen