LinuxCNC/linuxcnc #2061: Several components mentioned in provided config is missing in linuxcnc-uspace deb

petterreinholdtsen, 2022-09-22:
Via issue  #2048 and the associated pull request, I discovered that the provided configuration references several components that are not in the generated linuxcnc-uspace deb.  Should any of these be added to the deb?  Note, #2054 try to enable hal_speaker.

``
% for m in $(grep -r ^loadrt configs|awk '{print $2}'|sort -u|grep -v '\['|grep -v '::'); do if [ ! -f /usr/lib/linuxcnc/modules/$m.so ]; then echo $m; fi; done
freqgen
hal_ax5214h
hal_motenc
hal_speaker
hal_stg
hal_vti
pluto_servo
pluto_step
%
```

smoe, 2022-12-27:
I would have expected all components from he current source tree to be distributed via the .deb. Anything non-functional in the linuxcnc source tree should somehow be flagged as non-functional - and nonetheless be distributed :)

SebKuzminsky, 2023-01-16:
Looks like several of those are only built as kernel modules, for our RTAI configuration.  That might be because they (wrongly or rightly) use kernel-internal APIs, or because of a (hypothetical) bug in our build system that incorrectly limits the building of those drivers.

On the other hand...  I bet most or all of the hardware that those drivers are for can no longer be bought, and apparently any existing users don't care about upgrading.  It may be best to delete the drivers for obsolete hardware that no one cares about an more.

andypugh, 2023-01-16:
There may be users running the latest LinuxCNC and RTAI and this hardware. It's hard to be sure. 

But AFAIK Pluto never really worked, I think we can drop that.

SebKuzminsky, 2023-01-23:
As of a week ago, @cradek was running a Pluto on a machine he used frequently.  But he's probably not representative of our userbase 🤣