LinuxCNC/linuxcnc #2085: Gcode examples should have a heading of it's own

c-morley, 2022-10-08:
I think the examples write up is difficult to find.
I would like to add some docs on the support o code subroutines use with qtvcp widgets and almost made a new page.
If someone already knows how to program gcode, they might miss the example/support codes.

silopolis, 2022-10-09:
Le sam. 8 oct. 2022 à 20:52, c-morley ***@***.***> a écrit :

> I think the examples write up is difficult to find.
> I would like to add some docs on the support o code subroutines use with
> qtvcp widgets and almost made a new page.
> If someone already knows how to program gcode, they might miss the
> example/support codes.
>

I'm afraid my english betrays me here... Can you please rephrase/explain
please ?
Sorry

c-morley, 2022-10-09:
Lol no I think it was my English...

In the docs heading' Gcode' there is a sub heading of example geode. This is from memory, I'm at work.
I think we should have a separate main heading fie 'Gcode examples' to make it easier to find.

I wanted to add some descriptions of the example code that qtvcp uses and didn't realize there was a place for it.

revjmoney, 2026-11-09:
I'd like to pick this up. Is it still wanted?

Today `examples/gcode.adoc` is one item inside "G-code Programming", both in the HTML index (`docs/src/index.tmpl`) and in the PDF (`docs/src/Master_Documentation.adoc`, included at leveloffset 2). My plan:

- `index.tmpl`: a new collapsible section "G-code Examples" right after "G-code Programming", with the existing `examples/gcode.html` link moved into it (room for the QtVCP support-code page @c-morley mentioned).
- `Master_Documentation.adoc`: a new `= G-code Examples` part after "G-code Programming" that includes `examples/gcode.adoc`, so the PDF matches.
- `docs.xml`: move `examples_gcode` to match, for the prev/next buttons.
- No page content or anchors change, so existing links keep working.

Should this go to master or 2.9? It's a docs-only layout change, but it moves an index entry, so I'd guess master.

BsAtHome, 2026-11-09:
Yes, please target master. A lot has changed (for the better) in the docs in master.