LinuxCNC/linuxcnc #2093: Divide By Zero Check needed in glnav.py

satiowadahc, 2022-10-14:
Might look at it later, leaving here for reminding myself to do so.

My setup: Custom UI, Using gcodegraphics from qtvcp. Running 2.9.

```
  File "/usr/lib/python3/dist-packages/qt5_graphics.py", line 546, in set_current_view
    return getattr(self, 'set_view_%s' % self.current_view)()
  File "/usr/lib/python3/dist-packages/glnav.py", line 418, in set_view_z
    self.set_eyepoint_from_extents(size[0], size[1])
  File "/usr/lib/python3/dist-packages/glnav.py", line 253, in set_eyepoint_from_extents
    ztran = max(2.0, e1, e2 * w/h) ** 2
ZeroDivisionError: float division by zero
```

Relevant code in master:
```python
    def set_eyepoint_from_extents(self, e1, e2):
        """Set how far the eye is from the position we are looking
        based on the screen width and height of a subject."""
        w = self.winfo_width()
        h = self.winfo_height()

        ztran = max(2.0, e1, e2 * w/h) ** 2
        self.set_eyepoint(ztran - self.zcenter)
```

andypugh, 2022-10-14:
Checking for divide by zero is a good idea, but there is also the question of why winfo_height is returning zero? 

ztran = max(2.0, e1, e2 * w/(max(h, 1e-10))) ** 2 would be my simplistic fix. 

But there might be a good argument for not recalculating ztran at all if h == 0.

satiowadahc, 2022-10-15:
Two options I can see are either to set h to an epsilon value or not update ztran.

But it crashed LCNC with only the info in the terminal, at the very least it should notify the user or handle the exception.

andypugh, 2022-10-15:
@c-morley Rather than me bodge this without really knowing what e1 and e2 are, can you look at this?

c-morley, 2022-10-16:
you bet.

c-morley, 2022-10-16:
try:
exception:
would be easy and clear but I do wonder if there was some other error that is being masked.

c-morley, 2022-10-16:
Is the error message longer?

satiowadahc, 2022-10-16:
Try excepts are expensive in python compared to an if statement.

I was in the middle of debugging other things and missed copying this one.

c-morley, 2022-10-17:
I was thinking to add the try up in the qt_graphics code, which doesn't get called as often.
Trying to catch the error without crashing.
My guess is still that something else caused the error.
The error means the window has a height or width of zero - which doesn't seem right.
You can get all kinds of weird errors reported after a previous error makes things not initialize properly.

BsAtHome, 2026-05-10:
Is this error still present?

grandixximo, 2026-10-03:
Yes, still present in master. Opened a fix: #4620

closed 2026-10-11 by grandixximo