LinuxCNC/linuxcnc #2095: RTAI: LinuxCNC keeps saying to run `halrun -U`

NTULINUX, 2022-10-15:
## Here are the steps I follow to reproduce the issue:

 1. Compile LinuxCNC against RTAI
 2. Run latency-histogram, latency-test or LinuxCNC (I only used run-in-place for this test)
 3. Try to re-run any of those programs

## This is what I expected to happen:

Application would start back up

## This is what happened instead:

RTAI is already running, run `halrun -U` which doesn't work. I have to manually run `rmmod rtai_sched rtai_hal rtai_math`

## It worked properly before this:
(If the behavior changed after making a particular change in hardware or
software, describe the change you think is responsible.  E.g., "after upgrading
from LinuxCNC 2.7.3 to 2.7.4")

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`): Debian 11/Bullseye
 * I am using this kernel version (shown by `uname -a`): 4.19.2XX
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [X] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`): master checkouts from the last few weeks
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): Any
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): N/A

NTULINUX, 2022-10-15:
Yes I believe so. I understand that commit is useful, and I'm glad it's there, but `halrun -U` when manually invoked should do `rmmod rtai_sched rtai_hal rtai_math`.

andypugh, 2022-10-15:
I am not sure that halrun -U has ever worked properly, as my experience is that the RTAI modules are all mutually dependent and rmmod doesn't work either. 
If rmmod is working for you but halrun -U isn't then I am not entirely sure what it going on. 
Possibly @jepler has some ideas?

NTULINUX, 2022-10-15:
The testsuite module info is as such:

https://github.com/NTULINUX/RTAI/blob/master/testsuite/runinfo

rtai_hal.ko
rtai_sched.ko
rtai_mbx.ko
rtai_msg.ko

rmmod works as long as you do it in reverse order, rtai_math being the first one.

andypugh, 2022-10-15:
It doesn't look like the code tries to unload in any particular order, it just iterates through all components. 
https://github.com/LinuxCNC/linuxcnc/blob/master/src/hal/utils/halcmd_commands.cc#L1294

jepler, 2022-10-15:
Apparently, I broke this in https://github.com/LinuxCNC/linuxcnc/pull/1932

The check "is any kernel module that matches rtai loaded" as a prompt to "halrun -U" is now incorrect and needs to be removed. Only the exit status of `realtime status` should be used to determine if realtime is loaded, as halrun does:
```shell
if $REALTIME status > /dev/null; then
    echo "$THIS: Realtime already running.  Use 'halrun -U' to stop existing realtime session." 1>&2
    exit 1
fi
```
Other tests, such as those in latency-histogram, should not be used:
```
    if [catch {exec lsmod | grep rtai} msg] {
      # puts ok_to_start_rtai    
    } else {
      popup "RTAI is already running, (try: \$ halrun -U)"
      exit 1
    }
```

It looks like this affects latency-histogram and latency-plot at least are affected by this.

I do not believe 'halrun -U' should unload these modules. Absent new information, for the reasons given in #1932 we should never unload them; it just crashes the system (albeit with low probability).

jepler, 2022-10-15:
I was not able to reproduce your finding that the main 'linuxcnc' script was affected by this problem. However,  I filed #2097 to remove the checks I view as erroneous from the latency scripts.

I may need more testing or I may be wrong about the ability to "safely" leave rtai_sched loaded when using kmod+rtai. I can repeatedly start linuxcnc with axis (well, 4 times) but >50% of the time starting latency-histogram again hangs.

If you want to revert the part where we unload this for rtai, I won't stand in the way. It'll continue preventing us from testing kmod+rtai on buildbot, but I believe uspace+rtai is the way forward. (uspace+rtai is achieved by having rtai set up but using. `./configure --with-realtime=uspace`)

jepler, 2022-10-15:
`tests/threads.0` under kmod+rtai shows a similar problem: The threads appear to be loaded, but the thread code never actually executes, which can be seen by the thread time of `halcmd show thread` remaining at zero. Manually unloading `rtai_sched` between runs stops this failure mode.

All of my earlier testing concentrated on the single-thread case (tests/abs.0)

uspace+rtai gets regular _failures_ (like `line 937: got 0, expected 10 or 1`) indicating that rate monotonic scheduling is not being provided. but it doesn't hit this problem

so, in my opinion, restoring `MODULES_UNLOAD="hal_lib rtapi $MODULES_UNLOAD"` in the `(*rtai*)` case in realtime.in would be a sensible way to proceed, since it fixes the regression in kmod+rtai, and as this combination isn't buildbot-tested the fact that it has low probability full system freezes as a consequence, well, people who use kmod+rtai can work with upstream since we have a linuxcnc-free reproduction script at https://mail.rtai.org/pipermail/rtai/2020-June/028122.html

jepler, 2022-10-15:
This has got to be a pointer towards what's wrong... I must have seen a similar problem in 1-thread tests and found a nonsensical "solution" by chance. But this didn't carry through to 2-thread situations.

>    I couldn't determine why, but calling rt_free_timers made the     2nd and successive runs fail.  The effect of rt_free_timers should be     "undone" by rt_start_timers calling rt_request_timers, from what I could     see. But it wasn't.

NTULINUX, 2022-10-16:
I could be wrong, I thought this happened with main LinuxCNC as well.

I don't think https://github.com/LinuxCNC/linuxcnc/pull/1932 should be reverted, but if RTAI is already loaded, latency tests shouldn't care and just load themselves anyway. RTAI doesn't need to be **unloaded**, but once they're **loaded**, a check to see **IF** they're **loaded** in order to determine whether or not they should be **unloaded**, should be removed.

NTULINUX, 2022-10-18:
https://github.com/LinuxCNC/linuxcnc/pull/2097 fixes this issue, thanks!

NTULINUX, 2022-10-20:
SMASHIN DAH BUGS!

PR #2097 (2022-10-20): presence of rtai modules does not indicate realtime is running
I hope this Closes #2095 though there may be other instances of incorrect checking for modules matching "rtai" being loaded.