LinuxCNC/linuxcnc #2111: Gmoccapy handler

daiman444, 2022-10-24:
# Here are the steps I follow to reproduce the problem:

1. add pyngcgui to the [DISPLAY] section of the *.ini file for the gmoccapy configuration.
2. Run Gmoccapy configuration, do homing,
3. I press the "open touch off button list" button and press the "edit offsets" button: the "show user tabs" button becomes inactive.
4. after editing the offsets, press the "edit offsets" button again to exit this mode

## This is what I expected:

The "show user tabs" button becomes sensitive again.

## Instead, this happened:

the "show user tabs" button does not become sensitive again.

Problem found in "/usr/bin/gmoccapy" method "on_tbtn_edit_offsets_toggled" and part of it:

        if self.widgets.tbtn_user_tabs.get_sensitive():
            widgetlist.append("tbtn_user_tabs")
            
If replaced with:

        if self.get_ini_info.get_embedded_tabs()[2]:
            widgetlist.append("tbtn_user_tabs")
            
That all works great. Proven on several machines.

## Information about my hardware and software:
LinuxCNC_2.8.4

antoxa-kylibin, 2022-10-25:
I also faced this problem. But together we managed to solve this problem. I also tested this method on my machines, it really works correctly.

hansu, 2022-10-31:
Your patch is not working in every scenario. E.g. if you have only one embedded tab which is not in user tabs, it will activate the user tabs button after togging the edit offsets-button.
I provided a fix in #2120.

charlestonxorg, 2023-01-09:
X

PR #2120 (2022-11-07): gmoccapy: fixed sensitizing of user tab button (#2111)
The user tabs button was disabled after toggling the "edit offsets" button. 

Fixes #2111
closed 2023-01-10 by hansu