LinuxCNC/linuxcnc #2116: PID tuning chapter in documentation should explain use of FF1 and other parameters

petterreinholdtsen, 2022-10-30:
The chapter in the documentation on PID tuning, <URL: http://linuxcnc.org/docs/html/motion/pid-theory.html >, only talk about adjusting P, I and D, but nothing about adjusting FF1 like the recipe on <URL: https://wiki.linuxcnc.org/cgi-bin/wiki.pl?Tuning_LinuxCNC/HAL_PID_Loops >.  Perhaps some of the information from the wiki recipe should be added to the PID Tuning chapter?

hansu, 2022-10-30:
Maybe we can add the information from https://gnipsel.com/linuxcnc-old/tuning/servo.html ?

alangibson, 2024-01-29:
https://forum.linuxcnc.org/forum/39-pncconf/26860-tuning-values

petterreinholdtsen, 2024-01-30:
Both seem to have information that would fit well in the pid tuning
instructions in that chapter. :)

-- 
Happy hacking
Petter Reinholdtsen

pcw-mesa, 2026-10-26:
One might add that FF0 is useful for PID loops where the error and PID output are the same units
A common example would be velocity loops (say for spindle speed control with velocity feedback)