LinuxCNC/linuxcnc #2121: Docs: Update updating instructions

hansu, 2022-10-31:
This page needs a bit rework, as updating from 2.7 to 2.8 is a bit out of date:
https://linuxcnc.org/docs/2.9/html/getting-started/updating-linuxcnc.html#_upgrade_to_the_new_version

The chapter "3. Updating Configuration Files (for 2.8.x)" I would keep if someone wants to update from 2.7 to 2.9.

SebKuzminsky, 2022-10-31:
> The chapter "3. Updating Configuration Files (for 2.8.x)" I would keep if someone wants to update from 2.7 to 2.9.

We've always recommended upgrading "one version at a time" rather than skipping right to the newest release, mostly (I believe) in order to simplify the upgrade instructions (i.e. so that we don't have to reproduce the 2.7-to-2.8 upgrade instructions in all future versions).

So I suggest deleting all the old 2.7-to-2.8 contents of section 3 and replacing them with just whatever is needed to go from 2.8 to 2.9.

jethornton, 2022-10-31:
Yes, in the past if the upgrade was from 2.5 to 2.6 it was assumed the 
user would use the 2.6 documents to upgrade.

JT

hansu, 2022-11-01:
Okay but i think we should at least add a comment to upgrade in steps.
If I were a just a user and found a LinuxCNC 2.6 system, I would directly upgrade to 2.8. I would ask myself why I should put extra work in an intermediate step.

andypugh, 2022-11-01:
I think if I found a 2.6 system I would backup the linuxcnc directory and fresh install 2.8 and a new OS.

hansu, 2022-11-02:
Yes that's what I mean. You would just take the config from 2.6 and try to run it in 2.8, not installing a 2.7 in between.