LinuxCNC/linuxcnc #2125: Halscope crash bug

pcw-mesa, 2022-11-03:
Halscope in master has crashed multiple times when selecting the # of samples pane.
What is bad is that you must exit LinuxCNC to be able to run halscope again as you get
this error if you try to re-launch halscope from LinuxCNC:

HAL: ERROR: duplicate component name 'halscope'
SCOPE: ERROR: hal_init() failed

SebKuzminsky, 2022-11-03:
I just spent a bunch of time in Halscope (on Debian Bullseye and Debian Bookworm), including changing the number of samples, and did not notice any problems.  What platform did you see this on?

pcw-mesa, 2022-11-03:
X86, Mint 20.3
The times I have seen this happen are when halscope has been running for a long time (days)
I

pcw-mesa, 2022-11-04:
Thanks, that will help.
BTW here's a backtrace, not sure if it of any use:

#0  __strlen_avx2 () at ../sysdeps/x86_64/multiarch/strlen-avx2.S:65
#1  0x00007f930508ee95 in __vfprintf_internal
    (s=s@entry=0x7ffe1d515be0, format=format@entry=0x560f98f565d9 "%s\n%s", ap=ap@entry=0x7ffe1d515d60, mode_flags=mode_flags@entry=2) at vfprintf-internal.c:1688
#2  0x00007f93050a211a in __vsnprintf_internal (string=0x7ffe1d535330 "", maxlen=<optimized out>, 
    maxlen@entry=94, format=0x560f98f565d9 "%s\n%s", args=args@entry=0x7ffe1d515d60, mode_flags=mode_flags@entry=2)
    at vsnprintf.c:114
#3  0x00007f9305143fd1 in ___snprintf_chk
    (s=s@entry=0x7ffe1d535330 "", maxlen=maxlen@entry=94, flag=flag@entry=1, slen=slen@entry=94, format=format@entry=0x560f98f565d9 "%s\n%s") at snprintf_chk.c:38
#4  0x0000560f98f5205d in snprintf (__fmt=0x560f98f565d9 "%s\n%s", __n=94, __s=0x7ffe1d535330 "")
    at /usr/include/x86_64-linux-gnu/bits/stdio2.h:67
#5  draw_waveform (chan_num=<optimized out>, highlight=highlight@entry=1) at hal/utils/scope_disp.c:894
#6  0x0000560f98f52a82 in refresh_display () at hal/utils/scope_disp.c:258
#7  0x00007f9305b4f5ef in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#8  0x00007f9305af7ec3 in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#9  0x00007f9305491a56 in  () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#10 0x00007f93054afdf1 in g_signal_emit_valist () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0

SebKuzminsky, 2022-11-04:
The backtrace is helpful.  Looks like the `chan` data structure is invalid or has unexpected contents somehow.  Were you doing anything with halscope when it crashed?  Or was it doing something by itself, like running a Rolling trigger?

pcw-mesa, 2022-11-04:
It had been running a rolling trigger for a day or so and 
crashed when I pressed the samples button.

SebKuzminsky, 2023-01-27:
@jepler's fix for @cradek's arm64 halscope crash is now in master, give it a try and see if it fixes this problem too.

jepler, 2023-01-27:
based on this traceback I don't think it's fixed by my other change.

jepler, 2023-01-27:
well, maybe it could be the same thing, since it's also trying to draw the "selected" trace when it crashes. When I first started looking at it, I thought (based on line numbers) that maybe `format_scale_value` was implicated, but now I don't think so; the line numbers are different now than the version where the problem was reported.