LinuxCNC/linuxcnc #2129: Open discution for make something universal with axisui hal pin

alkabal, 2022-11-06:
Hi

I like to open a discussion about dedicated hal pin from axis

https://github.com/LinuxCNC/linuxcnc/blob/348b0d196d9fad662472589cc9882985732796d1/src/emc/usr_intf/axis/scripts/axis.py#L3888

I use some of this without any issue but i would like to speak about something similar for all GUI.

Maybee imo rhe best approach can be to detach this pin from GUI and replace by something attached to a more generic name like gui.xxxx

I hope this can be something interesting for all.

SebKuzminsky, 2022-11-06:
[halui](http://linuxcnc.org/docs/devel/html/gui/halui.html) has much of this functionality.

c-morley, 2022-11-06:
Halui has its own shortcoming when it comes to jogging controls.

But I think the premise of your idea is to decide a minimuim standard set of control pins and add them to all guis. It's actually easy to change the base name of pins so then just the standard base name has to be picked.

alkabal, 2022-11-06:
Thanks for reply

I appologize for unprecision for my own this is for some signal that are not availble in halui other GUI :
    comp.newpin("notifications-clear",hal.HAL_BIT,hal.HAL_IN)
    comp.newpin("notifications-clear-info",hal.HAL_BIT,hal.HAL_IN)
    comp.newpin("notifications-clear-error",hal.HAL_BIT,hal.HAL_IN)
    comp.newpin("resume-inhibit",hal.HAL_BIT,hal.HAL_IN)
    comp.newpin("error", hal.HAL_BIT, hal.HAL_OUT)
    comp.newpin("abort", hal.HAL_BIT, hal.HAL_OUT)

I use this deeply in some of my component and now this is difficult to test other GUI and also for a simpler HAL config file yes i really like the idea to have all the same name and all available.

c-morley, 2022-11-07:
Can you describe the uses of the non obvious ones?

c-morley, 2022-11-07:
The error notification communication in linuxcnc is super restrictive and really needs a rethink.

c-morley, 2022-11-07:
> Can you describe the uses of the non obvious ones?

I found it here:
http://linuxcnc.org/docs/devel/html/gui/axis.html#_user_hal_pins

alkabal, 2022-11-07:
https://github.com/LinuxCNC/linuxcnc/pull/1161
https://github.com/LinuxCNC/linuxcnc/pull/1024

silopolis, 2022-11-07:
Le dim. 6 nov. 2022 à 19:54, alkabal ***@***.***> a écrit :

> I use some of this without any issue but i would like to speak about
> something similar for all GUI.
>

I like that! 🙂

c-morley, 2022-11-07:
Can you explain any of the non obvious things you are using this for. For instance I suspect you are using the error or abort pin for another program (not just axmessage)

alkabal, 2022-11-07:
yes exactly this is for pass axis button stop action to other component, without that i have many time unpredictable state using macro

For error this allow to lock some function in other compoenent if we have error message displayed (need to be read and closed before continuing)

used here : 
https://github.com/alkabal/PSNG_V3/blob/064cf85c7a045b0505b6190a99d944d095808358/Your_Config_Folder/nc_subroutines/psng/python/base.py#L152

I have identified 3 way for stopping program that need to be managed if you use some macro from things like python
elif "TRUE" in str(abort_axisui) or "TRUE" in str(abort_halui) or "TRUE" in str(stop_halui):

c-morley, 2022-11-08:
Some related ramblings:

resume-inhibit seems an odd name for pause/resume don't you think?

Why do we need separate clear-notification pins for different type notifications?
In qtvcp you can close-but-remember the current shown messages or erase all the held messages or cycle through the held messages, which seem more useful in practice.

Why do you separate abort and error pins? Doesn't either stop your program? How does your program use them differently?

Curious why you watch both halui and axis ui for abort - or is that a work around?
In qtvcp I found sucsess by having abort send a message that can be picked up.
http://linuxcnc.org/docs/devel/html/remap/remap.html#_error_handling_dealing_with_abort

I humbly suggest hal.get_value(some.pin) is cleaner and probably faster the using popen (had a quick look at some of your code)

alkabal, 2022-11-08:
Hello

Yes you right resume-inhibit is a confusing name.

F or historical reason (previous psng) we have choose to auto clear GUI only for warn and regular message and force user to read and close error message manually.

Separate error and abort pin allow to know and display correct message to user from what is happened 
abort pin from axis is used if user push the stop button when a macro is started, without internal detection of this button my code is stoped in unpredictable states. So in reality this hal pin is a stop button action more than a abort.

I watch all due to different source of action but possible that is a work around because axis does not propagate correctly the stop button action.

For popen i use it for keep free of use pin for other hal config but if your suggestions work yes I agree it can be more clean and faster.

Thanks for interest

c-morley, 2022-11-12:
sorry you auto clear messages - could you elaborate on that, I guess mostly why?

Yes abort is not propagated through stat unfortunately. 
https://github.com/LinuxCNC/linuxcnc/issues/2013
The work around is to use on_abort.ngc to force propagation to the main GUI.

What generic pin base name?     uicore.some-pin?

alkabal, 2022-11-13:
Hello

Thanks for pointing that, you are right they are some issue that i have discovered until stress test and the error and abort pin from axis help a lot for doing a correct workaround.

I use a big amount of gcode message (maybe too much) so if message is only information they are auto cleared if you use other function for take the place for new message.

If this is a error message other functions are locked and ask user to read and close error before trying again 

I have do my best for use only gcode message for make it work with all gui (in the past I used machinekit and this is the only way finded for have message from remote/local interface, so I have kept the concept)

I use the on abort stuff but If I remember correctly this is not enough with the axis stop button.(sorry I’m unsure but I Hope i haven’t not spend time for something useless lool)

For name I have no real advice : uicore or guihal or other this is not the most important.

Thanks a lot for interest

alkabal, 2022-11-13:
At same time a thousands of thanks for suggestion about using « hal.get_value(some.pin) » that work fine !

I’m close to finish porting the code to 2.9/master, mostly GUI need rework due to spin button changed style (really better style) I think a lack of knowledge make it difficult for me to use glade without broking the code.

c-morley, 2022-11-14:
> if you use other function for take the place for new message.

Sorry I don't understand this part. You clear messages and replace them? or are you clearing messages so you can put up a new one? I guess I don't understand why you add messages just to have a separate program ignore them.
I'm just interested in what you are doing and why.

The on_abort.ngc idea. 
In AXIS we could have it look for the abort message and then set the pin. This has the advantage that any abort from anywhere will be caught.
It has the disadvantage that it requires the configuration to include the on_abort.ngc file.

Really linuxcnc should always send a message on abort through the error channel.
And maybe linuxcnc should broadcast the error channel using ZMQ or something

Anyways thanks for talking me through your thoughts.
I think can add the pins to qtvcp - well I mean make them work -(I can definitely add them.)

In qtdragon, we play a bit of a trick with the error channel, for similar reasons to you.
To have the qtdragon responsive during probing, we use a subprogram in a different context.
Of course that program needs to monitor the error channel which would race with qtdragon.
In our case the we block qtdragons error polling while the subroutine is running, then block the subroutine's error polling when not probing.
I guess we could do that with other GUIs too using HAL pins...
 We can also send out status messages from the subprogram using standard input.

alkabal, 2022-11-15:
i apologise allways difficult for me to explain what i think lool

yes i like message to clear all informative message each time i start a new function or at some place in the function for have only most usefull and recent message displayed. I think you can consider my code is too much verbous but usefull for me for debugging 

Too much time lost for debbug my own error but now it's close to be like i wan't it
I consider now my setup is consistant with correct tool offset from table, correct managing of offset from probescreen, machine with variable axis limit

If message is a error message i'm forced to close this one manually before continuing

alkabal, 2022-11-15:
Is qtdragon  now possible with RPI 4 Linuxcnc 2.9 ? (i like to test your GUI from long time)

At this moment with the axis patch for error and abort pin i think all work fine, this stuff come from Gmoccapy so i hope other people not have problem like we have encountered (mostly due to use too amount of macro imo) ??