LinuxCNC/linuxcnc #2132: Converting PSNG to Python3/Gtk3 (Possible bug with gladevcp combi_dro Master)

alkabal, 2022-11-08:
Hello

I run a fresh and updated bullseye instal with run in place linuxcnc from https://forum.linuxcnc.org/9-installing-linuxcnc/39779-rpi4-raspbian-64-bit-linuxcnc?start=150#253623

Trying to convert PSNGV3 to Python/Gtk3 i get some error with homing status

Traceback (most recent call last):
```
  File "/home/linuxcnc/dev/emc/lib/python/gladevcp/combi_dro.py", line 489, in _homed
    self.homed = self.status.homed[self.joint_no]
AttributeError: 'Combi_DRO' object has no attribute 'joint_no'Traceback (most recent call last):
  File "/home/linuxcnc/dev/emc/lib/python/gladevcp/combi_dro.py", line 509, in _homed
    self.homed = self.status.homed[self.joint_no]
AttributeError: 'Combi_DRO' object has no attribute 'joint_no'
```

https://github.com/LinuxCNC/linuxcnc/blob/348b0d196d9fad662472589cc9882985732796d1/lib/python/gladevcp/combi_dro.py#L489

At this moment i don't know if this is related to Linuxcnc master branch or related to my PSNGV3 python code.

I do not have this problem before when use buster and Python/Gtk2 (code for this parts is unchanged)

Any suggestion are welcome.

c-morley, 2022-11-14:
seems you need to call this function explicitly :
`    def set_joint_no(self, joint):
        '''
        changes the joint, not the joint number. This is handy for special
        cases, like Gantry configs, i.e. XYYZ, where joint 0 = X, joint 1 = Y1
        joint 2 = Y2 and joint 3 = Z, so the Z axis can be set to joint_number 2
        giving the axis letter Z and joint 3 being in this case the corresponding
        joint, joint 3 instead of 2
        '''
        self.joint_no = joint
`
maybe gmoccapy code could give you a clue

alkabal, 2022-11-14:
Hello

I need to check that more deeply but I haven’t this problem using 2.8 something is changed about this stuff in 2.9 ?

alkabal, 2022-11-28:
Hi

Lol i think i don't undersnatd what i need to do.

I find only 3 dro occurence in glade file from psng and nothing in python code

<object class="Combi_DRO" id="combi_dro_X">

hansu, 2022-11-28:
May I ask what is PSNG?

alkabal, 2022-11-28:
> May I ask what is PSNG?

Probe screen next generation
https://github.com/linuxcnc-probe-screen/probe-screen-ng (seem to be no more maintened by Kiall and i haven't the access...)
https://github.com/alkabal/PSNG_V3 actual is for 2.8 (i need to upload the 2.9 attemps)

alkabal, 2022-11-28:
You mean i need to do somethink like that in the python code ?

https://github.com/LinuxCNC/linuxcnc/blob/e8646b79ca87c18f0acb6ef4e2d45ab772f1202d/src/emc/usr_intf/gmoccapy/gmoccapy.py#L612

petterreinholdtsen, 2022-11-28:
[alkabal]
> Probe screen next generation
> https://github.com/linuxcnc-probe-screen/probe-screen-ng (seem to be
> no more maintened by Kiall and i haven't the access...)

If that repo is unmaintained, what about make it fit to be included in
the LinuxCNC main repo and get it under the community umbrella?

-- 
Happy hacking
Petter Reinholdtsen

alkabal, 2022-11-28:
For my own : i work on something more advanced and flexible so i have no real interest to continu maintaining this old version (too many difference).

I think new feature are only accepted in 2.10 and this repo is for 2.8 only.

In all way decision related to what is proposed to include in Linuxcnc is unrelated to me lol.

alkabal, 2022-12-02:
master branch updated with 2.9/2.10 support for GTK3/Python3

But for the bug reported in this issue for now i do not understand what i need to do, i think the only real problem is color change for DRO related to Homed/Not Homed.

Any suggestion welcome.

alkabal, 2022-12-02:
Other things that need to be reworked but i'm actually not able to do myself is allow inteface to be resizable

c-morley, 2022-12-03:
> Hi
> 
> Lol i think i don't undersnatd what i need to do.
> 
> I find only 3 dro occurence in glade file from psng and nothing in python code
> 
> `<object class="Combi_DRO" id="combi_dro_X">`

from gmoccapy.py:

`
        # we build one DRO for each axis
        self.dro_dic = {}
        for pos, axis in enumerate(self.axis_list):
            joint = self._get_joint_from_joint_axis_dic(axis)
            dro = Combi_DRO()
            dro.set_joint_no(joint)
            dro.set_axis(axis)
            dro.change_axisletter(axis.upper())
            dro.show()
            dro.set_property("name", "Combi_DRO_{0}".format(pos))
            dro.set_property("abs_color", self._get_RGBA_color(self.abs_color))
            dro.set_property("rel_color", self._get_RGBA_color(self.rel_color))
            dro.set_property("dtg_color", self._get_RGBA_color(self.dtg_color))
            dro.set_property("homed_color", self._get_RGBA_color(self.homed_color))
            dro.set_property("unhomed_color", self._get_RGBA_color(self.unhomed_color))
            dro.set_property("actual", self.dro_actual)
            dro.connect("clicked", self._on_DRO_clicked)
            dro.connect('axis_clicked', self._on_DRO_axis_clicked)
            self.dro_dic[dro.get_property("name")] = dro

`

c-morley, 2022-12-03:
So I would think you could set these from glade directly. My quick test seems to indicate that many of the properties are not set by default.
 So i guess I'm asking how does your program set properties such as joint number in your program?

alkabal, 2022-12-03:
Hello
Thank you i have also see this code but was unsure.

Regarding to second comment i think you push to evidence something that is fully unmanaged for the moment, the code is from 2.7 compatibility and i think nobody do the real change need for set axis and joint.

Damned another big rework is needed. (i'm unsure about saying many thanks lool)

for reference python code for jog button : 
https://github.com/alkabal/PSNG_V3/blob/bf2f037b45534920354c5dd90e284213fbfbfc98/Your_Config_Folder/nc_subroutines/psng/python/jog.py#L154

and glade code for Dro : (nothing related to dro find inside pyton)
https://github.com/alkabal/PSNG_V3/blob/bf2f037b45534920354c5dd90e284213fbfbfc98/Your_Config_Folder/nc_subroutines/psng/psng_mod.glade#L423

alkabal, 2022-12-03:
i hope that i'm on the good road i stop for today but adding some code from Gmoccapy i see in my logs : 

- Number of joints = 3
- 3 COORDINATES found = xyz
- joint 0 = axis x
- joint 1 = axis y
- joint 2 = axis z
- {0: 'x', 1: 'y', 2: 'z'}
- gladevcp: trouble looking for handlers in 'python': 'ProbeScreenJog' object has no attribute '_get_RGBA_color'

alkabal, 2022-12-03:
Success but do not ask me for what the code do, i have only copy paste from gmoccapy. 
Really many greetings to you !!

alkabal, 2022-12-03:
Can i ask what did you think using import or is better to copy the things needed ?: 
        from gmoccapy import widgets       # a class to handle the widgets
        from gmoccapy import getiniinfo    # this handles the INI File reading so checking is done in that module

alkabal, 2022-12-05:
Another day another (two) problems

It is normal toggling machine power ON/OFF without any move (fresh startup) i get executed the ON_ABORT_COMMAND ?

Any suggestion for the reason i can't get italic change anymore with 2.9 ?
https://github.com/alkabal/PSNG_V3/blob/cd1744427f6ef062fd573d6d8c1d6b0a5c6b40f8/Your_Config_Folder/nc_subroutines/psng/python/base.py#L887

The code seem to be excuted correctly related to Return or Escape but never get the italic

c-morley, 2022-12-05:
Yes you must look for it in the on_abort_command ngc script.

http://linuxcnc.org/docs/devel/html/remap/remap.html#_error_handling_dealing_with_abort

c-morley, 2022-12-05:
gmoccapy's widgets and get_info would be very handy.
also http://linuxcnc.org/docs/devel/html/gui/gladevcp-libraries.html

alkabal, 2022-12-05:
Oh verry helpfull upgraded on_abort i did not show that before hopefully i understand (and verified) this does not exist in 2.8.

Sorry i did not fully understand if you says yes use "from gmoccapy import" is bette than include the whole code in psng 

Another big thanks without your help imo i can't find some of the solution proposed, thanks a lot.

alkabal, 2022-12-05:
I suspect a error in the doc (swapped estop/machine-off) must be something like that:

```
o100 if [#1 EQ 5]
    (estop released)
o100 elseif [#1 EQ 6]
    (estop pressed)
o100 elseif [#1 EQ 7]
    (MSG, machine turned off) 
o100 elseif [#1 EQ 8]
    (MSG, abort pressed)
o100 else
    (DEBUG, error parameter is [#1])
o100 endif
```

Roguish000, 2023-01-03:
alkabal,  hello,    have you gotten PNSG to work in 2.10?   I would really like to use it.   I currently use ProbeScreen v2 on my old mill, and am converting a new mill to 2.10.

alkabal, 2023-01-04:
HelloYes it work fine but I need to update my linuxcnc for verify if nothing have change and I also need to update my gut.Other things to consider I have do many change (enhancement) from probescreen v2If you like to test and have some difficulty you can ask for help Br

alkabal, 2023-01-08:
Code on my GitHub is the last revision.Br

Roguish000, 2023-04-20:
Hello again.   Is your probe code working on Master?  I was forced to update my install to Bullseye and the latest 2.9.0-pre1 Linuxcnc.  and I really need Probing.

alkabal, 2023-04-20:
HelloI do not update from some month but in theory yes this is do for 2.9 2.10Maybee something changed recently need to patch the code, I apologize vu I haven’t free enough time for the moment so I can’t update et test again before some months.But I keep contact if you ask something I will try to reply. Br

andypugh, 2023-10-03:
I have forked psng. Please raise issues there rather than on LinuxCNC. 
https://github.com/andypugh/probe-screen-ng/commits/master

I will try to update it to Py3 / GTK3

closed 2023-10-03 by andypugh