LinuxCNC/linuxcnc #2136: Commit #2e2d75c breaks arm64 builds

phillc54, 2022-11-10:
As the title says, commit #2e2d75c breaks arm64 builds with the following output in terminal during building:

`Compiling realtime hal/drivers/hal_speaker.c
hal/drivers/hal_speaker.c:76:10: fatal error: sys/io.h: No such file or directory
   76 | #include <sys/io.h>
      |          ^~~~~~~~~~
compilation terminated.
make: *** [Makefile:1227: objects/rthal/drivers/hal_speaker.o] Error 1
`

petterreinholdtsen, 2022-11-10:
[Phillip Carter]
> As the title says, commit #2e2d75c breaks arm64 builds with the
> following output in terminal during building:

I believe the RTAI build was already broken on arm64, as this driver was
built when RTAI was enabled.  The change only exposed the same build
failure for uspace builds.

> `Compiling realtime hal/drivers/hal_speaker.c
> hal/drivers/hal_speaker.c:76:10: fatal error: sys/io.h: No such file or directory
>    76 | #include <sys/io.h>
>       |          ^~~~~~~~~~
> compilation terminated.
> make: *** [Makefile:1227: objects/rthal/drivers/hal_speaker.o] Error 1
> `

According to
<URL: https://stackoverflow.com/questions/68300986/linux-c-sys-io-h-not-found-on-arm64-but-on-amd64-it-is-there >
this is an architectural difference issue.  I believe the solution is to
disable this driver on all non-x86 platforms, as it will only work
there.

-- 
Happy hacking
Petter Reinholdtsen

phillc54, 2022-11-10:
If I checkout the commit before that one it builds fine.

petterreinholdtsen, 2022-11-10:
[Phillip Carter]
> If I checkout the commit before that one it builds fine.

Are you using the RTAI or uspace build profile?  I assume you use the
uspace profile, as the commit in question move the hal_speaker module
out of the RTAI specific block and into the common block.

-- 
Happy hacking
Petter Reinholdtsen

phillc54, 2022-11-10:
It is a uspace build. AFAIK there is no RTAI for arm processors.

petterreinholdtsen, 2022-11-10:
Perhaps something along these lines can work?

diff --git a/src/Makefile b/src/Makefile
index 4349f19378..86d50f6181 100644
--- a/src/Makefile
+++ b/src/Makefile
@@ -63,6 +63,12 @@ TRIVIAL_BUILD=no
 endif
 endif
 
+ifeq ($(shell uname -m),i686)
+X86ARCH = true
+endif
+ifeq ($(shell uname -m),x86_64)
+X86ARCH = true
+endif
 
 # Beautify output
 # ---------------------------------------------------------------------------
@@ -949,8 +955,10 @@ sampler-objs := hal/components/sampler.o $(MATHSTUB)
 # Subdirectory: hal/drivers
 obj-$(CONFIG_HAL_PARPORT) += hal_parport.o
 hal_parport-objs := hal/drivers/hal_parport.o $(MATHSTUB)
+ifeq ($(X86ARCH),true)
 obj-$(CONFIG_HAL_SPEAKER) += hal_speaker.o
 hal_speaker-objs := hal/drivers/hal_speaker.o $(MATHSTUB)
+endif
 ifneq ($(BUILD_SYS),uspace)
 obj-$(CONFIG_PCI_8255) += pci_8255.o
 pci_8255-objs := hal/drivers/pci_8255.o
@@ -1284,7 +1292,9 @@ endif
 ../rtlib/hal_ax5214h$(MODULE_EXT): $(addprefix objects/rt,$(hal_ax5214h-objs))
 ../rtlib/hal_ppmc$(MODULE_EXT): $(addprefix objects/rt,$(hal_ppmc-objs))
 ../rtlib/hal_skeleton$(MODULE_EXT): $(addprefix objects/rt,$(hal_skeleton-objs))
+ifeq ($(X86ARCH),true)
 ../rtlib/hal_speaker$(MODULE_EXT): $(addprefix objects/rt,$(hal_speaker-objs))
+endif
 ../rtlib/opto_ac5$(MODULE_EXT): $(addprefix objects/rt,$(opto_ac5-objs))
 ../rtlib/scope_rt$(MODULE_EXT): $(addprefix objects/rt,$(scope_rt-objs))
 ../rtlib/hal_lib$(MODULE_EXT): $(addprefix objects/rt,$(hal_lib-objs))

There is also the <rtapi_io.h> include used by hal_parport.c which might
be an abstraction layer for this, but I suspect it will just provide a
dummy hal_speaker driver on arm platforms.

-- 
Happy hacking
Petter Reinholdtsen

phillc54, 2022-11-10:
Thank you, it built successfully with no apparent errors. I ran a few sim configs with no issues.

PR #2138 (2022-11-12): Only build hal_speaker driver on x86, and adjust to work for uspace and RTAI
The ioperm() code is only needed in uspace mode, and the <sys/io.h> file is most useful in user space.

Fixes #2136.
commit 23c8c37d (2022-11-12): Only build hal_speaker driver on x86, and adjust to work for uspace and RTAI

The ioperm() code is only needed in uspace mode, and the <sys/io.h> file
is most useful in user space.

Fixes #2136.
commit 001a4eb2 (2022-11-13): Only build hal_speaker driver on x86, and adjust to work for uspace and RTAI

The ioperm() code is only needed in uspace mode, and the <sys/io.h> file
is most useful in user space.

Fixes #2136. Added by cherry-pick
closed 2022-11-12 by andypugh