LinuxCNC/linuxcnc #2140: Unicode characters from man pages not correctly converted to HTML

hansu, 2022-11-12:
Man pages like this
[image]

Are converted to HTML like this (http://linuxcnc.org/docs/2.9/html/man/man9/toggle.9.html):

<h2>NAME
<a name="NAME"></a>
</h2>

<p style="margin-left:11%; margin-top: 1em">toggle -
’push-on, push-off’ from momentary
pushbuttons</p>

<h2>SYNOPSIS
<a name="SYNOPSIS"></a>
</h2>

loadrt toggle [count=N\|names=name1[,name2...]] 

<h2>DESCRIPTION
<a name="DESCRIPTION"></a>
</h2>

<p style="margin-left:11%; margin-top: 1em">â
ââââ ââââ
<br>
in : âââââââ
ââââââââââ
âââ <br>
â
âââââââââââââ
<br>
out: âââââââ
ââââââ</p>

petterreinholdtsen, 2022-11-13:
[Hans Unzner]
> Are converted to HTML like this
> (http://linuxcnc.org/docs/2.9/html/man/man9/toggle.9.html):

This rule in docs/src/Submakefile seem to be doing the conversion:

$(DOC_DIR)/html/man/%.html: $(DOC_DIR)/man/%
        @echo Formatting $(notdir $<) as HTML
        @mkdir -p $(dir $@)
#   Attention! This is a temporary fix until Groff 1.23 is released. Should be
#   set back to "-m man" when 1.23 is available.
#       This fixes the formatting of subsections and appearance in the toc.
#   It uses the local an-old.tmac instead of the one which is shipped with Groff.
        @(cd $(DOC_DIR)/man; groff -Thtml -m an-old-fixed -M . stylesheet.9 $(patsubst $(DOC_DIR)/man/%,%,$<)) > $@

My Bullseye machine uses groff version 1.22.4-6.  If we instead use
pandoc to convert (pandoc -t html), we get UTF-8 characters in the HTML
document, but sadly the space character used are not displayed properly,
so the end result is unreadable too.  Perhaps we can tell pandoc somehow
to preserve the whitespace in that block?

<URL: https://stackoverflow.com/questions/52732988/nroff-groff-does-not-properly-convert-utf-8-encoded-file >
claim groff do not really understand UTF-8, which is my experience too.
Luckily the 'man' command handle UTF-8 correctly.

-- 
Happy hacking
Petter Reinholdtsen

petterreinholdtsen, 2024-07-06:
I managed to find a solution this issue during this weekends developer
gathering using preconv.

-- 
Happy hacking
Petter Reinholdtsen

commit 1130c9b2 (2024-07-06): Pass generated component man pages through preconv for UTF-8 support.

Adjusted halcompile to output UTF-8 header for all generated man pages,
to ensure preconv always know the charset, and adjust the component
build rules to filter the output through preconv to convert any UTF-8
characters to the format understood by groff.  Added groff-base
as a build dependency for preconv.

Fixes #2140.
closed 2024-07-06 by andypugh