LinuxCNC/linuxcnc #2143: linuxcnc multi-channel support

TealerLinkGuo, 2022-11-16:
Is it possible for linuxcnc to provide multi-channel support or ideas for source code modification？
Multi-channel means that linuxcnc can control multiple different devices through multiple channels. Each channel operates independently of each other, they have their own separate RS274 parser and EMCMOT path planner. Channel waiting and synchronization can be realized by reading flag bits between multiple channels.
Does anyone have a good idea for implementing this?

andypugh, 2022-11-16:
I feel that one way to do this would be to allow multiple LinuxCNC instances sharing a common shared-memory area.

This is currently not allowed, though I think that if the code that checks was disabled then something would work. 

As things stand each instance would be unaware of the other, so there would be no HAL namespace clash, but there would be issues with each instance using the same memory for different things. 

There might also be issues with sharing the realtime threads. 

So, the two instances would need to be aware of each other at the HAL level. And at that point there would need to be steps taken to avoid namespace clashes. 

I think that sharing HAL is the best way to achieve interprocess cooperation. 

The alternative is to have two completely separate LinuxCNC instances and some other type of interprocess communication, but the possible issues of sharing the realtime threads would still need to be addressed. There might be ways to run separate threads on separate cores. 

It is possible to run LinuxCNC on two separate PCs and communicate between them with IO pins, linuxcncrsh, udp packets  (and anything else that you can think of)

c-morley, 2022-11-17:
It's really too bad some more of Michael code didn't get in:
https://www.youtube.com/watch?v=Zk_KiWrhi5Y
Not sure if the code is anywhere to look at.

c-morley, 2022-11-17:
It might be in here:
https://github.com/LinuxCNC/linuxcnc/tree/unified-build-candidate-3

andrewheeler82, 2022-11-17:
As far as I know, only the Siemens 840D sl currently offers real multi-channel operation. We have a few machines with the 840D in our company. Heidenhain TNC 640 is a single-channel control like current LinuxCNC.

https://youtu.be/c_mJeGuolZ8

GPaso, 2024-07-31:
I am looking forward to an implementation of something like this to be used on rotary transfer machines.

rmu75, 2025-12-10:
It is possible to run multiple linuxcnc instances on one machine (with a patch, see #2722), synchronization would have to be implemented in something external that connects to both instances.