LinuxCNC/linuxcnc #2150: linux cnc at burdelox

alejandrolleida, 2022-11-19:
in bourdeloz we are glad to have linux cnc
it comes with a lot of remote controls for your f-uked pc like vnc server ssh server ftp server samba server all are connected with webcam server and the kit comes with ritual people like dount dot com all time of day week year f ukin g with you for rituals

closed 2022-11-19 by SebKuzminsky

petterreinholdtsen, 2022-11-19:
[alejandrolleida]
> in bourdeloz we are glad to have linux cnc

Glad to hear that.

Did you just want to give some positive feedback to the LinuxCNC
developers, or is there some issue hidden in your message I just fail to
pick up?

Btw, where/what is bourdeloz?

-- 
Happy hacking
Petter Reinholdtsen

alejandrolleida, 2022-11-20:
yes all 0positive feetback
bourdeldeaoz is a french city is very libertine from there is esay to go to badajodeoz and to bujaradeoz this sites have pretty rituals with a lot of colectives organizations them are very social until the end, I kwnwn a lot of people in therapy until them understand all of us Must to Love this collectives, industries, commerce,  organizations, business, loving, rituals, clubs of the world inclusive eastkolm victims.

alejandrolleida, 2022-11-20:
""""""/*Happy hacking
Petter Reinholdtsen
Did you just want to give some positive feedback to the LinuxCNC
developers, or is there some issue hidden in your message 
"*I just fail to pick up?*"
****/////
Did you said fall in to your pickup 4x4?
With me?
Have you a short dress?
$?