LinuxCNC/linuxcnc #2162: MB2HAL: Error when more than 5 `PIN_NAMES`

hansu, 2022-11-28:
This works:
`PIN_NAMES=INPUT1,INPUT2,INPUT3,INPUT4,INPUT5`
This not:
`PIN_NAMES=INPUT1,INPUT2,INPUT3,INPUT4,INPUT5,INPUT6,INPUT`

Debugoutput:
```
mb2hal parse_transaction_section DEBUG: [TRANSACTION_00] [PIN_NAMES] 
[INPUT1,INPUT2,INPUT3,INPUT4,INPUT5,INPUT6,INPUT7]
mb2hal parse_transaction_section DEBUG: [TRANSACTION_00] [NELEMENTS] [17]   <- This is not correct (should be 7 as stated above)
```

Reported here: https://forum.linuxcnc.org/24-hal-components/47495-mb2hal-reading-multiple-coils-limited-to-5-when-using-names

petterreinholdtsen, 2022-11-28:
[Hans Unzner]
> mb2hal parse_transaction_section DEBUG: [TRANSACTION_00] [PIN_NAMES] 
> [INPUT1,INPUT2,INPUT3,INPUT4,INPUT5,INPUT6,INPUT7]
> mb2hal parse_transaction_section DEBUG: [TRANSACTION_00] [NELEMENTS]
> [17]   <- This is not correct (should be 7 as stated above)

Perhaps you can write a script in tests/ to trigger the error, to ensure
the error do not sneak up on us again once fixed?

-- 
Happy hacking
Petter Reinholdtsen

hansu, 2022-11-28:
Yeah that should be possible

PR #2167 (2022-12-01): mb2hal: fix error when PIN_NAMES > 5
Do you guys agree that this was the intention of the author?

It increases the size of the buffer holding the pin names by steps of `NAME_ALLOC_SIZE` when more pins are defined.

Resolves #2162
closed 2022-12-28 by hansu