LinuxCNC/linuxcnc #2169: Linear Z move executing G2. Does not appear in G3

tlight9, 2022-11-29:
## Here are the steps I follow to reproduce the issue:

 1. In LCNC 2.9.0 running this program with G2 before any other movement after homing
%
G20
G2 X0 Y0 Z-1 J4 F1000
G91
G2 X0 Y0 Z-1 J4 F1000
G2 X0 Y0 Z-1 J4 F1000
G2 X0 Y0 Z-1 J4 F1000
M30
%
 2.
 3.

## This is what I expected to happen: I expect a clockwise helix

## This is what happened instead: See attached file of expected path and actual in AXIS SIm

## It worked properly before this: It is believed this bug may have been introduced in late 2.8.0 version
## Information about my hardware and software: Intel(R) Core(TM) i5-6600 CPU @ 3.30GHz
Also tested on Rpi4 with the same results
 
 * I am using this Linux distribution and version (often, shown by `lsb_release -a`):
 * I am using this kernel version (shown by `uname -a`):Linux bookworm 6.0.0-3-rt-amd64 #1 SMP PREEMPT_RT Debian 6.0.7-1 (2022-11-05) x86_64 GNU/Linux
 
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):AXIS Sim
 
[image]

powerio25, 2022-12-03:
Hi at all.
I had this problem, some time ago, because I had added on my master, the patch:
#1528 Short arcs of very large radius were being interpreted as full … .
I reversed the patch and the problem disappeared.

This is the patch, return at - situation and problem disappeared.

src/emc/rs274ngc/gcodemodule.cc
@@ -24,6 +24,7 @@
#include "interp_return.hh"
#include "canon.hh"
#include "config.h"             // LINELEN
+#include "units.h"

int _task = 0; // control preview behaviour when remapping

@@ -998,11 +999,17 @@ static PyObject *rs274_arc_to_segments(PyObject
*self, PyObject *args) {

    double theta1 = atan2(o[Y]-cy, o[X]-cx);
    double theta2 = atan2(n[Y]-cy, n[X]-cx);
-
-   if(rot < 0) {
-        while(theta2 - theta1 > -CIRCLE_FUZZ) theta2 -= 2*M_PI;
-    } else {
-        while(theta2 - theta1 < CIRCLE_FUZZ) theta2 += 2*M_PI;
 +   /* Issue #1528 1/2/22 andypugh */
 +   /*_posemath checks for small arcs too, but uses config units */
 +   double len = hypot(o[X]-n[X], o[Y]-n[Y]) * (25.4 *
GET_EXTERNAL_LENGTH_UNITS());
    /* If the signs of the angles differ, make them the same to allow
monotonic progress through the arc */
 +   /* If start and end points are nearly identical, then interpret as
a full turn */
  + if(rot < 0) { // CW G2
  +     if (theta1 < theta2) theta2 -= 2*M_PI;
  +      if (len < CART_FUZZ) theta2 -= 2*M_PI;
  +  } else { // CCW G3
  +      if (theta1 > theta2) theta2 += 2*M_PI;
  +      if (len < CART_FUZZ) theta2 += 2*M_PI;
    }

    // if multi-turn, add the right number of full circles
10

src/libnml/posemath/_posemath.c
@@ -1812,6 +1812,7 @@ int pmCircleInit(PmCircle * const circle,
    } else {
        circle->angle = acos(dot);
    }

    /* now angle is in range 0..PI . Check if cross is antiparallel to
       normal. If so, true angle is between PI..2PI. Need to subtract
from
       2PI. */
@@ -1820,8 +1821,10 @@ int pmCircleInit(PmCircle * const circle,
    if (d < 0.0) {
        circle->angle = PM_2_PI - circle->angle;
    }
-
-   if (circle->angle > -(CIRCLE_FUZZ) && circle->angle <
(CIRCLE_FUZZ)) {
 +   /* Issue #1528 24-Jan-2022. Additional test for nearly-
straight   *
  +   * short arcs of very large radius becoming
circles               */
   + pmCartCartDisp(start, end, &d);
   + if (d < CART_FUZZ){
        circle->angle = PM_2_PI;
    }

@@ -1830,6 +1833,9 @@ int pmCircleInit(PmCircle * const circle,
        circle->angle += turn * 2.0 * PM_PI;
    }

   + /* FIXME: some code has an unguarded division by circle->angle */
   +if (circle->angle == 0) circle->angle += CIRCLE_FUZZ;
   +
    //Default to invalid
/* if 0'ed out while not debugging*/
#if 0
1
src/libnml/posemath/posemath.h
@@ -713,6 +713,7 @@ extern "C" {
// FIXME setting this to be an order of magnitude smaller than canon's
shortest
// allowed segment. This is still larger than TP's smallest position,
so it may
// be silently causing trouble.
// andypugh 5/2/22 This seems to be interpreted to be in config units.
#define CART_FUZZ (1.0e-8)
/* how close a cartesian vector's magnitude must be for it to be
considered
   a zero vector */

andypugh, 2022-12-03:
I am starting to think that this is a more serious problem than I thought, or it has pointed out a more serious problem. 

Using this G-code: 
````
%
G21
G90
G2 X0 Y0 Z-1 J4 F1000
G2 X0 Y0 Z-2 J4 F1000
G2 X0 Y0 Z-3 J4 F1000
G2 X0 Y0 Z-4 J4 F1000
G0 X0 Y0 Z0
M30
%
````
Which should be a simple spiral, as might be used for thread milling (by a CAM unaware of the P parameter) 

With the current 2.9 version gives:

[image]

Thinking that this line that was added by the at-fault patch:
````
    /* FIXME: some code has an unguarded division by circle->angle */
    if (circle->angle == 0) circle->angle += CIRCLE_FUZZ;
````
And that 0 + CIRCLE_FUZZ is not less than CIRCLE_FUZZ (a test elsewhere) I tried
````
    /* FIXME: some code has an unguarded division by circle->angle */
    if (circle->angle == 0) circle->angle += CIRCLE_FUZZ / 2;
````
Which gives: 
[image]

Which is worse. 

So then I tried commenting-out that entire line:
````
    /* FIXME: some code has an unguarded division by circle->angle */
    //if (circle->angle == 0) circle->angle += CIRCLE_FUZZ / 2;
````
And that was even worse. 

[image]

With the spiral being mainly ignored. 

This particular line might be a red-herring as the real issue seems to be that the cartesian distance between the end points of a spiral is never small, the test should have been limited to the plane of the arc. The surprise might be that it works at all. 

I can't do any more on this today, but will try to look at it tomorrow.

andypugh, 2022-12-21:
The original reporter confirms this fixed with f72bd0bd0e31989765df06b9a3478555efe36e40

closed 2022-12-21 by andypugh