LinuxCNC/linuxcnc #2173: pncconf does not insert stepgen dpll setup boilerplate on 7I96S

pcw-mesa, 2022-12-01:
## Here are the steps I follow to reproduce the issue:

 1. Run pncconf and select 7I96S
 2. Create 7I96S configuration
 3.

## This is what I expected to happen:
7I96S configurations (and all Ethernet card configurations)  should insert DPLL stepgen setup in hal file
## This is what happened instead:
No DPLL stepgen setup was added
## It worked properly before this:

## Information about my hardware and software:

 * I am using this Linux distribution and version  2.10

phillc54, 2022-12-01:
I just ran up a config and I am not really sure what is missing. I have attached the resulting hal file.

[my_LinuxCNC_machine.hal.txt](https://github.com/LinuxCNC/linuxcnc/files/10136850/my_LinuxCNC_machine.hal.txt)

pcw-mesa, 2022-12-02:
That's strange, a forum user reported the issue and I duplicated it on a test installation 
at work but now it seems OK on my home machine. fairly mysterious...

pcw-mesa, 2022-12-02:
I will close this since the current code works on this test PC

closed 2022-12-02 by pcw-mesa