LinuxCNC/linuxcnc #2200: PNCconf Spindle Setup Issue

BSHoekstra, 2022-12-10:
We want to enage and disengage a relay through a MESA card GPIO that turns a spindle motor on and off, but PNCconf does not setup the .hal file correctly for M3, M4 and M5 commands to active and deactivate the hal pin.  I was able to identify the solution after many hours of troubleshooting, so I wanted to share this information in hopes that this solution can be automatically programmed into PNCconf by the developers.  Note that this is an issue with both Linuxcnc/Axis Version 2.8.4 and Linux/Axis Buildbot 2.9.

## Here are the steps to follow to reproduce the issue:

 1.  Run PNCconf.  
 2.  On the Mesa Card 0 Tab, select
         1)  Board name:  5i25-Internal Data
         2)  7i76X2 - With One 7i76
3.  On the I/O Connector 3 Tab, select
        Num 012: Function - Spindle Enable, Pin Type:  GPIO Output
4.  Complete the PNCconf installation
5.  Open the .hal file and you will find the following 
                  # --- SPINDLE-ENABLE ---
                  setp hm2_5i25.0.gpio.012.is_output true
                  net spindle-enable  =>     hm2_5i25.0.gpio.012.out
6.  For the program to work correctly you have add a line manually as shown below:
                  # --- SPINDLE-ENABLE ---
                  net spindle-enable <= spindle.0.on
                  setp hm2_5i25.0.gpio.012.is_output true
                  net spindle-enable  =>     hm2_5i25.0.gpio.012.out

Hope this helps others.

Regards, 

Barry :)

c-morley, 2022-12-18:
I tested 2.9 and don't find this problem.
The line: net spindle-enable <= spindle.0.on
is placed under the heading: # ---setup spindle control signals---
in the same file.
Did you select spindle-enable off the drop down list or type it in?
Actually it shouldn't matter, net spindle-enable <= spindle.0.on is always placed in the file IFAICT.

BSHoekstra, 2022-12-22:
You are right.   I started a new PNCConf Config Wizard setup and sure enough, as you pointed out, it setup the Hal File correctly and the Hal file worked perfectly. 

                  The line: net spindle-enable <= spindle.0.on
              was under the heading: # ---setup spindle control signals--- 

Sorry for completely missing this as I was going through the trouble-shooting process to get my CNC Machine working (which it is now), and thank you for responding to help me up the learning curve!  :)  I very much appreciated it.

Regards,

Barry :)

closed 2022-12-22 by BSHoekstra