LinuxCNC/linuxcnc #2204: Halcompile does not include modbus headers..............

Roguish000, 2022-12-12:
Ok.  running on both a real machine, and a VM, of debian 12 bookworm, with the latest and greatest versions of linuxcnc and everything associated, including a RIP build, all master braches.

Testing on an **unmodified** existing user comp.
I tried to use halcompile to create a very slightly version of the vfdb_vfd.c comp.  Tried all kinds of things.
Finally was able to read the halcompile message better.    See the image.
Note the -I in the line.  '-I/home/noel/comp -I/usr/include -I/usr/include/linuxcnc'
But the modbus header files are in  /usr/include/modbus
So they are not found.
I copied all 4 modbus header files into /usr/include/linuxcnc.....

and voila.  it worked.  well not really.  got past that error...............now more errors.  see the second, longer image.
 still not sure, not a programmer. 
but this is an existing and working user comp.
shouldn't it compile correctly?

A little help please......

[image]

[image]

petterreinholdtsen, 2022-12-12:
[Roguish]
> A little help please......

I had a look, and to me it look like your component need some extra -I
to the compile step and -l arguments to the linker step.  I had a look
at the source of halcompile to try to figure out how to do so, and the
trick seem to be two options, extra_compile_args and extra_link_args
that can be added to a .comp file, for example like this:

  option extra_compile_args "-I/usr/include/modbus";
  option extra_link_args "-lm -lmodbus";

The former should fix your problem including <modbus.h>, and the latter
should fix the linking issue with round and modbus_* methods.  Not sure
where the iniFindInt method come from, so no library suggestion there.

It was not obvious to me how to activate similar options for .c based
components, so I am not sure if this solve your problem.

In any case I suspect the should be documented in the halcompile(1)
manual page, which where I failed to find them.  They are documented in
<URL: http://linuxcnc.org/docs/devel/html/hal/comp.html#_options >.

-- 
Happy hacking
Petter Reinholdtsen

andypugh, 2022-12-12:
There might be a way to pass the flag via a 

#pragma GCC ... 

Line in the source file. 

But, pragmatically (sic) you can probably look for Makefile.modinc on your system and add the options above to an EXTRA_CFLAGS line.

andypugh, 2022-12-25:
I have created a test branch (https://github.com/LinuxCNC/linuxcnc/tree/andypugh/halcompile_opts )which allows the following command line:

````
halcompile --userspace --install --extra_compile_args="-I/usr/include/modbus" --extra_link_args="-lm -lmodbus" hal/user_comps/vfdb_vfd/vfdb_vfd.c
````
Which nearly works (no more modbus issues) but then has trouble linking the inifile.h functions. And I haven't got anywhere with figuring how to link that.

petterreinholdtsen, 2022-12-25:
[andypugh]
> I have created a test branch
> (https://github.com/LinuxCNC/linuxcnc/tree/andypugh/halcompile_opts)
> which allows the following command line:

Cool.    What about a test case in tests/?

> Which nearly works (no more modbus issues) but then has trouble
> linking the inifile.h functions. And I haven't got anywhere with
> figuring how to link that.

What is the problem with listing several -I and -l entries?

-- 
Happy hacking
Petter Reinholdtsen

andypugh, 2022-12-26:
> Cool. What about a test case in tests/?

Seems a lot of trouble for something that it is unlikely anyone will ever use...

> has trouble linking the inifile.h functions. And I haven't got anywhere with figuring how to link that.
> What is the problem with listing several -I and -l entries?

No problem at all, I just haven't found a -l (as in, L) that links inifile.h 
(And isn't it a bit daft of github to choose a font where I and l are indistinguishable)

petterreinholdtsen, 2022-12-26:
[andypugh]
>> Cool. What about a test case in tests/?
>
> Seems a lot of trouble for something that it is unlikely anyone will
> ever use...

Is there something I can do to make it less of a trouble?  I've written
<URL: http://linuxcnc.org/docs/devel/html/code/writing-tests.html >
trying to make it easier.  The point is really to ensure linuxcnc behave
predictably and according to documentation, the number of users is
beside the point.

Every feature tested make it easier to have confidence in new releases.

> (And isn't it a bit daft of github to choose a font where I and l are
> indistinguishable)

Absolutely. :)

--
Happy hacking
Petter Reinholdtsen

andypugh, 2022-12-26:
A test would presumably require me to write a userspace C component that relied on a compiler and linker flag, and that would fail to compile without it. 
So, it needs to be a valid component, it needs to rely on an external library (not modbus, modbus is optional), and needs to give predictable output both with and without the extra flags. 
Unless the only user who has ever complained about this returns to the thread I will probably not even bother putting the modification in.

Roguish000, 2023-01-03:
andypugh.  I'm here.  been occupied just trying to get my system to run, with all the ethernet timing problems, and qtvcp changes.   I definitely want to continue this issue......

andypugh, 2023-01-07:
Something which has just occurred to me... Are the modbus libraries realtime-safe? Part of the reason we have rtapi is to be sure that all the functions used never busy-wait and perform operations atomically (so that if they are interrupted they don't leave a variable or data half-updated)

Roguish000, 2023-01-08:
Way beyond my paygrade........ and knowledge level........  
but one should hope so.

andypugh, 2023-01-08:
> Way beyond my paygrade........ and knowledge level........
> but one should hope so.

Actually, you can ignore the question, this is a userspace component isn't it?

jepler, 2023-01-08:
I think the solution of making the comp "option extra_compile_args" and "extra_link_args" controllable from the commandline is a fine idea. It might be good to say that (for now) the effect of having them specified in both places (commandline and .comp) is unspecified, so that we can later figure out if there's a sensible way to combine options from both sources, or learn which one should override the other based on experience.

jepler, 2023-01-08:
Also I'm not sure how or where it should be noted that not all .comp files in the source tree can simply be compiled by halcompile; but I think this has tripped folks up in the past as well. (more so for user comps than rt comps)

andypugh, 2023-01-09:
I managed to compile vfdb_vfd using this updated halcompile and

````
halcompile --userspace --install --extra_compile_args="-I/usr/include/modbus" --extra_link_args="-lm -lmodbus -llinuxcncini" hal/user_comps/vfdb_vfd/vfdb_vfd.c
````

@Roguish000 You should be able to overwrite the halcompile on your system with the halcompile.g file from here: https://github.com/LinuxCNC/linuxcnc/blob/andypugh/halcompile_opts/src/hal/utils/halcompile.g
(need to click "raw" then save-as (probably as .py not .g)

Roguish000, 2023-01-14:
andypugh,  thanks for working on this.   apologies for being a bit late...
I have been able to get the file you created and referred to above.  I have also just done a git clone and RIP (yesterday, jan 13).
Your file and the one in the git source appear the same (I did a diff on them).
,,,,what is the difference between a .g and a .py file?  I know what a .py  python file is.  just never heard of a .g  ....

So if I want to halcompile my modified vfdb_vfd.c file, where should it place it?  like exactly what directory?  particulary if I'm using the RIP (after I have done the environment).
I've also copied your command line and it seems to make sense, but I do not see those --extra...  parameters in any of the docs, either man pages or html.
I'm a strong believer in RTFM, which I rely on.
So any little help would be greatly appreciated.

andypugh, 2023-01-14:
The new options are documented here: http://linuxcnc.org/docs/2.9/html/hal/comp.html (including the exact command line that you need)
I think that it doesn't matter where the file is as long as it is in the same one as you run halcompile from.

Roguish000, 2023-01-14:
Andy, well, I went back and did a --enable-build-documentation.   took a bit to get firefox to open a local file, but I got it.
reading the docs now.

Roguish000, 2023-01-14:
Andy, once again, thanks.   but no success yet.
I did notice one thing in the docs.  in the description of the --extra_link_args  you cite the temp directory  "-L." and in the example command line you use lower case L (l).  not sure which is correct,  but I know linux is very case sensitive.
The extra_compile_args  description and example use are the OK.

I noticed you referred to 2.9 docs.  So I uninstalled 2.10 and installed 2.9 from Seb's secret buildbot.

Now, I just can't get it to work.   see the image.  working in the directory with the comp file, I have tried numerous combinations of the command to try and determine the problem with not luck.  Running as installed, not RIP which is 2.10.

[image]

andypugh, 2023-01-15:
The new stuff should be in both 2.9 and 2.10. 
Possibly the version on the secret buildbot (so secret that I don't know about it...) hasn't caught the update yet?

andypugh, 2025-04-13:
I think this is probably sorted out now? Re-open if not.

closed 2025-04-13 by andypugh